OpenCPN/OpenCPN #2054: glChartCanvas.cpp missing include GL/glx.h

mgrouch, 2020-10-06:
glChartCanvas.cpp missing #include <GL/glx.h>
Needed to compile at least in raspberry pi

leamas, 2020-10-06:
Please make a PR linked to this issue. It would also be good if you could give some more context: exact built version (commit), cmake commands, compiler version, os version, ...

Is this the same as #2053, or a separate issue?

mgrouch, 2020-10-06:
On Raspberry pi 4 with 64 bit kernel

build wxWidgets 3.1.4 like this:

sudo apt-get -y install build-essential cmake gettext git-core libgtk-3-dev libgtk-3-0

git clone -b v3.1.4 --recurse-submodules https://github.com/wxWidgets/wxWidgets.git
cd wxWidgets || exit 255

mkdir build-gtk
cd build-gtk || exit 255

../configure --with-gtk=3 --with-libpng=builtin --with-libjpeg=builtin --with-libtiff=builtin --with-regex=builtin
--with-zlib=builtin --with-expat=builtin --enable-unicode --enable-aui --with-opengl
LDFLAGS="-static-libgcc -static-libstdc++"
make -sj5
sudo make install
sudo ldconfig

then build OpenCPN against it:

sudo apt-get -y install build-essential cmake gettext git-core gpsd gpsd-clients libgps-dev
libglu1-mesa-dev libgtk-3-dev libbz2-dev libtinyxml-dev libportaudio2 portaudio19-dev libcurl4-openssl-dev
libexpat1-dev libcairo2-dev libarchive-dev liblzma-dev libexif-dev libelf-dev libsqlite3-dev
bc bison flex libssl-dev python3 ddd htop

git clone https://github.com/opencpn/opencpn.git

cd opencpn && mkdir build
cd build || exit 255

cmake -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_GSHHS=ON -DOCPN_USE_GL=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo
-DOCPN_USE_CRASHREPORT=ON -DOCPN_ENABLE_PORTAUDIO=OFF -DOCPN_ENABLE_SYSTEM_CMD_SOUND=ON -DOCPN_FORCE_GTK3=ON ..

make -sj5

sudo make install

leamas, 2020-10-06:
Still lacking os and compiler info. What does `lsb_release -a `and` gcc --version `say?

That said, an added #include shouldn't make any harm; perhaps a PR is the proper way to address this.

mgrouch, 2020-10-06:
```
pi@raspberrypi:~/OCPN/opencpn/build $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
pi@raspberrypi:~/OCPN/opencpn/build $ gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
```

leamas, 2020-10-07:
OK, thanks. I still think a PR is appropriate here.

mgrouch, 2021-04-10:
fixed

closed 2021-04-10 by mgrouch