OpenCPN/OpenCPN #2055: Safe restart dialog crashes the whole app

mgrouch, 2020-10-06:
new OCPN_TimedHTMLMessageDialog() in safe_mode.cpp
why is it HTML dialog? what does it have to do with HTML?
It opens with size 0 and crashes the app.

Thanks

mgrouch, 2020-10-06:
On Raspberry pi 4 with 64 bit kernel

build wxWidgets 3.1.4 like this:

sudo apt-get -y install build-essential cmake gettext git-core libgtk-3-dev libgtk-3-0

git clone -b v3.1.4 --recurse-submodules https://github.com/wxWidgets/wxWidgets.git
cd wxWidgets || exit 255

mkdir build-gtk
cd build-gtk || exit 255

../configure --with-gtk=3 --with-libpng=builtin --with-libjpeg=builtin --with-libtiff=builtin --with-regex=builtin
--with-zlib=builtin --with-expat=builtin --enable-unicode --enable-aui --with-opengl
LDFLAGS="-static-libgcc -static-libstdc++"
make -sj5
sudo make install
sudo ldconfig

then build OpenCPN against it:

sudo apt-get -y install build-essential cmake gettext git-core gpsd gpsd-clients libgps-dev
libglu1-mesa-dev libgtk-3-dev libbz2-dev libtinyxml-dev libportaudio2 portaudio19-dev libcurl4-openssl-dev
libexpat1-dev libcairo2-dev libarchive-dev liblzma-dev libexif-dev libelf-dev libsqlite3-dev
bc bison flex libssl-dev python3 ddd htop

git clone https://github.com/opencpn/opencpn.git

cd opencpn && mkdir build
cd build || exit 255

cmake -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_GSHHS=ON -DOCPN_USE_GL=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo
-DOCPN_USE_CRASHREPORT=ON -DOCPN_ENABLE_PORTAUDIO=OFF -DOCPN_ENABLE_SYSTEM_CMD_SOUND=ON -DOCPN_FORCE_GTK3=ON ..

make -sj5

sudo make install

leamas, 2020-10-06:
There might be too many wheels turning here. You are on a unchartered platform (arm64), and also using an odd wxWidgets version.   Stable builds on Linux always uses the packaged versions, currently on the 3.0 series.

I don't know how used you are to the debian toolchain. If I was in your situation, I would start with downloading the official package which you can rebuild from source using standard debian tools. This will certify that you use the correct settings.

But to look into this we still need the output from 

  - `lsb_release -a`
  - `gcc --version`

mgrouch, 2020-10-06:
pi@raspberrypi:~/OCPN/opencpn/build $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
pi@raspberrypi:~/OCPN/opencpn/build $ gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0

mgrouch, 2020-10-06:
I want to look at touch gestures support. Adding listeners for two finger zoom, long touch as right click, etc. This API is available in wxWidgets 3.1 but not in 3.0.
So I need to compile with 3.1.x.

leamas, 2020-10-07:
Fair enough (and interesting!). 

That said, your situation is that you have odd errors not encountered on other Debian platforms after changing platform, wxWidgets version, using unstable master and build setup. If updating to latest master doesn't help I'd suggest that you first try move back to a working state using for example the packaged code or the release tag. If necessary, make the same build on x86_64.  When you have something working it might be easier to change piece after piece. 

BTW: I'm a bit sceptical about the wxWidgets roadmap. As late as in March it said it would be due "this spring". We'll see...

mgrouch, 2020-10-07:
I've tried with main brunch of wxWidgets both on raspberry pi (raspberry pi OS) and my PC (ubuntu) and OpenCPN crashes
at start-up consistently.

There was this warning on first attempt to start it under ubuntu
```
mikhail@mikhail-VirtualBox:~/opencpn/build$ /usr/local/bin/opencpn 
Gtk-Message: 04:18:51.056: GtkDialog mapped without a transient parent. This is discouraged.
Unhandled unknown exception; terminating the application.

```

leamas, 2020-10-07:
OK,. What happens if you build from the release tag v5.2.0?

mgrouch, 2020-10-07:
Tried that already. Exactly the same error happens with OpenCPN PPA-5.2.0.8 tag.

leamas, 2020-10-07:
Well, it's not the code -- then we would have tons of bug reports on this, but there isn't. So, something else:

  - Platform/compiler?
  - wxWidgets version?
  - Build options. 

Looking at the options, they _look_ sane. What happens if you link against the packaged wxWidgets library instead?

leamas, 2020-10-07:
And... what is your display manager? If it's wayland, please make a try using X11.

mgrouch, 2020-10-07:
> Well, it's not the code -- then we would have tons of bug reports on this, but there isn't. So, something else:
>
>     * Platform/compiler?
>
>     * wxWidgets version?
>
>     * Build options.
>
>
> Looking at [...]

There is no packaged wxWidgets 3.1. I had to build it see the script above.
3.0 is no-go cause it doesn't support gestures and this is what I'm building it for.

Display manager is X11, not wayland.

leamas, 2020-10-07:
But we still need to know whether this particular problem is bound to wxWidgets 3.1, so for that reason I suggest you make a try with linking against 3.0. It's IMHO hard tp proceed without this test.

mgrouch, 2020-10-07:
I've found the issue. I needed to build wxWidgets 3.1 with options from debian packaging of wxWidgets 3.0.

This script creates workable version:
```

#!/bin/bash

sudo apt-get -y install build-essential cmake gettext git-core libgtk-3-dev libgtk-3-0 libgl1-mesa-dev libglu1-mesa-dev \
 zlib1g-dev libjpeg-dev libpng-dev libtiff5-dev libsm-dev autotools-dev autoconf libexpat1-dev libxt-dev \
 libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libwebkit2gtk-4.0-dev libnotify-dev

git clone --recurse-submodules https://github.com/wxWidgets/wxWidgets.git
cd wxWidgets || exit 255

git checkout 301e5ec9fe3719fac8a2786e8675b19e2242183e

mkdir build-gtk
cd build-gtk || exit 255

../configure \
 --with-gtk=3 \
 --with-libpng=sys \
 --with-libjpeg=sys \
 --with-libtiff=sys \
 --with-regex=sys \
 --with-zlib=sys \
 --with-expat=sys \
 --enable-display \
 --enable-geometry \
 --enable-graphics_ctx \
 --enable-mediactrl \
 --enable-sound \
 --with-opengl \
 --enable-webview
make -sj5
sudo make install
sudo ldconfig
```

leamas, 2020-10-07:
Great work! Seems that this bug could be closed (?)

One thought crosses my mind: it would be nice if we could  have a common place where we could pickup a a pre-configured wxWidgets 3.1 package. You are certainly right in that it's a good idea to look into testing 3.1 during this cycle.

mgrouch, 2020-10-07:
Or you can just use my script and build it. 

The bug can be closed.

closed 2020-10-07 by mgrouch