OpenCPN/OpenCPN #2057: Add touch gestures hangling for two finger zoom and long press as a right click emulation

mgrouch, 2020-10-06:
Add touch gestures hangling for two finger zoom and long press as a right click emulation.
It is possible with wxWidgets 3.1.x 

For examples see:
grep TOUCH samples/event/*.cpp 
grep GESTURE samples/event/*.cpp
grep LONG_PRESS samples/event/*.cpp

in wxWidgets 3.1.4 code.

With this there will be no need for twofing and evdev-rce hacks.
It will work on linux (raspberry pi, etc) right after installing OpenCPN.

Thanks,
--MG

mgrouch, 2020-10-07:
Script to build wxWidgets 3.1.x for it and OpenCPN

For wxWidgets
```
sudo apt-get -y install build-essential cmake gettext git-core libgtk-3-dev libgtk-3-0 libgl1-mesa-dev libglu1-mesa-dev \
 zlib1g-dev libjpeg-dev libpng-dev libtiff5-dev libsm-dev autotools-dev autoconf libexpat1-dev libxt-dev \
 libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libwebkit2gtk-4.0-dev libnotify-dev

git clone --recurse-submodules https://github.com/wxWidgets/wxWidgets.git
cd wxWidgets || exit 255

git checkout 301e5ec9fe3719fac8a2786e8675b19e2242183e

mkdir build-gtk
cd build-gtk || exit 255

../configure \
 --with-gtk=3 \
 --with-libpng=sys \
 --with-libjpeg=sys \
 --with-libtiff=sys \
 --with-regex=sys \
 --with-zlib=sys \
 --with-expat=sys \
 --enable-display \
 --enable-geometry \
 --enable-graphics_ctx \
 --enable-mediactrl \
 --enable-sound \
 --with-opengl \
 --enable-webview
make -sj5
sudo make install
sudo ldconfig

```

For OpenCPN (see comments below for manual steps):

```
sudo apt-get -y install build-essential cmake gettext git-core gpsd gpsd-clients libgps-dev \
 libglu1-mesa-dev libgtk-3-dev libbz2-dev libtinyxml-dev libportaudio2 portaudio19-dev libcurl4-openssl-dev \
 libexpat1-dev libcairo2-dev libarchive-dev liblzma-dev libexif-dev libelf-dev libsqlite3-dev \
 bc bison flex libssl-dev python3 ddd htop

git clone -b PPA-5.2.0.8 https://github.com/opencpn/opencpn.git

cd opencpn && mkdir build

# in CMakeLists.txt replace references to 3.0 wxWidgets to 3.1 wxWidgets

cd build || exit 255

cmake -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_GSHHS=ON -DOCPN_USE_GL=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo \
 -DOCPN_USE_CRASHREPORT=ON -DOCPN_ENABLE_PORTAUDIO=OFF -DOCPN_ENABLE_SYSTEM_CMD_SOUND=ON -DOCPN_FORCE_GTK3=ON ..

# in ../src/glChartCanvas.cpp add #include <GL/glx.h>

make -sj5

sudo make install

```

mgrouch, 2020-10-08:
Ok. I'm now able to add touch event listeners and receive them in OpenCPN.
Now the issue is how to make it work with fewer code changes. I tried to post right click event from long touch event handler
in MyFrame class but it doesn't seem have any effect. I've tried to post wxContextMenuEvent to no effect.
Great that I receive touch events in the code but my total lack of experience with wxWidgets stops me there.

Any help?
Thanks

bdbcat, 2020-10-08:
Simplest thing to do is to post virtual mouse events into the ChartCanvas class.

Long touch->Right click emulation:  chcanv.cpp:8587
Pinch zoom:  Suggest following logic of "wheel zoom"...chcanv.cpp:8620

Good job so far.....
Good Luck going forward
Dave

mgrouch, 2020-10-08:
I've put in chcanv.cpp
into constructor

```
        if ( !EnableTouchEvents( wxTOUCH_PRESS_GESTURES ) )
        {
            wxLogError("Failed to enable touch events");

            // Still bind event handlers just in case they still work?
        }

        Bind(wxEVT_LONG_PRESS, &ChartCanvas::OnLongPress, this);
        Bind(wxEVT_TWO_FINGER_TAP, &ChartCanvas::OnTwoFingerTap, this);
        Bind(wxEVT_PRESS_AND_TAP, &ChartCanvas::OnPressAndTap, this);

```

and added methods

```
void ChartCanvas::sendRightClick( wxLongPressEvent& event )
{
    wxEvtHandler *evthp = GetEventHandler();
    {
        wxContextMenuEvent ev( wxEVT_NULL, 0, event.GetPosition() );
        wxPostEvent( this, ev );
    }
}

void ChartCanvas::OnLongPress( wxLongPressEvent& event )
{

    printf("%s\n", "long press!!!");
    if ( event.IsGestureEnd() )
    {
        sendRightClick( event );
    }
}

void ChartCanvas::OnTwoFingerTap( wxTwoFingerTapEvent& event )
{
}

void ChartCanvas::OnPressAndTap(wxPressAndTapEvent& event)
{
}

```

I see that long touch gets caught but context menu doesn't pop up

bdbcat, 2020-10-08:
For tight click, you need something like:
       CallPopupMenu(mx , my);
 See the code...

mgrouch, 2020-10-08:
Ok

with this
in chcanv.cpp
into constructor

```
        printf("%s\n", "registering events");
        if ( !EnableTouchEvents( wxTOUCH_PRESS_GESTURES ) )
        {
            printf("%s\n", "failed to enable touch events");

            // Still bind event handlers just in case they still work?
        }

        Bind(wxEVT_LONG_PRESS, &ChartCanvas::OnLongPress, this);
        Bind(wxEVT_TWO_FINGER_TAP, &ChartCanvas::OnTwoFingerTap, this);
        Bind(wxEVT_PRESS_AND_TAP, &ChartCanvas::OnPressAndTap, this);
```
and added methods

```
void ChartCanvas::OnLongPress( wxLongPressEvent& event )
{
    printf("%s\n", "long press!!!");
    CallPopupMenu(event.GetPosition().x, event.GetPosition().y);
}

void ChartCanvas::OnTwoFingerTap( wxTwoFingerTapEvent& event )
{
}

void ChartCanvas::OnPressAndTap(wxPressAndTapEvent& event)
{
}

```

I see popup menu, but it goes away as I let long touch go.

mgrouch, 2020-10-09:
GTK3 has a bug with touch menus. I have to rollback to using GTK2

See
https://gitlab.gnome.org/GNOME/gtk/-/issues/945

and merge request

https://gitlab.gnome.org/GNOME/gtk/-/merge_requests/1794

mgrouch, 2020-10-10:
Going back to GTK2 doesn't help. Touch events do not work at all. Handlers fail to register.
So instead I've went with upgrade of GTK3 to the latest tag.
Now menu works fine, touch two finger zoom work fine, long touch right click works (only on chart canvas so far).
But dragging a map with the finger stopped working.
Here is how I could build it:

Step1 Build GTK3 from debian source
```

sudo apt-get -y install build-essential cmake bc bison flex \
 at-spi2-core \
 dh-sequence-gir \
 fonts-cantarell \
 gnome-pkg-tools \
 gobject-introspection \
 gtk-doc-tools \
 libcolord-dev \
 libcups2-dev \
 libgdk-pixbuf2.0-dev \
 libgirepository1.0-dev \
 libjson-glib-dev \
 librest-dev \
 libxkbfile-dev \
 sassc \
 xvfb \
 libcairo2-doc \
 xsltproc

git clone -b "debian/2.40.0+dfsg-5" https://salsa.debian.org/gnome-team/gdk-pixbuf.git

cd gdk-pixbuf || exit 255

sudo apt-get -y install dh-exec meson
dpkg-buildpackage -rfakeroot -b -uc -us

sudo dpkg -i ../libgdk-pixbuf2.0-bin_2.40.0+dfsg-5_armhf.deb
sudo dpkg -i ../libgdk-pixbuf2.0-common_2.40.0+dfsg-5_all.deb
sudo dpkg -i ../libgdk-pixbuf2.0-0_2.40.0+dfsg-5_armhf.deb
sudo dpkg -i ../gir1.2-gdkpixbuf-2.0_2.40.0+dfsg-5_armhf.deb
sudo dpkg -i ../libgdk-pixbuf2.0-dev_2.40.0+dfsg-5_armhf.deb

cd ..

git clone -b "debian/3.24.23-2" https://salsa.debian.org/gnome-team/gtk3.git

cd gtk3 || exit 255

dpkg-buildpackage -rfakeroot -b -uc -us

sudo dpkg -i ../libgtk-3-common_3.24.23-2_all.deb
sudo dpkg -i ../libgtk-3-0_3.24.23-2_armhf.deb
sudo dpkg -i ../gir1.2-gtk-3.0_3.24.23-2_armhf.deb
sudo dpkg -i ../libgtk-3-dev_3.24.23-2_armhf.deb

```

Step2. Build wxWidgets 3.1.x from source against GTK3

```

# Build gtk3 before
# See ../gtk3-rpi/

sudo apt-get -y install build-essential cmake gettext git-core libgtk-3-dev libgl1-mesa-dev libglu1-mesa-dev \
 libgtk2.0-dev \
 zlib1g-dev libjpeg-dev libpng-dev libtiff5-dev libsm-dev autotools-dev autoconf libexpat1-dev libxt-dev \
 libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libwebkit2gtk-4.0-dev libnotify-dev

git clone --recurse-submodules https://github.com/wxWidgets/wxWidgets.git
cd wxWidgets || exit 255

#git checkout 301e5ec9fe3719fac8a2786e8675b19e2242183e

mkdir build-gtk3
cd build-gtk3 || exit 255

../configure \
 --with-gtk=3 \
 --with-libpng=sys \
 --with-libjpeg=sys \
 --with-libtiff=sys \
 --with-regex=sys \
 --with-zlib=sys \
 --with-expat=sys \
 --enable-display \
 --enable-geometry \
 --enable-graphics_ctx \
 --enable-mediactrl \
 --enable-sound \
 --with-opengl \
 --enable-webview
make -sj5
sudo make install
sudo ldconfig

```

Step3 Build OpenCPN against wxWidgets3.1.x (see manual steps in comments and the changes I applied in .h, .cpp code
in the posts above)

```

# Build wxWidgets before
# See ../wxWidgets-rpi/

sudo apt-get -y install build-essential cmake gettext git-core gpsd gpsd-clients libgps-dev \
 libglu1-mesa-dev libgtk-3-dev libbz2-dev libtinyxml-dev libportaudio2 portaudio19-dev libcurl4-openssl-dev \
 libexpat1-dev libcairo2-dev libarchive-dev liblzma-dev libexif-dev libelf-dev libsqlite3-dev \
 bc bison flex libssl-dev python3 ddd htop

git clone -b PPA-5.2.0.8 https://github.com/opencpn/opencpn.git

cd opencpn && mkdir build

# in CMakeLists.txt replace references to 3.0 wxWidgets to 3.1 wxWidgets

cd build || exit 255

cmake -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_GSHHS=ON -DOCPN_USE_GL=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo \
 -DOCPN_USE_CRASHREPORT=ON -DOCPN_ENABLE_PORTAUDIO=OFF -DOCPN_ENABLE_SYSTEM_CMD_SOUND=ON -DOCPN_FORCE_GTK3=ON ..

# in ../src/glChartCanvas.cpp add #include <GL/glx.h>

make -sj5

sudo make install

```

mgrouch, 2020-10-11:
Ok

with this in chcanv.cpp into constructor (and obvious include file additions)

```
    if ( !EnableTouchEvents( wxTOUCH_PRESS_GESTURES |  wxTOUCH_ZOOM_GESTURE | wxTOUCH_PAN_GESTURES ))
    {
        wxLogError("Failed to enable touch events");
        // Still bind event handlers just in case they still work?
    }

    Bind(wxEVT_LONG_PRESS, &ChartCanvas::OnLongPress, this);
    Bind(wxEVT_TWO_FINGER_TAP, &ChartCanvas::OnTwoFingerTap, this);
    Bind(wxEVT_PRESS_AND_TAP, &ChartCanvas::OnPressAndTap, this);
    Bind(wxEVT_GESTURE_ZOOM, &ChartCanvas::OnZoom, this);
    Bind(wxEVT_GESTURE_PAN, &ChartCanvas::OnPan, this);
```
and added methods

```
void ChartCanvas::OnPan(wxPanGestureEvent& event)
{
    printf("%s\n", "Pan gesture");
    wxEvtHandler *evthp = GetEventHandler();
    wxMouseEvent ev( wxEVT_MOTION );
    ev.m_x = event.GetPosition().x + event.GetDelta().x; //mouse_x;
    ev.m_y = event.GetPosition().y + event.GetDelta().y; //mouse_y;
    ev.m_leftDown = true;
    ::wxPostEvent( evthp, ev );
}

void ChartCanvas::OnZoom(wxZoomGestureEvent& event)
{
    if ( event.IsGestureStart() )
    {
        printf("%s\n", "Zoom gesture started");
    }

    if ( event.IsGestureEnd() )
    {
        if (event.GetZoomFactor() > 1.1 || event.GetZoomFactor() < 0.9) {
           printf("%s\n", "Zoom gesture ended");
           ZoomCanvas( event.GetZoomFactor(), true, false );
        }
    }
}

void ChartCanvas::OnLongPress( wxLongPressEvent& event )
{
    printf("%s\n", "long press!!!");
    wxEvtHandler *evthp = GetEventHandler();
    wxMouseEvent ev( wxEVT_RIGHT_DOWN );
    ev.m_x = event.GetPosition().x;
    ev.m_y = event.GetPosition().y;
    ::wxPostEvent( evthp, ev );
}

void ChartCanvas::OnTwoFingerTap( wxTwoFingerTapEvent& event )
{
}

void ChartCanvas::OnPressAndTap(wxPressAndTapEvent& event)
{
}

```

Now finger zoom work again. Right click with long touch works too.

There are couple of issues:

1. Context menu disappears after touch finger is released.
2. How to add long touch for right mouse click globally across whole OpenCPN app?

Thanks,
--MG

mgrouch, 2020-10-11:
OK. Finally. I've found solution on how to do right click emulation with long touch using external separate program.

So in previous patch replace EnableTouchEvents call with 

`if ( !EnableTouchEvents( wxTOUCH_ZOOM_GESTURE | wxTOUCH_PAN_GESTURES ))`

This (and the code changes above) will make OpenCPN to support zoom and pan with touch gestures.

Right click gesture will be supported on whole linux desktop using this solution:
```

wget https://raw.githubusercontent.com/bareboat-necessities/my-bareboat/master/right-click-emu/right-click-emu.sh
chmod +x right-click-emu.sh
./right-click-emu.sh
reboot

```

Thanks,
--MG

mgrouch, 2020-10-11:
Found one more issue.
Two finger zoom with OpenGL enabled most of the time is recognized as pan instead of zoom.
Weird...

mgrouch, 2020-10-11:
Hmm.
Somehow with openGL enabled  zoom action requires 3 fingers.
2 fingers do pan.

mgrouch, 2020-10-11:
With this and OpenGL enabled it looks better. However zoom requires 3 fingers.

```

void ChartCanvas::OnPan(wxPanGestureEvent& event)
{
    if ( event.IsGestureEnd() ) 
    {
        wxEvtHandler *evthp = GetEventHandler();
        wxMouseEvent ev( wxEVT_MOTION );
        ev.m_x = event.GetPosition().x + event.GetDelta().x; //mouse_x;
        ev.m_y = event.GetPosition().y + event.GetDelta().y; //mouse_y;
        ev.m_leftDown = true;
        ::wxPostEvent( evthp, ev );
    }
}
```

mgrouch, 2020-10-12:
I've put my commits in this branch 

https://github.com/bareboat-necessities/OpenCPN

So you can build it now like this:
```

# Build wxWidgets before
# See ../wxWidgets-rpi/

sudo apt-get -y install build-essential cmake gettext git-core gpsd gpsd-clients libgps-dev \
 libglu1-mesa-dev libgtk-3-dev libbz2-dev libtinyxml-dev libportaudio2 portaudio19-dev libcurl4-openssl-dev \
 libexpat1-dev libcairo2-dev libarchive-dev liblzma-dev libexif-dev libelf-dev libsqlite3-dev \
 bc bison flex libssl-dev python3 ddd htop

git clone https://github.com/bareboat-necessities/OpenCPN.git
cd OpenCPN && mkdir build

#git clone -b PPA-5.2.0.8 https://github.com/opencpn/opencpn.git
#cd opencpn && mkdir build

cd build || exit 255

cmake -DOCPN_BUNDLE_TCDATA=ON -DOCPN_BUNDLE_GSHHS=ON -DOCPN_USE_GL=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo \
 -DOCPN_USE_CRASHREPORT=ON -DOCPN_ENABLE_PORTAUDIO=OFF -DOCPN_ENABLE_SYSTEM_CMD_SOUND=ON -DOCPN_FORCE_GTK3=ON ..

make -sj5

sudo make install

```

mgrouch, 2020-11-16:
I've put it under circleCI build from here:

https://github.com/bareboat-necessities/OpenCPN

which gets deployed to https://cloudsmith.io/~bbn-projects/repos/bbn-repo/packages/
(Click "Set me up" on that page for instructions for your distro)

FredericGuilbault, 2020-12-12:
Is this have any chance to become a pull request or be manually implemented by opencpn team ?

bdbcat, 2020-12-12:
This function also relates to recent Android work.  So I expect we can see this in the master branch fairly soon.
I'll study your PR, and co-ordinate with Android implementation.
Thanks for the reminder

rgleason, 2021-04-23:
Off Topic, but related  https://github.com/OpenCPN/OpenCPN/issues/762
Is this still an issue, or is it fixed?

mgrouch, 2021-04-23:
2 fingers zoom still an issue

rgleason, 2021-04-23:
Mike, Thanks.   Have you noticed what Didier wrote about in https://github.com/OpenCPN/OpenCPN/issues/762
"Don't defer Leftup Event when dragging" ?   He provided some code for that too, but I wonder if you have noticed this problem while using the touch interface?

mgrouch, 2021-04-23:
I haven’t had many issues with dragging. I’ve seen some small jumps at the beginning of drag which were not too big of a deal. Might that’s what that fix trying to address.

tbultel, 2021-08-12:
See https://github.com/OpenCPN/OpenCPN/pull/2333 that fixes all the mentioned issues

mgrouch, 2021-09-11:
FYI. Even with current released version 5.2.4 I've managed to make it work
without changing it with external program twofing and udev rules.

You can check how it works in BBN OS 2021-09-10

https://bareboat-necessities.github.io/my-bareboat/bareboat-os.html

Thanks

tbultel, 2021-09-12:
Mikhail,
I think you should have a look on master branch of OpenCPN.
My changed have been integrated, without the need for an external program,
simplifying the integration. It needs wxWidgets > 3.1.1, though.
Thanks !

rgleason, 2021-09-12:
Mgrouch can this be closed now?

mgrouch, 2021-09-12:
Closing has to wait till wxWidgets comes out as 3.2.x

3.1.x versions are not going to be in Debian

leamas, 2021-09-25:
See also #2396

mgrouch, 2022-07-16:
One of the reasons to use WxWidgets 3.2.x

bdbcat, 2024-06-15:
Mgrouch...
Reviving this for discussion.
In OCPN 5.9.1, I have implemented the EnableTouchEvents() in chcanv.cpp.
But for some reason, using RPi4/bookworm, I do not ever get any events.
Any ideas?

```
#ifdef HAVE_WX_GESTURE_EVENTS
  if (!EnableTouchEvents(wxTOUCH_ZOOM_GESTURE |
                         wxTOUCH_PRESS_GESTURES)) {
    wxLogError("Failed to enable touch events");
  }

  Bind(wxEVT_GESTURE_ZOOM, &ChartCanvas::OnZoom, this);

  Bind(wxEVT_LONG_PRESS, &ChartCanvas::OnLongPress, this);

```
Dave

mgrouch, 2024-06-15:
The is also touch PAN event. 
and there was PR more recent than my original patches. 

https://github.com/OpenCPN/OpenCPN/pull/2333

mgrouch, 2024-06-15:
Might be worth trying enabling all touch events and adding handles only for some. 

Also long touch for right click should be enabled somewhere else. Otherwise it works only on chart canvas. And it’s needed in other places too. 
select all in chart downloader would be one example.

bdbcat, 2024-06-15:
I get no events, not even long touch.  If I can get  that one simple touch event to work, the rest come easily.
Funny thing is that these touch events apparently work on flatpak build, as reported on CF.
I begin to suspect the wxWidgets build has error.
Very strange.

mgrouch, 2024-06-16:
On Wayland or on X11?

mgrouch, 2024-06-16:
Also we might watch out for an impact by this:

https://github.com/wxWidgets/wxWidgets/issues/24595

bdbcat, 2024-06-16:
I am testing on X11.
Now rebuilding wxWidgets to current github master, to test.
More info as I find it.

mgrouch, 2024-06-16:
What’s the version of the wxWidgets which upholds having issue with?
Debian or Ubuntu packaged?

leamas, 2024-06-16:
Debian, wxw 3.2.2

bdbcat, 2024-06-17:
New hardware.
Running locally built wx3.2.2.1, I can now get LongTouch events in OCPN.  This is some progress.
However, I have not succeeded in getting two-finger zoom events.  The events are enabled.they just never come.

Same with wx3.2.4
Clearly something is missing here...

bdbcat, 2024-06-18:
More debug, testing new touchscreen monitor and "official" rPI 7" display.
Neither setup will produce multi-touch events.

Built the "event" sample from wxWidgets distribution.  This has a test module that catches all wxGesture events, and reports as debug output.  Only gestures recognized by this sample are "long-touch" and "pan", both of which are single point gestures.

Built gtk3 locally, adding printf debug looking for gesture events.  No multi-point gestures seen.

For good measure, tested flatpak OCPN beta.  And pinch zoom gestures are handled correctly, actually works very nicely.

So, some tentative conclusions:
1. The default input stack from X11->kernel->gtk->wxWidgets does not handle multi-touch.  Feels like some sort of config error, but I am unable to see where.
2. Something about the flatpak gtk library is different, enabling the multi-touch support by default.
3. The core gesture code in OCPN is functional, else flatpak could not possibly do pinch-zoom on OCPN canvas.

Looking for any/all input on this.  I'm running out of ideas.

bdbcat, 2024-06-18:
mgrouch...
Have you tried the wxWidgets sample called "event" on your current hardware?

mgrouch, 2024-06-18:
Does two finger zoom work in Chromium browser on your system? Worth checking to eliminate lower lever set up issues

bdbcat, 2024-06-18:
Yes, it does.
I wonder if Chromium is talking directly to the hardware, not depending on the linux kernel stack?

mgrouch, 2024-06-18:
> Yes, it does. I wonder if Chromium is talking directly to the hardware, not depending on the linux kernel stack?

No it doesn’t talk directly to hardware. It uses input event stack libraries.

mgrouch, 2024-06-18:
It would be interesting to compare output of

ldd opencpn 

of regular build 
with flatpak build. 

Which libraries versions would be different?

bdbcat, 2024-06-18:
Checked ldd.
There are huge differences in the system libraries.  Almost every one is different.
Will study.
Any hints on libs critical to multitouch?

mgrouch, 2024-06-18:
I checked horse example and two finger zoom didn’t work in it. WxWidgets 3.2.2 example. 

Trying to build WxWidgets 3.2.5

bdbcat, 2024-06-18:
OK, thanks, same result as me with the horse, wx3.2.4

leamas, 2024-06-18:
> Trying to build WxWidgets 3.2.5

There are wx3.2.5 packages available at http://mumin.crabdance.com/wx/3.25/. This is just the source package from Debian Sid, backported to bookwoem

bdbcat, 2024-06-18:
We are working to test the wxWidgets samples, included when built locally, to remove any OCPN questions.
There is a sample called "event", which tracks and reports gesture events as seen by wx.  If this sample demo fails on multi-touch, O will also fail.

leamas, 2024-06-18:
Indeed. It should be perfectly possible to build this sample against the 3.2.5 packages to test also that combination without having t make yet another painful arm wxw rebuild.

leamas, 2024-06-18:
We might want to compare the compilation flags used when compiling wxWidgets in Flatpak vs in Debian.

Flatpak, from the manifest:
```
    config-opts:
          - --with-gtk=3
          - --with-opengl
          - --with-sdl
          - --with-libmspack
          - --enable-intl
          - --disable-rpath
          - --enable-ipv6
```

Debian is  governed by d/rules which reads:

```
override_dh_auto_configure: 
        dh_auto_configure -- \ 
                GREP=/bin/grep \
                --disable-glcanvasegl \
                --enable-repro-build
```

when compiling, this is expanded to: 
```
./configure --build=aarch64-linux-gnu --prefix=/usr --includedir=\${prefix}/include --mandir=\${prefix}/share/man --infodir=\${prefix}/share/info --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir=\${prefix}/lib/aarch64-linux-gnu --runstatedir=/run --disable-maintainer-mode --disable-dependency-tracking GREP=/bin/grep --disable-glcanvasegl --enable-repro-build
```

perhaps `--disable-glcanvasegl`  should be checked?

mgrouch, 2024-06-18:
Might be needs

 --with-sdl

mgrouch, 2024-06-18:
Look. .deb options make this:

````
Configured wxWidgets 3.2.5 for `aarch64-unknown-linux-gnu'

  Which GUI toolkit should wxWidgets use?                 GTK+ 3 with support for GTK+ printing libnotify
  Should wxWidgets be compiled into single library?       no
  Should wxWidgets be linked as a shared library?         yes
  Should wxWidgets support Unicode?                       yes (using wchar_t)
  What level of wxWidgets compatibility should be enabled?
                                       wxWidgets 2.8      no
                                       wxWidgets 3.0      yes
  Which libraries should wxWidgets use?
                                       STL                no
                                       jpeg               sys
                                       png                sys
                                       regex              sys
                                       tiff               sys
                                       lzma               sys
                                       zlib               sys
                                       expat              sys
                                       libmspack          no
                                       sdl                no
````

leamas, 2024-06-18:
Too late here, need some sleep. But next step should probably be to compare the full build logs for Flatpak and Debian, the latter from an isolated build using pbuilder or a docker container.

I assume the the test program fails in the same way on wx3.2.5, right?

mgrouch, 2024-06-18:
Also does flatpak version enable 3.0 compatibility?

````
  What level of wxWidgets compatibility should be enabled?
                                       wxWidgets 2.8      no
                                       wxWidgets 3.0      yes
````

mgrouch, 2024-06-18:
Does the horse example work with two finger zoom on flatpak version?

mgrouch, 2024-06-19:
I feel you just always need to compile wxWidgets and link statically to it. With WxWidgets 3.2 API without 3.0 compatibility. On all platforms. 

Relying on what each platform provides as shared libraries is too much maintenance.

nohal, 2024-06-19:
That is exactly what can't and should not be done.

mgrouch, 2024-06-19:
> That is exactly what can't and should not be done.

Why?

nohal, 2024-06-19:
a) Your brilliant idea immediately makes the plugins impossible
b) OpenCPN will never be accepted in any Linux distribution, ever

[60 more replies]