OpenCPN/OpenCPN #2060: pop up sometimes comes garbled clicking on right bottom hamburger icon

mgrouch, 2020-10-12:
(opencpn:1838): Gtk-CRITICAL **: 10:29:15.804: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkCheckButton

while clicking right bottom hamburger icon on chart canvas tools

and pop up after that sometimes comes garbled

Thanks

mgrouch, 2020-10-12:
raspberry pi.
Error present on both 32 bit and 64 OS v ersions

leamas, 2020-10-12:
There are a vast amount of this kind of assertions, most of which problems in wxWidgets. In one case I have been able to upstream the problem, and a fix is now committed in wxW.  However, it's a complicated process  starting with  isolating the problem which is not that easy.

mgrouch, 2020-10-12:
This seems to be only one place left to fix it

mgrouch, 2020-11-06:
I can confirm that with latest GTK and WxWidgets it happens every time

mgrouch, 2020-11-16:
It breaks with GTK3. No such issue with GTK2

rgleason, 2021-04-22:
https://github.com/OpenCPN/OpenCPN/issues/2168

mgrouch, 2021-04-22:
There are 2 distinct issues. 
One is for toolbar with plugin icons. 
Another is for right bottom toolbar with chart displayed objects selectors.

leamas, 2024-04-28:
@mgrouch : there has been substantial updates of the relevant code since this was filed three years ago.

 Is this still an issue or can it be closed?

mgrouch, 2024-04-28:
Can be closed

closed 2024-04-28 by mgrouch