OpenCPN/OpenCPN #2061: mingw plugins are deemed incompatible.

leamas, 2020-10-12:
As heading says. A mingw build has an empty list of plugins after loading the catalog, and attempts to import a plugin   fails with a "Incompatible plugin" message.

leamas, 2020-10-12:
Seems like a plugin problem. OpenCPN expects `<target-arch>` to be _mingw-x86_64,_ but plugins generally just uses _mingw_. Patching a test plugin solves the problem.

leamas, 2020-10-12:
Patching the catalog shows all available plugins: ` sed -i '/target/s/mingw/mingw-x86_64/' /c/ProgramData/opencpn/ocpn-plugins.xml`

There are issues when installing plugin: It is installed what looks loooks like two items. After a restart, the plugin  settings looks sane and seems to work, but the uninstall/update buttons are gone.  

Leaving this for now, I don't really understand all the effects of patching the catalog this way without actually fixing the plugins. OTOH, importing a fixed plugin has the same problem.

[image]

leamas, 2020-10-12:
Attaching debug log:

[opencpn.log.gz](https://github.com/OpenCPN/OpenCPN/files/5367107/opencpn.log.gz)

leamas, 2020-10-13:
The installation issues are present in v5.2.0 but fixed in current master at a44756712.

This means that remaining issue is plugins using mingw  instead of mingw-x86_64 as `<target>`.

leamas, 2020-10-13:
Downstream plugin bugs:
  - https://github.com/opencpn-radar-pi/radar_pi/issues/150
  - https://github.com/jongough/testplugin_pi/issues/109
  - https://github.com/Rasbats/shipdriver_pi/issues/55
  - https://github.com/bdbcat/oesenc_pi/issues/102

rgleason, 2020-10-13:
Thanks Leamas,
VDR mingw has this in metadata

```
target: mingw
target-version: 10
target-arch:  x86_64
```

Is this correct?

rgleason, 2020-10-13:
So how do I get this changed?

`This means that remaining issue is plugins using mingw instead of mingw-x86_64 as <target>.`

leamas, 2020-10-13:
No idea, this is basically frontend2 stuff. I have filed a bug against testplugin, see above.

rgleason, 2020-10-13:
BTW  vdr_pi is frontend1  but I think I have found it.

I think LINE 13 should be 
    set(PKG_TARGET "mingw-x86_64") 

**cmake/PluginSetup.cmake**

```
if (OCPN_FLATPAK)
    set(PKG_TARGET "flatpak-x86_64")
    set(PKG_TARGET_VERSION "18.08")    # As of flatpak/*yaml
elseif (MINGW)
    set(PKG_TARGET "mingw")     <------ LINE 13
    if (CMAKE_SYSTEM_VERSION)
        set(PKG_TARGET_VERSION ${CMAKE_SYSTEM_VERSION})
    else ()
	set(PKG_TARGET_VERSION 10)
    endif ()
elseif (MSVC)
    set(PKG_TARGET "msvc")
    if (CMAKE_SYSTEM_VERSION)
        set(PKG_TARGET_VERSION ${CMAKE_SYSTEM_VERSION})
    elseif (CMAKE_VS_WINDOWS_TARGET_PLATFORM_VERSION)
        set(PKG_TARGET_VERSION ${CMAKE_VS_WINDOWS_TARGET_PLATFORM_VERSION})
    else ()
        set(PKG_TARGET_VERSION 10)
    endif ()
 
```

bdbcat, 2020-10-13:
Rick...
Looks right to me.
Give it a test run.

rgleason, 2020-10-13:
Thanks Dave,
Just so everyone understands, this change for Mingw target, will be made in the normal course of updating (Sean's & other) plugins for other reasons, howver my branch for Jon's Testplugin will be updated immediately.

rgleason, 2020-10-14:
This is unacceptable
https://github.com/OpenCPN/plugins/pull/258/checks?check_run_id=1250995277

bdbcat, 2020-10-14:
Rick...
A little more detail, please.  What is the problem?

leamas, 2020-10-14:
@bdbcat @rgleason: I see  the problem, should have thought of it: The xsd file does not allow` <target>` to be correct value _mingw-x86_64_, and it allows the bad value _mingw_. 

Will make a PR. During a transition window it might make sense to allow both values so we don't block things on the currently low-priority mingw build, right?

leamas, 2020-10-14:
@rgleason: Good catch!

rgleason, 2020-10-14:
Thanks. Whole we are at this aren't there some other changes needed for MacOS?

If possible I would like to make those changes too as I update PI. Are there some good examples for this change somewhere that I can try to follow?

leamas, 2020-10-14:
>  While we are at this aren't there some other changes needed for MacOS?

Not that I'm aware of.

rgleason, 2020-10-16:
XCode 11.0

jongough, 2020-11-24:
Why is <target> & <target-arch> being compounded? <target> is mingw, <target-arch> is x86_64, so all the information needed is available. The XML says that this package is for mingw running on 64bit architecture. Why include the 64bit twice, once in the name and once in the architecture. This appears to be a misuse of xml and making unneeded complexity.

leamas, 2020-11-27:
This has already been discussed, for exampled in #2002 and #2003. I agree that here is some useless redundancy.  As a consequence, I think neither of us are happy about the situation.

However, the decision is made and we have moved on. Given that almost all plugins have accepted the general contract my opinion is that it's better to fix the few  who doesn't rather than to try to fix the format.

Of course, it might be possible to make a PR to clean things up  (either drop `<arch>` or not having a full ABI specification in `<target>`). However, since it must be backwards compatible it's non-trivial, and at least I is not motivated to do such a thing.

leamas, 2020-12-28:
Things are sorted out, the xsd file is updated. Whatever problems left it is about specific plugins. Closing.

closed 2020-12-28 by leamas