OpenCPN/OpenCPN #2062: Travis CI builds not starting on pushes to master branch

bdbcat, 2020-10-13:
Anyone know why?

leamas, 2020-10-13:
It does build for me, so something with your user.

Also, here is this travis-ci.org vs travis-ci.com mess. Seems that they are moving from the .org domain to .com, and that the .com offer is slightly better.  I have moved to travis-ci.com

bdbcat, 2020-10-13:
leamas...
I assume you mean that your forks of OpenCPN/OpenCPN (master branch) build OK on your travis account?
But we seem to not have a travis account for "OpenCPN",  so the canonical master branch never gets built by travis, as it does for CircleCI and Appveyor.  Am I missing something?

bdbcat, 2020-10-13:
leamas...
Never mind my blathering.  I see that Travis builds for OpenCPN/OpenCPN are happening for master commits, and PRs.  I was looking in the wrong place...

closed 2020-10-14 by bdbcat