OpenCPN/OpenCPN #2064: Allow one OpenCPN instance to exchange date with another

mgrouch, 2020-10-14:
I think I do not like to do route planning standing next to binnacle and trying to program waypoints
on a screen hard mounted in sailpod.
I would rather put waypoints on Android tablet in laying on a bunk abut then 
I would like to push them into OpenCPN instance running on raspberry pi which serves cockpit.

It would be nice to have a way to do that.

Thanks,
--MG

rgleason, 2021-04-10:
There are some related requests in Tracker.  Perhaps you can find them and add your vote and thoughts?
It would be good because this is something that is being discussed.

mgrouch, 2022-09-21:
See discussion: https://github.com/OpenCPN/OpenCPN/discussions/2762

nohal, 2023-04-30:
Implemented.

closed 2023-04-30 by nohal