OpenCPN/OpenCPN #2066: Docker Account for MacOS

rgleason, 2020-10-16:
The info is in the link: : https://circleci.com/docs/2.0/private-images

Basically:
- Set up a docker account (normal free registration).
- Add the password to the environment, similar to CLOUDSMITH_API_KEY
- Add username and name of environment variable to .circleci/config

As I said, the details are in the link. As I understand it, docker.io will limit all anonymous downloads, be the images public or not.

rgleason, 2020-10-16:
This will also affect plugins building MacOS.
According the Alec you need to have at least a  Docker Username and Password.
I started by signing up for a free  "Docker Hub" account (cloud type) https://hub.docker.com/
I was then promped to download the Mac or Windows Desktop (CLI) (for Windows it is 407mb.)
I did not complete the installation as I am not sure I need it for plugins.

This depends on whether we have to create our own MacOS Container and upload it to our Docker repository, 
or if we can use a  MacOS Image using XCode 11.0  (as Alec has suggested) that is already created and stored in a public Docker repository  (either Circleci or other source)..

rgleason, 2020-10-16:
Opensource docker images are retained 6 months.   "Free accounts may retain inactive images for up to 6 months"
docker/roadmap#152

Docker allows organizations https://hub.docker.com/orgs   but there is a limit to the number of users (3).
however OpenSource Projects can also apply https://docs.google.com/forms/d/e/1FAIpQLSd11DjfgeaKSRdqp6Br85MzQQuM4igt-xmn6rmzA2StFGLndQ/viewform

rgleason, 2020-10-16:
For plugins, use Shipdriver as an example  https://github.com/Rasbats/shipdriver_pi

leamas, 2020-10-17:
We should not create any docker for building macos, there is absolutely no need. It is readily available in many versions including 11.0.

To be honest, I don't follow the reasoning here. The starting point seems to be the message from circleci.  But that does not involve the need of any new images, just that we authenticate when pulling our already available images from docker, see my last commits on #2050.

rgleason, 2020-10-17:
I see this to add docker authentication
https://github.com/leamas/OpenCPN/commit/ad35611f8b8bf4d14730d3c883c127dce90d4857

and in CircleCI  environment (similar to CLOUDSMITH_API_KEY) we add
DOCKER_USER    'user value'
DOCKER_PW   'pw value'

and in circleci/config.yml we have to add   "- run: ci/docker-auth.sh"  as you show.

rgleason, 2020-12-25:
I think this is settled without everyone having to have a docker account.

closed 2020-12-25 by rgleason