OpenCPN/OpenCPN #2067: Use git submodule feature to include all 3rd party plugins into OpenCPN build

mgrouch, 2020-10-18:
Use git submodule feature to include all 3rd party plugins into OpenCPN build

nohal, 2020-10-18:
Why?

mgrouch, 2020-10-18:
So one build in CI 
builds OpenCPN with all plugins available for all platforms. 
Simplifies maintenance and reduces release cycles efforts

nohal, 2020-10-18:
I probably miss the point here. Do you really think moving the QA and release coordination from the actual authors of the plugins over to the OpenCPN team is a good idea? I certainly don't.

mgrouch, 2020-10-18:
It’s not moving it from them to OpenCPN team. It will make binaries available with OpenCPN release. 
Submodules Can refer to tagged released versions of plugins. It’s more about making binaries available at same time and at same place.

https://gist.github.com/gitaarik/8735255

nohal, 2020-10-18:
I of course do know what git submodules are.
How do you imagine the OpenCPN core team knows in what state my plugin code is and what exactly I as it's author want to include?
And as I have both writen several of these thirdpaty plugins myself and spent a LOT of time making others at least somehow usable, I think I know what I talk about when I'm saying it will not work.

mgrouch, 2020-10-18:
Whatever developer tags as release. 
They do not necessarily have CI in place to build on all platforms. 
This will allow them to be parts of your CI build.

mgrouch, 2020-10-18:
The reason you spending so much time because you can’t test it in time cause they need to be built somewhere separately. This way they are always built and can be tested right away

nohal, 2020-10-18:
> The reason you spending so much time because you can’t test it in time cause they need to be built somewhere separately. This way they are always built and can be tested right away

Believe me, this is not the reason.

mgrouch, 2020-10-18:
I think it’s a win win. 
You basically giving plugin developers your build infrastructure for all platforms. They wouldn’t need their own CI builds. You are getting plug-in built and you see immediately if the changes in OpenCPN code break other plugin builds. When you release you have all plugins built. Someone cloning OpenCPN gets source code with all plugins. 
I see this approach working in other projects and I have no reason to 
believe it’s not going to work in OpenCPN. It forces both plugin developers and OpenCPN developers play nicer together. Forces some more conventions, speeds up development cycles.

nohal, 2020-10-18:
All it really does is it adds work to the core team.

mgrouch, 2020-10-18:
They would have an option to build without sub modules. 
Regarding extra work for core team. 
I disagree. There was work on safe mode to disable faulty plugins,  there was work on plugin manager. If the plugins were part of the same continuous build it might have not been necessary. Having them as git submodules leaves developers with ability to have own release cycles. 
But the build infrastructure is one and unified. It actually would free core developers from answering lot of questions and coordination efforts. Once some effort is invested into this common build it’s going to a big timesaver and effort saver

nohal, 2020-10-18:
Based on the experience I have you wasty overestimate the theoretical benefits and underestimate the issues this would bring.
Implementing CI for an OpenCPN plugin is a matter of minutes (Is there still realy some plugin that does not have it and is not a complete abandonware?).
Unless you disallow independent plugin releases completely, you will never be able to benefit from the effort savings you try to outline because there simply will still be faulty plugins from the thirdparties out there.
And doing it you effectively kill the plugins alltogether and make all the effort invested in them useless as the actual resulting workflow will be exactly the same as if all the functionality was directly implemented in the core application.
Worse, you will force the core team to invest efforts in the plugins as, obviously, they should be in usable shape to make any OpenCPN release feasible.

mgrouch, 2020-10-18:
Faulty plugins will be easier detected. 

OpenCPN will still be able to release 
with the flag turning off submodules. 

It will bring the issues, but the issues will be same which would had to be solved anyway. So it’s same effort for the team. 

It will also help to remediate repeating issues with every release. 
Not sure why are you arguing against continuous build of everything 
especially with having an option to build it as before (just conditional statement somewhere in the scripts to turn off sub modules build).

nohal, 2020-10-18:
I "argue" against it because I clearly do not think the benefits you try to outline really are benefits and are worth any effort. And because it has been litrally years since I have seen any issue that would have been detected by doing this type of a build.
3rd party plugins are 3rd party plugins for a reason.

mgrouch, 2020-10-18:
Btw 

Do you have OpenCPN continuous builds of OpenCPN for
1. Windows
2. Non Debian based Linux flavors
3. Arm 32 and 64 bit Linux 
4. OpenCPN build against trunk versions of Gtk and wxWidgets?
5. Against all plugin trunks 

The more continuous build are there the easier to find breaks and easier to test. There are many platforms and combinations OpenCPN is used on. There need to be CI builds for all these combos.

nohal, 2020-10-18:
All we consider we need, yes. Some you outline, like for example builds against trunks of all the 3rd party plugins would be a nonsense as far as I'm concerned.

mgrouch, 2020-10-18:
Why would plug-ins be discriminated against as far as continuous build is concerned? They are just like 3rd party libraries which you want to test against. You also want to discover compile time errors as soon as possible. And be able to run and test the executables produced by good builds. I simply do not understand the reasoning for objections to that.

nohal, 2020-10-18:
3rd party plugins are responsibility of the 3rd party developing and releasing them, not of core OpenCPN.
I suppose you know how OpenCPN's plugin system actually works - the 3rd party plugins do include a copy of https://github.com/OpenCPN/OpenCPN/blob/master/include/ocpn_plugin.h in their source tree, implement a specific version of the API that never changes and (except on Windows), do not even link against OpenCPN core binary build preducts, so the chance that the kind of errors you talk about will be discovered by the CI builds is very very close to not exist.

leamas, 2020-10-18:
The CI setup currently builds 
   - windows 10 (appveyor), mingw32
   - flatpak 18.08 and ubuntu trusty .. focal
   - MacOS, Xcode 10.3
   - raspbian/arm32

And that's we what we currently manage, it's a question of resources... We do no intend to support or build for other Linux distros, flatpak covers this niche. Vanilla Debian/arm64 seems like the next, probable platform, time permitting.

leamas, 2020-10-18:
@nohal: I concur.

Also, as for the overall coordination between plugins and main OpenCPN the plugins project at https://github.com/OpenCPN/plugins plays an important role, basically vetting the current plugin set including version when it comes to the new installer. 

This project reflects the nature of the plugins as downstream, independent projects with a rather loose dependency using the plugin API.

mgrouch, 2020-10-18:
You have dependencies via shared libraries. If you upgrade OpenCPN to wxWidgets 3.2 all plugins would have to be rebuilt. That header file is not only how you coupled with plugins. 
I agree that it’s plugin developer responsibility to maintain its code and releases (code releases) but they might not be able to do binaries for all cases. They better be on your CI continuous build. Plus not just plugin developer should test. Anybody should be able to pick a build result and test it.

nohal, 2020-10-18:
That is why the plugin developer should implament CI for his plugin, yes.
When we switch the framework major version, we by definition implicitly break everything, in a planned and controlled way, we do not need CI to tell us that everything is broken in such a case.
Also when some platform breaks it's ABI compatibility between releases, everything needs to be rebuilt. Still it is not the reason to transfer this workload from 3rd  party plugins to the core OpenCPN.

leamas, 2020-10-18:
If not already so, I'm convinced that the new installer will be the major way to install plugins -- it's just so dead simple for users.

And in the current workflow for our plugins, we have CI solutions for our platforms in place. These are  reasonably simple to copy-paste, and the plugins in the catalog are available on all supported platforms. So no, this is not a problem.

leamas, 2020-10-18:
@mgrouch : Bottom line: Why is this importan for you? What are you trying to achieve?

We have been looking at the browsers as a reference. They are the same in this respect, and they certainly don't build the plugins together with the browser. OpenCPN basically works the same way...

mgrouch, 2020-10-18:
You are not transferring work to core developers. You are transferring it to your CI process. 
You do not need to break anything after you upgrade to new upstream library because your CI will automatically build plugins with that library too. Why every plugin developer would have to implement their own CI. It’s unnecessary. They should be on OpenCPN CI. 

I’ve built OpenCPN with latest Gtk3 and latest wxWidgets. How I test it with all plugins? Are you saying I need to build each plugin one by one? When they could have been already built with OpenCPN as submodules. 

I really can’t understand why OpenCPN CI can’t build plugins.

mgrouch, 2020-10-18:
How can I test my build I complied with latest Gtk3 and latest wxWidgets with all the plugins?

nohal, 2020-10-18:
If you want to build and test OpenCPN and all the plugins with a build configuration not supported by the upstream, it is your responsability to get such a build for all the components (and dependencies, if required, as it is in the particular case you describe). And nothing at all prevents you from using git submodules for that (I actually do it like this all the time locally, for the plugins I am contributing to), CI platform of your liking or whatever else.

Every developer of an independent component, like for example a 3rd party plugin for OpenCPN, should implement the CI the way he considers viable for his product because it is his product, not someone else's. And everybody else is of couse free to contribute such workflow in an open source environment as ours.

leamas, 2020-10-18:
> I really can’t understand why OpenCPN CI can’t build plugins.

Because upstream does not depend on downstreams, it's plugins who tests against latest version of opencpn.

> How can I test my build I complied with latest Gtk3 and latest wxWidgets with all the plugins?

Ah.... *This* is the problem. It would have been easier if you had that as the subnject rather than the git submodules stuff. 

With reasonable resources you basically cannot do this for all plugins. That said, to fork a limited set, set up the CI and add yet another builder shouldn't that much work. One you have the plugins built you can install them using the Import Plugin button/feature.

mgrouch, 2020-10-18:
Plugin and OpenCPN share same address space. Should be built against same libraries. There should have been a way to build them in one build and make it easier to test. 
And moreover requiring a developer of some small plugin to set up whole separate CI environment and distribution channels is not nice. 
Build with submodules flag would really resolve a lot of issues. This what this request is about. 
There is a page for plugins with link and so on. To build them one by one is a big effort for me. Could have been easily avoided.

mgrouch, 2020-10-18:
Plugins and OpenCPN is just one stream.

leamas, 2020-10-18:
Rather than looking into that page, I suggest the you look into the plugin catalog at the plugins project. That's the source for plugins with working CI setup (some of which forked from the original plugin, don't blame me).

Let's drop the submodules discussion. There are no more arguments, we have shared our positions.  As you see, it just won't happen and we need to find other solutions.

leamas, 2020-10-18:
To be constructive: Give me a list of 3-5 plugins and I'll try to locate the github  repo with working CI setup for you.

mgrouch, 2020-10-18:
Let’s say hypothetically I could
manage in my fork of OpenCPN to add all plugins as sub modules.
And have a flag to build all and only OpenCPN with another flag value would you accept and merge this change into OpenCPN mainline?

leamas, 2020-10-18:
Not my decision. But I would strongly advice against doing so. 

As I said: we have shared our positions on this, there are no more arguments. Submodules won't happen and we need to find other solutions. I'm heading for bed, so if you want to locate the repos it either now or tomorrow.

nohal, 2020-10-18:
I really don't think having the plugin downstreams as submodules in the core repository is a good idea. It simply generates maintenance costs that are completely unnecessary.

mgrouch, 2020-10-18:
Time is the cost. I could  have been testing will all plugins on latest gtk3 latest wxWidgets. 
There is going to be major lap in plugins releases after you lets say start building with wxWidgets 3.2. And plugins still be unbuilt. Totally avoidable and preventable.

leamas, 2020-10-18:
Well, these *are* the same arguments, again. I will not take part in any more discussion on the submodules subject, but I'm more than willing to help you arrange some testing with my limited ability.

rgleason, 2020-10-18:
Also not all plugins are in the same state, ready for PI Manager and subsequent small changes. This is an impossibility.

leamas, 2020-10-19:
See also #2047

CarCode, 2020-10-19:
The different attemps to make plugins better shows that the design of plugins was not well-thought-out when it has been established years ago. The best way now is to publish all the good and usable plugins together with the main program and not only the 4 plugins as right now. Such a decision might be also a way to stimulate 3rd.party programmers for enhancing their plugin not already part of the main to get the step into the main program.

leamas, 2020-10-19:
> The best way now is to publish all the good and usable plugins together with the main program and not only the 4 plugins as right now.

But isn't this what we currently do in in the plugin installer? When a plugin is published here (i. e., in the plugin catalog) it's sort of part of, or published, by the main project. From a user perspective this means here is a well-defined set o about 30 "official" plugins, all of which easily installed with a single button click.

transmitterdan, 2020-10-20:
Other projects such as wxWidgets use git-submodules.  But it's primary purpose is to maintain consistency with dependencies of the core project.  OpenCPN does not depend on plugins (except the 4 core plugins), rather the other way around, plugins depend on OpenCPN.  I suppose a plugin could consider OpenCPN as a submodule but in my view not the other way around.

mgrouch, 2020-10-20:
OopenCPN depends on them. It executes them in same address space and depends of them being compiled against same libraries.

bdbcat, 2020-10-20:
This is simply not workable.
Take a simple example, using your latest OCPN branch linking against your private wxWidgets v3.1.4 build.  Add a bunch of known good managed plugins, using git submodules added to OCPN Cmake scripts.. 
I predict that virtually every linux/debian CI build will fail, simply because wx3.1.4 is not available in any of the docker images used to build those plugins.  If any managed to build, they would likely not be loadable in OCPN, rejected as incompatible due to strict library ABI incompatibilities.
So, what have we got?  A broken CI build.  And no compatible plugins.
So what have we learned?  We have a bunch of plugins that cannot currently be CI-built against wx3.1.4, or won't load  if they do build.  Who will fix them?  
Answer:  It is up to the plugin devs or packagers to update their work as required, staying in sync with API/ABI changes in OCPN core.  This is not just philosophical pablum.  It is actually a critical element of the distributed development model we are using under present resource constraints.  The OCPN core devs simply cannot come to "own" the entire plugin universe in order to assure a clean CI build.  Just not practical...

I chose your wx3.1.4 example on purpose.  We will come back to this for more discussion, I'm sure. But let's be clear that today we have no CI workflow that supports custom wxWidgets builds on debian based distros.  None.  So if a new feature (e.g. integrated touch control) requires new wxWidgets libraries which are not available in some distro as a dependent package, then this type of upstream change to OCPN will necessarily require significant CI work, as well as custom-built plugins.

transmitterdan, 2020-10-20:
OpenCPN will build and run without 3rd party plugins.  Ergo, OpenCPN does not "depend" on 3rd party plugins for its core function. Many, many users find the program fully sufficient as it is and never add a 3rd party plugin.

Anyway, it would take an army of developers to do what you are suggesting.  Absent that there would have to be some kind of approval process for allowing 3rd party plugins to become submodules to the core.  Otherwise, scarce programming resources would be completely used up in keeping the CI process building with countless plugin changes.  Who is to decide which plugins get put into a second class status as "non-core" plugins?  I think no one wants that job.  As it is today, the CI process nearly ties up one developer full time what with constant changes to the CI ecosystem itself.

As for new processors, let's keep in mind the purpose of OpenCPN.  It is a helpful program for boaters.  It is not meant to live on the bleeding edge of the newest processors and countless OS variants that come out every week or two.  The target user isn't all that interested in the processor du jour.  Their needs should be given priority over the hobbyists who can't wait to try out the latest Arm.  The hobbyists desiring to run on the latest hardware will be happy to experience the cuts and scrapes that come with it.  But 99% of boaters will not see the value in it.  There are plenty of fully capable and well supported processors available today.  The cost starts at less than $100 for a highly capable Pi4. Nearly all boaters can simply pick any one of them and go boating.

mgrouch, 2020-10-20:
> 
> 
> This is simply not workable.

Disagree

> Take a simple example, using your latest OCPN branch linking against your private wxWidgets v3.1.4 build. Add a bunch of known good managed plugins, using git submodules added to OCPN Cmake scripts..

I've now made my wxWidgets as well as latest GTK3 build public here:
https://cloudsmith.io/~bbn-projects/repos/bbn-repo/ but that's not the point. 
You want CI to build as many combinations as possible. Where you target your product. As well as combinations where it is going to be targeted in future. You actually want to build against beta software of upstreams to be ready to release. Travis-CI martix makes it easier.
If it fails you learn that it fails and where it fails the moment change introduced upstream. For example they pulled one API from WxWidgets and I found you were using it
So they actually put it back. 
https://github.com/wxWidgets/wxWidgets/commit/365c3a2ae92fa2746f52678220c0864be4814d20
Because I found it. Otherwise your CI build would have found it if you had that combo in CI.

> I predict that virtually every linux/debian CI build will fail, simply because wx3.1.4 is not available in any of the docker images used to build those plugins. If any managed to build, they would likely not be loadable in OCPN, rejected as incompatible due to strict library ABI incompatibilities.

Initially it will fail. Cause its broken anyway. But CI will let you fix it and KEEP it stable.
You have CI. You can add a script how wxWidgets get built and you build OCPN with exactly same version as a part of the same build. Not going to be rejected because of ABI. Best part you do not do it manually. CI does it for automatically.
You just watch and fix the builds with your pace. But when you fixed it it's done and repeatable. And when it breaks you look what changed if the build is taking anything changing.

> So, what have we got? A broken CI build. And no compatible plugins.

You know now where it breaks. But if it works you have compatible plugins ready to be tested and possibly released.
BTW you can have a flag to turn off building plugins to make the core team focused on the OpenCPN itself if they build outside CI.
But if they change something in OpenCPN downstream will know right away. Cause it's going to break the full build.

> So what have we learned? We have a bunch of plugins that cannot currently be CI-built against wx3.1.4, or won't load if they do build. Who will fix them?

Those are breakage that would have been fixed anyway, right? This just let's you to discover them earlier in the development cycle.

> Answer: It is up to the plugin devs or packagers to update their work as required, staying in sync with API/ABI changes in OCPN core. 

True. But your release and your product is incomplete. Plugins are not working, do not exist in binaries, etc.
Not the case with CI.

>This is not just philosophical pablum. It is actually a critical element of the distributed development model we are using under present resource constraints. The OCPN core devs simply cannot come to "own" the entire plugin universe in order to assure a clean CI build. Just not practical...

CI is a time saver for all the projects. Takes time to set up but benefits are great. Plus you are sharing your build CI
all your matrix of possible platform/processor/GTK/WxWidgets/23-64 bit/Wayland-X11 combinations. So
plugins developers Do-Not-Repeat-Yourself. Key principal to rapid development. You could reduce your development cycles.

> 
> I chose your wx3.1.4 example on purpose. We will come back to this for more discussion, I'm sure. But let's be clear that today we have no CI workflow that supports custom wxWidgets builds on debian based distros. None.

Add it to the matrix. You can have build of GTK and wxWidgets as additional steps. (I provided the scripts).
I've now even added scripts to build .deb packages.

> So if a new feature (e.g. integrated touch control) requires new wxWidgets libraries which are not available in some distro as a dependent package, 

You can and SHOULD also have a build for upcoming release against trunk of UNRELEASED upstream key software.

>then this type of upstream change to OCPN will necessarily require significant CI work, 

Adding to a build matrix in CI.

>as well as custom-built plugins.

Attempt of the build will be made by CI. You will see where it breaks and plugin developers would need to fix it, or you could branch the plugin (which you already do).

Thanks

leamas, 2020-10-20:
As everyone here besides Mikhail I agree that we should reject the idea to include the plugins as git submodules. After bdbcat and transmitterdan chiming in, I guess case could be considered closed unless there is some really new arguments. Repeating the same just wont change anything.

However, there is IMHO a risk that we throw out the baby with the bathwater. What Mikhail has done is to explore the gestures support in wxWidgets 3.1, and this might be valuable. When wxWidgets 3.2 is out, probably not that long from now, this might be really interesting on devices like a raspberry. Developers exploring such things "in advance" is a great thing making us prepared for the upcoming releases.

This is similar to #2047, where @hreuver0183 tries to explore the arm64 support. I can see the move from raspbian to vanilla Debian be under way on the RPi (after all, Raspbian is kind of  kludge since Debian did not support RPi). So also here, we should welcome devs exploring the new stuff. 

Bottom line: while the submodules approach should be rejected, let's welcome devs exploring different new paths of OpenCPN, of course without commitments on if/when such things are included in mainline.

leamas, 2020-10-20:
As for the wxTransformMatrix issue above: Here is a point. Not to include the plugins as submodules, but that it might make sense to include builds of OpenCPN against latest wxWidgets 3.1 snapshot or master in the CI setup. Probably best done like a nightly job rather than triggered by OpenCPN commits.

Just to check we stay compatible, the wxTransformMatrix was a good catch. But still, this does not require any submodules.

IMHO:  "patches welcome" on this.

bdbcat, 2020-10-20:
@mgrouch 
Thanks for the detailed response.  Some valid points made.
We "should" do a lot of things. However, the reality is that dev resources will be better spent in fixing bugs and adding functionality to the most widely used platforms, rather than completing checkboxes in the distro/arch/plugin matrix.  So that is where the energy will be focused.
Thanks.

bdbcat, 2020-10-20:
@leamas 
"include builds of OpenCPN against latest wxWidgets 3.1 snapshot or master in the CI setup"
Good idea, but I'm not sure how we could cleanly package and distribute something like this.  debian mentors are not gonna like it....

leamas, 2020-10-20:
> Good idea, but I'm not sure how we could cleanly package and distribute something like this. debian mentors are not gonna like it

Clearly we shouldn't do that. What I envision is just a compilation check, so we can stay compatible with the upcoming  wx 32.release.  If @mgrouch hadn't found  the wxTransformMatrix issue this would have caused  an API break. We  could avoid such things by just ensure that OpenCPN compiles (in particular, the API)  against the latest wxWidgets master.

Packaging comes when 3.2 is here. The current roadmap claims that this will be December, but it has slipped before. We'll see.

closed 2021-04-10 by mgrouch