OpenCPN/OpenCPN #2069: flatpak: Appstream metadata validation fails.

leamas, 2020-10-20:
From the flatpak nightly build log:
```

+ appstreamcli validate app/files/share/appdata/org.opencpn.OpenCPN.appdata.xml
File 'app/files/share/appdata/org.opencpn.OpenCPN.appdata.xml' does not exist.
Validation failed.
+ :

```

bdbcat, 2020-10-20:
I guess I am confused.
In the ubuntu focal build, we create and package:
usr/share/metainfo/opencpn.metainfo.xml

Do I understand that flatpak wants
app/files/share/appdata/org.opencpn.OpenCPN.appdata.xml

Are they in fact the same file?

leamas, 2020-10-20:
I don't know for the moment, but consider it assigned to me. Will look into it. It definitely not urgen

leamas, 2020-10-21:
Playing around a little. Some points:
  - We supply the same file used in the Ubuntu software directory and in flathub.
  - The appdata file is renamed and is now called metainfo, living in /usr/share/metainfo
  - The spec docs, all there is for flatpak: https://www.freedesktop.org/software/appstream/docs/
  - Debian/Ubuntu docs: https://wiki.debian.org/AppStream, https://wiki.debian.org/AppStream/Guidelines
  - flatpak requires filename == id == org.opencpn.OpenCPN.yaml.
  - Debian requires desktop filename -> yaml file name i. e. desktop file is called org.opencpn.OpenCPN.desktop.
  - As for desktop file, flatpak can rename file at installation, allowing for opencpn.desktop.
  - However, corresponding rename for metainfo does not work (it did foir appdata).

All in all: to be consistent we need to rename the desktop file opencpn.desktop -> org.openpcn.OpenCPN.desktop. I don't think this should break anything (?).

PR #2070 (2020-10-22): Flatpak: Install appstream metadata as appdata.xml instead of metainfo.xml (#2069)
Use the appdata name instead, using the metainfo name was premature.

This should satisfy both the needs for flatpak (fully qualified name on org.opencpn.OpenCPN) as well as Debian (desktop and appdata should files should have the same basename).

Closes: #2069
closed 2020-10-22 by bdbcat