OpenCPN/OpenCPN #2071: RFE: Convert documentation to a plugin

leamas, 2020-10-22:
It is possible to publish the documentation as a plugin. Doing so, there would be some advantages:

  - For users, here is a streamlined workflow for downloading and not least installing the docs in proper 
    locations which can be linked in the main application.
  - Documentation authors would benefit from being able to update the docs after the main release. Overall,
    a simpler documentation update cycle might benefit both authors and users.
  - Packagers would have better options. Excluding the docs, the current default, is not really ideal since 
    downloading the documentation package is a bit complicated for users. However, including the docs  
    more than doubles the package size (15M vs 41M on linux), and also has problems since the docs are
    not FOSS.

A documentation plugin would mean that we introduce a new type of data-only plugins, compatible with all platforms. This should not be complicated, and I see no particular drawbacks.

Thougths?

rgleason, 2020-10-22:
Alec, I like this idea but don't know enough about the process.
The documentation should be updated and rewritten and broken down into several packages in my opinion.

- Basics
- Advanced
- External Plugin Documentation

Can the plugin be made to use Dokuwiki so that changes to the online documentation are simply imported to the plugin?
Can the plugin support adding "packages" or "books"  as above. 

I wonder if dokuwiki has some tools or API that would help?  I will ask on the dokuwiki website.

leamas, 2020-10-22:
Not everything at once, please... The natural starting point to me is the zipped HTML manual as available at opencpn.org. Converting this to a plugin would be pretty straight forward.

The rest of you questions are really if you could generate this zipped HTML manual automatically, and if it could/should be split to  smaller ones. These questions are IMHO not really related to this bug. After all, you can split the manual to smaller ones already today, offering multiple doc packages at opencpn.org. And as long as the zipped manual can be created,  it doesn't really matter for the plugin approach if it's automated or not.

Don't get me wrong, it would of course be nice if the zipped manual could be generated automatically. But if you want to do this, it should IMHO be a separate bug. Likewise when it comes to split the manual to subpackages.

rgleason, 2020-10-22:
I wouldn't call this a "bug".  I would elevate it to a "feature", but of course you and I have different perspectives.
If you want to zip  about 700 mb of Dokuwiki that is fine with me. I think smaller packages that could be added by the user would more appropriate.

You say HTML, Dokuwiki is very similar to HTML but different.

rgleason, 2020-10-22:
So you are planning on diverting and splitting the documentation efforts when we have a real need for more and better editing? I am not at all in favor of that.

rgleason, 2020-10-22:
Where exactly is this html that you are referring to? The manual uses docuwiki format.

CarCode, 2020-10-22:
Are you talking about documentation or the help file for OpenCPN?
As far as I know OpenCPN is multi language. But neither one is right now.

leamas, 2020-10-22:
I'm talking about the documentation package which is installed when setting OCPN_BUNDLE_DOCS, also available as a package to download at opencpn.org.

There are multiple problems with the docs, the missing translation is yet another one. But, these are the docs we have...

leamas, 2020-10-22:
@rgleason : Im talking about http://opencpn.navnux.org/data/doc482.7z, available for download at https://opencpn.org/OpenCPN/info/downloads.html, look for User Manual

leamas, 2020-10-22:
@rgleason. That package seems, at a glance, to be the same as what's installed in /usr/local/share/opencpn/docs when bundling the docs in the build.

leamas, 2020-10-23:
@rgleason: Do you know how these files are created i. e., the files under data/doc in the git repo?

rgleason, 2020-10-23:
We have the Site Export Plugin which does export XHTML 
I have had great trouble with this in the past because it fails on the entire wiki and takes ages to complete.
Perhaps it has gotten better.
The only way to get it to complete is to download and install dokuwiki locally and copy the Opencpn Contents and then set it up and run the Site Export Plugin.  Then it completes.
I have not tried it recently, as the whole process is painful.

rgleason, 2020-10-23:
We may need to create the document packages locally. 

I notice that using [dw2pdf](https://www.dokuwiki.org/plugin:dw2pdf) and [bookcreator](https://www.dokuwiki.org/plugin:bookcreator) which creates pdf's also has memory problems and timeouts which can be solved by using a local setup.

[SiteExport Plugin](https://www.dokuwiki.org/plugin:siteexport)  can use various ""Render Engines" including dw2pdr, XHTML
[Mandatory reading for SiteExport setup](https://github.com/i-net-software/dokuwiki-plugin-siteexport#dokuwiki-site-export)

leamas, 2020-10-23:
But then we have a bigger problem, right?  That the bundled documentation needs to be updated, and we really don't know how to do this. Or?

I'd say that the reasonable approach is to use HTML, this widely supported. I know we have discussed this before, but if my memory doesn't fail me the "sources" are reallly big (?). Do you have any data on this?

rgleason, 2020-10-23:
I think that if we must convert dokuwiki to HTML, we should get the documentation updated and in smaller packages, with new screenshots using a naming convention and all images stored more naturally.  

I think we should start fresh by creating a new Book on the wiki, and first build the Basics package, using the current documents as reference. We should try to set this document up to support other languages and implement the dokuwiki language feature as well.

rgleason, 2020-10-23:
Alec, I am heavily involved in updating Sean's plugins. We do not have enough good editors to maintain two formats for documentation.  Dokuwiki was selected by Dave as the opensource wiki to use. I implemented with the help of a number of others, the move from Drupal.  We have learned this system and I am not prepared to change it at this point. 

IMO any offline documentation for the program/plugin will have to be derivative (which means occasional) or use dokuwiki directly.

If you want to start your own separate totally open documentation effort, fine, it's opensource, but I am not going to participate.

leamas, 2020-10-23:
One possible approach is  that we work together on this:

You work with the process ending up zipped archive of HTML-files, an update of the current manual package at opencpn.org (could also be used to update the bundled docs).

And I work with the process starting with the zipped HTML manual, ending up in a new "Manual plugin".  This would also include some patches to OpenCPN.

Thoughts?

rgleason, 2020-10-23:
Yes, that would work I think and prevent the two processes from holding each other up!

leamas, 2020-10-23:
> I assume your plugin would use the current html document as a start.

Yes

> Then as we get updated, we would simply switch out the zip file?

Yes, more or less

> Then later after we complete the BASICS package, and actually have an ADVANCED package [...]

Let's complete the first steps, close this bug and then figure possible enhancements.

leamas, 2020-10-23:
OK? 

I'll not have much time until later next week, we'll see what happens then.

rgleason, 2020-10-23:
OK fine. Please close.

rgleason, 2020-10-24:
Started outline for "2020" manual.  Was kicked off, but will continue tomorrow.

rgleason, 2020-12-06:
I have been watching what Petri is doing for dashboard_tactics wiki and it is pretty darn good.
It has a linked index and the wiki is very legible. 
Why not move all plugins documentation to the the github wiki using Canne's methods?
Then the documentation could be downloaded with the code.
Here is Petri's example:
 https://dashboard-tactics-pi.readthedocs.io/en/v2.0.0/gettingstarted.html

rgleason, 2020-12-06:
Also, it is an expandable index, similar to dokuwiki
https://dashboard-tactics-pi.readthedocs.io/en/v2.0.0/influxdb/influxdb.html

I think this might work for all the plugins, the documentation would be distributed with each of the plugins.
Someone would have to figure out how this would be used when downloaded.
What would we do with the main OpenCPN dokuwiki content? 
Reconfigure it as I had just started to do, into separate and smaller sections and downloads 
1. [2020 Basic User Manual ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual)  -Distributed with OpenCPN
2. [2020 Advanced User Manual](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_advanced)  -Optional
3. [2020 External Plugins](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:plugins_external) -Optional

Maybe the External Plugins section just is a page that links to each of the github.com plugin wikis?
Questions;
1. Ease of making an epub for the current Basic & Advanced User Manual from Dokuwiki on an ongoing basis to be included with distribution?
2. How to handle linking of the External plugin documentation downloads on a user's computer.
3. Distributing the Plugin Documentation helps with download sizes, but creates a User problem regarding quick access.
4 Perhaps the plugin itself handles that with a link in the About section of Preferences, then nothing else needs to be done, it is just a self contained package with no links or connections to the main documentation.?

rgleason, 2020-12-06:
See https://tiddlywiki.com/dev/

rgleason, 2020-12-06:
I have not grasped the daily operational aspect of editing, it seems more like using a computer language, but there is a way to have language translations which is mapped out here, 
https://tiddlywiki.com/dev/#WikiText

rgleason, 2020-12-06:
A tiddly is a single html file with javascript which supports markup, contents, expandable table of contents, importing images, external images,installing a plugin, tiddlymaps plugin, etc all in one html file. It has a desktop application, firefox extension, etc.  It is quite powerful and might work, although images might have to be smaller than we currently are using. 

See these [videos by Francis Meetze](https://www.youtube.com/watch?v=ZMGpAW0z_Bo&list=PLzZCajspPU_UjFn0uy-J9URz0LP4zhxRK)  

How cn others edit these files when they are in the github wiki?

rgleason, 2020-12-06:
I may have been barking up the wrong tree. I wrote Petri and also copy his response below:

```
Dear Petri,
There is an ongoing discussion under issues about documentation.  You  have shown that we can have an expandible index and a decent content control with Markup.  Would you be able to comment regarding how you did this and how difficult it would be to convert all plugins to this approach?
Would other Editors be able to access the wiki to make changes/improvements an add content?
https://github.com/OpenCPN/OpenCPN/issues/2071#issuecomment-739510364
```
I hope he does not mind my sharing this:
```
Dear Rick,

I just got the keel detached from my boat and I am currently pondering what to do with 2 tons of lead, so I am quite busy, also with work.

I have not invented anything, my documentation has always been incorporated with the source code and as integrated part of my CI/CD process like in so many other projects.

For dashboard_tactics_pi v2 I just moved from github.io to [readthedocs.io](https://docs.readthedocs.io/en/stable/intro/getting-started-with-sphinx.html). It is, apparently the holy grail of open source build-integrating documentation hosting service - today...

Basically, you would need to decide do you want to go on with a wiki where made-with-good-will-but-unfortunately-now-out-of-the-date text blocks will stick on forever since nobody comes back to maintain them - the curse of any wiki open to the wind.

Or, do you want to go with the documentation-in-software where the documentation gets outdated only together with the software, when its development stops? After which nobody cares, anyway.

That basically answers your question about the editors: In the CI/CD incorporated documentation the editors need to make commits to the repository of the plug-in to change its documentation. If their submission is approved, readthedocs.io would chunk a new version of the documentation as part of CI/CD process.

Nothing to do with a Wiki.

Yes, that limits the number of the potential contributors, most of the CF would not bother.

For the source code incorporated documentation structure Markup seems to be the easiest to maintain since it converts easily and is now widely accepted in the FOSS community. It certainly works with Sphinx, which is used by readthedocs.io. And even if one does not want to use readthedocs.io for rendering, it is rendered on GitHub and probably also elsewhere.

[Without a document build tool, like Sphinx,](https://docs.readthedocs.io/en/stable/intro/getting-started-with-sphinx.html) making a structured document from a bunch of Markup-files is not impossible. It would just give, say "quite different results" depending on who is behind the keyboard.

Turning old documentation into source-code-incorporated Markdown... you must mean turning HTML-wiki text to markdown? There are tools for it, like html2markdown in Python. Would that go without sweat - certainly not!

Finally, don't bother to do marketing for Jupyterlab which I am using, it creates JSON files and would be probably just too much for most of the developers, increasing the complexity and boggling their minds. (I use Jupyterlab because it allows mathematical formulas, incorporated Python scripts for their calculations and all that scientific stuff which a plug-in developer would probably never need - but it provides WYSIWYG editing on a browser, exports Markup and works with Sphinx if you want to try it!)

I hope that you find the above thoughts useful,

Petri
```

rgleason, 2020-12-07:
So there is a choice: 

1. Tiddlywiki  https://tiddlywiki.com/   (simpler single html file with javacript, compatible with github wiki)
2. Sphinx  https://www.sphinx-doc.org/en/master/index.html which uses [reStructuredText Directives](https://docutils.sourceforge.io/docs/ref/rst/directives.html#include)  to output to HTML on github, etc, and other formats like PDF.  (More sophisticated than tiddlywiki, more features and versatility, needs python.)
3. Continue with Dokuwiki, build new packaged and current documentation after solving export to html or pdf.  Which is not really the way "CI/CD incorporated documentation the editors need to make commits to the repository of the plug-in to change its documentation." works. 

Any thoughts about this?

nohal, 2020-12-07:
After using all of them heavily I much prefer AsciiDoc or Markdown over reStructuredText. But anything is better than HTML.

rgleason, 2020-12-07:
Thanks Pavel,
https://asciidoc.org/index.html

This is worthwhile considering and discussing before embarking on documentation updates to Dokuwiki.  Also we should be aware of the strengths of dokuwiki.

Dokuwiki has a number of very nice features which we use:
- Authentication for users, which allows all users to contribute relatively easily. (Don't underestimate this kind of access!!)
- Expandable TOC and ordering, use of title
- Decent handling of images (could be better)
- Orphan finding and repair to clean up documentation.
- Some very good plugins that we use.
- Backup and previous pages (attic)
- Good support and forum
- We have not yet found an easy way to support CI/CD Documentation by being able to push html up to github.com Wiki.

rgleason, 2020-12-07:
I did a search on [dokuwiki to github markdown  ](https://www.google.com/search?channel=tus2&client=firefox-b-1-d&q=dokuwiki+to+github+markdown) and then found that there is a current [Dokuwiki Markdown plugin](https://www.dokuwiki.org/plugin:mdpage) compatible with our version of Dokuwiki. It supposedly will output to Github Markdown version  GSM.  Maybe this will work, but I doubt it.

leamas, 2020-12-07:
@nohal said:

> After using all of them heavily I much prefer AsciiDoc or Markdown over reStructuredText. But anything is better than HTML

Agreed. And, I'm pretty sure you would get similar answers from almost anyone involved in active development, opencpn or not.

The readthedocs.io approach looks interesting, being based on plain markdown but also able to create common document structure from different sub-projects maintained independently. In our case, we could have a common document including at least some plugins with the docs maintained in the plugins' repos.

As for your advantages with current approach:

> Authentication for users, which allows all users to contribute relatively easily. 

This is actually a drawback when compared to git pull requests. These does not require any authentication, and the technique is ubiquitous these days.

>    Expandable TOC and ordering, use of title
>   Decent handling of images (could be better)

At a glance, sphinx + markdown offers the same.

>   Orphan finding and repair to clean up documentation.

This is just tools to handle wiki-specific problems.  A markdown structure could (will) contain broken links, but there are multiple tools to locate such things.

> Some very good plugins that we use.

To achieve what?

> Backup and previous pages (attic)
>  Good support and forum

The git community  certainly beats any wiki one... And git not just offers backup, it offers complete history, branching...

>  We have not yet found an easy way to support CI/CD Documentation by being able to push html up to github.com Wiki.

Moving to something like sphinx + markdown would solve this. Note that the idea to push html probably can't  work, it  would be a maintenance nightmare. Any reasonable way to handle this would be to base the html on some kind of sources, markdown or not. And to update these sources rather than markdown.

leamas, 2020-12-07:
BTW: I'm currently trying to understand the Android build process to make an overhaul of it. It consumes all free cycles I have, making this documetnation issue to a lower priority for me.

rgleason, 2020-12-07:
Users are not programmers. They do not commonly use or are familiar with git and they do not generally have git accounts. I would estimate that of those who edit the wiki, programmers & users with git accounts are in the minority.  We have asked the User Community to maintain and build the documentation to help remove this burden from hard pressed programmers. Do we plan to change that and require that programmers spend more time documenting everything?  

How do you propose that we accomplish the huge rewrite that is needed?

Leamas wrote:
```
As for your advantages with current approach:

>    Authentication for users, which allows all users to contribute relatively easily.

This is actually a drawback when compared to git pull requests. These does not require any authentication, and the technique is ubiquitous these days.
```

leamas, 2020-12-07:
>  We have asked the User Community to maintain and build the documentation to help remove this burden from hard pressed programmers.

Does it work?

>  They do not commonly use or are familiar with git and they do not generally have git accounts

They certainly doesn't have an account on our wiki either; creating a github account isn't harder than any other account. With this,  a user could clone our repo, edit the file directly in the github GUI and make a PR by pushing some buttons. I don't really think this is harder than the steps necessary in current solution.

For non-developer users we could make a short guide how to do it. It's actually easy.

> How do you propose that we accomplish the huge rewrite that is needed?

One option could be to do it piecemeal, starting with the developer manual. This is clearly geared to developers, and moving these docs to where they belong in the the actual git repo(s) would make maintenance much simpler. 

Once we have some experience we could decide on possible more steps.

> While Sphinx may be the most robust and flexible tool we could use, it is clearly more complicated.

Perhaps, perhaps not. Clearly, setting up the overall structure is a major undertaking. Such an effort would of course also mean that each and every page will be reviewed when transferred to a new format.

OTOH, with the major structure in place (and some new templates for backgrounds etc.) it should actually be really easy to maintain, using the vast universe of git tools available including github. Most user contributions will be some simple editing of text files.

>  will try to learn more about it and this process.

I recommend the  screencast at https://docs.readthedocs.io/en/stable/intro/getting-started-with-sphinx.html.

rgleason, 2020-12-07:
Please see Andi's comments  (Andreas Gohr andi [at] splitbrain )  (head programmer for Dokuwiki) in the [Dokuwiki Forum here
](https://forum.dokuwiki.org/d/18568-github-cicd-documentation-workflow-commits/2)  - If you need to register and log in please do so to read the entire dialog.

In summary, there currently no plugin that does this. He writes:

> I'd think exporting DokuWiki to HTML would be the more reasonable way and it could for sure be automated. But might need some coding by someone who understands DokuWiki and PHP. One could also use the XMLRPC API to use any language for fetching HTML exports directly from DokuWiki via HTTP. This for sure could be done within the CI Pipeline of Github (probably best via scheduled actions). But again, it needs a programmer to do it.

I found this advice to be very useful

> I just saw your post, and appreciate your advice.
> 
>     Continue to use Dokuwiki and begin the long task of rewrite with our User/Editors helping.
>     Point the programmers to your post, to work with you solving the translation for Github Markdown or Html to support github CI/CD Documentation requirements.
> 
> That is a pretty simple direct solution which puts the responsibilities where they need to be.

Andi has offered to consult.

Therefore, I believe it is the programmer's responsibility to make the translation from Dokuwiiki to Github CI/CD Documentation happen, perhaps using html and it is the Users responsibility to take on the rewrite of the documentation using Dokuwiki.  I think it is beyond time to get going on the rewrite and we do not need to go searching for other tools.  The programming side can occur whenever it can happen, independent of documentation.

By the way, Andi wrote the dw2pdf translator and the html translator which are both available for reference.

leamas, 2020-12-07:
Well, the current Developers Manual is about 40 pages, not all actual enough to be kept. It should be perfectly possible to convert these to markdown using a manual process, with or without some pandoc pre-processing.

But agreed, it would be easier for someone with a programming background. 

To repeat myself: the arguments for moving manuals to the git repo are IMHO stronger for the developer manual than the rest.

leamas, 2020-12-07:
Found this: https://earthscinotebook.readthedocs.io/en/latest/wiki/transition2markdown/

Our issue isn't unique...

rgleason, 2020-12-07:
Maybe its time for me to take a back seat on documentation and let the programmers handle this. I would of course support and be involved, but I no longer have the interest or time to change everything again, learn and setup an entirely new system, etc. 

If the resulting solution is easy for me to use, I will use it and be able to help. If it isn't I won't.  I am certain that the many Opencpn User/Editors who have devoted time to this will do the same, but I fear that by moving to github you will be losing many Editors and valuable assistance and involvement. 

I do not have the knowledge or skills to decide on the future course of the documentation and I am not ready to move forward on an big effort to update Dokuwiki without these things being resolved.   Sorry, but that's the situation.

rgleason, 2020-12-07:
I tried pandoc online converter to convert AISradar dokuwiki to github markdown

https://pandoc.org/try/?text=&from=dokuwiki&to=gfm&standalone=0
1. log into dokuwiki, pick a page, select edit, copy the text.
2. go to pandoc, select dokuwiki in the left field, paste text from dokuwiki
3. in pandoc right field pick markdown flavored for git
4. pick convert, copy the text
5. go to yourr AISradar repository make a new wiki page, and copy the text.

This is the result
https://github.com/rgleason/AISradar_pi/wiki

leamas, 2020-12-07:
Well, I didn't assume the pandoc conversion should be good enough. But  should be a reasonable starting point for manual editing. The link above seems to give some hints on this.

I don't have the time for this right now. I see a path for the developer docs, which should benefit most from living in the git repo as plain text. But it will take time I don't have right now.

It would be great if you could do pandoc conversion on the developers manual and make it available somehow (I assume this requires some administrative privileges I don't have).  But besides this, we could perhaps just leave this bug open until there is some time?

rgleason, 2020-12-07:
I could dump it into my Opencpn wiki and give you collaboration rights.
But what about all the namespaces? 
How is that done?

The only thing I worked on in gitub markdown was Chartdownloader Documentation with Pavel and TransmitterDan. That was a pretty small document really.
Here is Pavel's final version with no screenshots https://github.com/nohal/chartdldr_pi/wiki
     I see that you can have separate page and there is TOC index to the right. (Similar to your wiki, Alec)
Here is my earlier version done for him  but it appears Pavel edited it so there are no images https://github.com/rgleason/chartdldr_pi/wiki    --Not sure how wikis work.

leamas, 2020-12-07:
Let's leave it for now. I don*t have the time anyway...

rgleason, 2020-12-07:
ok

rgleason, 2020-12-10:
I've tried pasting source Tiddlywiki code for AISradar into a Page in github.com  Wiki  and found that Tiddlywiki is not supported because I cannot select HTML.   https://github.com/OpenCPN/OpenCPN/wiki/TiddlyWiki-with-editor.html 

However there is a dropdown selection for Editing AsciiDoc, Markdown, Creole, Org-Pod, RDoc, Pod, Testtile, and ReStructureText (used by Petri for DashT ), so those will work with Github Wiki.   

If github Wiki is going to be used, I don't know if there are any good reasons not to use MarkDown, or a very good reason to use one of the others.

rgleason, 2020-12-10:
After working with markdown, once again, I remember it now.
1. It is a little cumbersome to me right now. 
2. Translation is not really seamless as all the links have to be reworked.
3. Graphics and screenshots must be stored elsewhere.
4. It is going to take some serious time to move Plugins to Wiki
5. User/editors are going to have to get a free github account and learn something else.

Before doing all this work, I would like to experience for on plugin 
1. How this wiki gets included in the tarballs and packages.
2. How seamless the access to the Manual is once the plugin is installed.

leamas, 2020-12-10:
While markdown is fine, I don't think github's wiki implementation scales to the full-fledged manuals We need something else, tailored for larger, more complex docs.

> Graphics and screenshots must be stored elsewhere. Where?

Markdown is typically stored in a git repo, and pictures etc. can be stored in the same repo.

> The table of contents or Index must be hand coded for a multilevel index

This cannot be maintained manually. That's the reason for a tool like sphinx.

> User/editors are going to have to get a free github account and learn something else.

Indeed. OTOH, developers already have the skills and can update docs in the same way as they update code, in the same workflow. This is a compelling feature, especially for the developer manual.

rgleason, 2020-12-10:
Alec,  First congratulations to you and Bdbcat on navigating a very difficult transition! Kudos to you both.
 Re above
- 1 Agreed.
- 2 Are you suggesting that the Markdown and screenshots be in  github.com/[user][plugin]/**docs?**  Or are you suggesting an entirely new repo? I thought github discouraged that, at least that is what Pavel told me. Example?
- 3 We absolutely need good translation, namespace management, TOC tools and good graphics/screenshot controls, auto internal links for titles, good link translation. Does sphinx doe this?  How difficult is sphinx?  Will it translate without a lot of work? How much work to set it up? How user friendly?
- 4  I am afraid we are going to lose a lot of the troop of Editors and there is no way to get this done otherwise, unless you, altready stressed programers, or I do it. I don't see me doing it without that help.

 Can you explain to me the advantages of having the documentation in the "CI/CD flow"?  How will things be automated? What will happen? What is the User experience as a result of this change? 

I certainly don't want Users to have to piece together their documentation!  It should be downloaded with each plugin and accessible from both the Opencpn Help  Manual button and a button in preferences in the plugin, requiring no preconfiguration, separate download etc.

I fully support CI/CD workflow provided it can be done reasonably such that we have the manpower to transition and the tools necessary to make it easy. I don't think I understand its advantages yet and I don't think we have found the right tools yet.

What was your proposal for a plugin that would translate Dokuwiki?  That could be run just before releases to update the github repository documentation and be included with the release.  Then we would just continue to use Dokuwiki. I thought that was a great solution.

At this point I don't see myself migrating the Dev Manual because I don't think we have the right tools yet.. I do think your MarkDown Plugin Manual is good, (sans graphics) but getting there with the Dev Manual will be very "expensive" to put it in your terms.
Best. Rickk

leamas, 2020-12-10:
The more I think about this I get the feeling the developer manual is a different beast than the user manual. In short, the developer manual is (should be) maintained by developers and consumed by developers. OTOH, the user manual is written by and for end users, sort of. So perhaps the one size fits all approach isn't the optimal.

Let's put the user manual aside from now.

Looking at the developer manual, good such manuals are always, no exceptions, part of the source somehow. The basic idea is that developers has the docs close enough so they an update the docs when updating the code. In our case, this would mean that the developer manual was part of Opencpn git repo, perhaps under new directory like 'Manual'. 

Also, the requirements for the developer manual  is different. In particular, there is no need for translations -- developers knows English (more or less...). In general, I also think that the developer manual is less about pictures, layout and such.  

OTOH, the overall need for TOC and other structure is needed also here. I do *think* that Sphinx should fix this, but I have just skimmed it on the surface. And to be honest, that was long ago and in a python context.

How hard is it? No way to tell without trying, for sure.

What I envision is a directory with pages like the .md files l leave around here and there. Plus an overall structure maintained using sphinx or some other tool.

leamas, 2020-12-10:
The workflow advantage: Non-trivial PRs would include the documentation update in this context. Developers are lazy, they don't want to mess with specific tools for documentation. OTOH, if the docs are part of of the git PR, all is fine and as they are in other projects out there.

leamas, 2020-12-10:
As soon as look into aspects like internationalization and structure sphinx actually looks good. As for i18n, it seems to use gettext which means that we could translate the manual using the same tools and services as used when translating the program. 

But then again, focusing on the developer manual there is no need for translations.

rgleason, 2020-12-10:
"the developer manual is (should be) maintained by developers and consumed by developers. "

I agree with you about developing manual, but one part, the development beta plugins should have a docuwiki beta plugin page withl lnks to all beta plugin documentation that can be in sphinx and stored in git somehow. This gets advanced users into testing too.

Dev manual will become much better maintained easily by programmers. I would like to have access too.

leamas, 2020-12-10:
As I understand it, sphinx shines here. It has the notion of independently maintained subprojects which (again, as I understand it) not just can be linked but really integrated to a common manual. This maps perfectly to our distributed development model.

Assuming the devel manual is just a subdirectory in main OpenCPN everyone has "access" since anyone can make a clone, make changes and eventually also a PR to get the changes into the main site.

The plugin docs should then live in the plugin source code git repos. And everything kept together (linked, common look & feel, ...) by sphinx.

nohal, 2020-12-10:
I personally have abandoned both Markdown and Sphinx for documentation and switched to [AsciiDoc](https://asciidoc.org) with [Asciidoctor](https://asciidoctor.org). The main reason for the swich were AsciiDocs's superior formatting capabilities over Markdown and Asciicdoctor's HTML and PDF theming support fitting with my particular needs. While I do not need it for anything, [Antora](https://antora.org) does a lot of the stuff Sphinx does in this ecosystem out of the box.

For example everything you can see at https://docs.kicad.org is completely built with a toolchain similar to mine (And has answers to most if not all of OpenCPN's needs - multiple versions, multiple languages, multiple output formats. The source for the docs is at at https://github.com/KiCad/kicad-doc, where you can also read [why they decided to go that way](https://github.com/KiCad/kicad-doc/tree/master/doc_alternatives))

rgleason, 2020-12-10:
(I see that Pavel has posted just ahead of me about AsciiDoc which is supported by github wiki.  Thank you very much Pavel, we need to look into this too!)

Sphinx ...I do hope this will considerably simplify everything!  
If, for the time being, User Manual (Basic & Advanced) stays as Dokuwiki, is it possible to have links to the plugin manuals at the github locations?  
Maybe the next step is to get sphinx working and learn more, and get the right setup?  
Petri's Dash-T is an example:
- [github.com/canne/docs](https://github.com/canne/dashboard_tactics_pi/tree/dev220/docs) 
- [v2.0.0 using sphinx and readthedocs.io](https://dashboard-tactics-pi.readthedocs.io/en/v2.0.0/)

Then how do we get the manpower to move this documentation?  Put out a call on Cruiser Forum? Contact existing editors/managers of Dokuwiki directly? (We have about 65 listed.)

If the move of Development Manual and Plugins goes well and meets your expectations, then maybe we try doing the next big chunk:  
1. Basic User Manual  so it gets automatically and seamlessly included with releases. This will inevitably require a big rewrite, helped by an army of volunteers.
2. Advanced User Manual would be an optional download, and could follow in some way.

rgleason, 2020-12-11:
In process of setup. Have nvm and nodejs installed so far.

nohal, 2020-12-11:
If you start with Antora, I don't think it is the right end of the chain.

rgleason, 2020-12-11:
Yes, I will next install asciidoc and asciidoctor then antgora.  Thanks.

rgleason, 2020-12-12:
Pavel, Do I need Ruby + Devkit 2.7.2-1 x64  or just Ruby 2.7.2.-1 x64?
I am down to #4

**Installation**
1. Download and Install Python 3.9.1    https://www.python.org/downloads/   AsciiDoc needs python >=3.5
2. Download and Install AsciiDoc  9.0.4 https://github.com/asciidoc/asciidoc-py3/releases
3. How to install AsciiDoctor on Windows https://prj2.fontysvenlo.org/install_asciidoctor.html
4. Download and Install Ruby + Devkit 2.7.2-1 x64 https://rubyinstaller.org/downloads/  
  - check the Option "Add Ruby executable to your path" during installation routine 
6. Install AsciiDoctor https://asciidoctor.org/ just below
7. Now Open a Windows Command Shell (WIN + type CMD) and install the asciidoctor gem:
  - gem install asciidoctor
  - asciidoctor --version
          Asciidoctor 1.5.5 [http://asciidoctor.org]
          Runtime Environment (ruby 2.0.0p648 (2015-12-16 revision 53162) [universal.x86_64-darwin16]) (lc:UTF-8 fs:UTF-8 in:- ex:UTF-8)
6. Antora Requirement for Windows https://docs.antora.org/antora/2.3/install/windows-requirements/
7. Install Chocolatey  https://docs.antora.org/antora/2.3/install/windows-requirements/#install-choco
8. Install nvm  https://docs.antora.org/antora/2.3/install/windows-requirements/#install-nvm 
9. Install Node (Long Term Support LTS)
10.Install Antora Globally https://docs.antora.org/antora/2.3/install/install-antora/
11. When we say “globally” here, it does not imply system-wide. It means the location where Node is installed. If you used nvm to install Node, this location will be inside your home directory (thus not requiring elevated permissions).
Install using the latest long term support (LTS) release of Node. While you can use other versions of Node, Antora is only tested against active LTS releases.

**Additional Information about use**
1. Asciidoc writers guide  https://asciidoctor.org/docs/asciidoc-writers-guide/
2. Asciidoc is just plain text! So it is helpful to have a text editor that colors text for legibility.
3. You can use Windows Notepad++, Atom Linux Vim, Emacs  MacOS Textmate

rgleason, 2020-12-12:
While looking into Energy Software Interfaces with Energy+ I found [Honeybee by Ladybug Tools](https://github.com/ladybug-tools/honeybee-legacy/wiki) which has an excellent example for Markdown use for plugins.  This Github Wiki has a decent Table of Contents and supports images within the text, using github storage. 
You can click on edit for both the content and the table of contents to view the code.
It appears everyone can edit it.
https://github.com/ladybug-tools/honeybee-legacy/wiki

The graphics are stored in this way, EG

`![depth of analysis](https://user-images.githubusercontent.com/44324576/51753145-60685680-20b9-11e9-8526-299586429511.png)`

[image]

[177 more replies]

closed 2021-04-23 by nohal