OpenCPN/OpenCPN #2075: 5.2.4 Build failure in Mageia Cauldron - Cannot run wxWidgets version test (!)

barjac, 2020-10-28:
We patched 5.2.0 CMakeLists.txt as follows to allow build, but this patch does not fix 5.2.4:
```
diff -ur OpenCPN-5.2.0_o/CMakeLists.txt OpenCPN-5.2.0/CMakeLists.txt
--- OpenCPN-5.2.0_o/CMakeLists.txt	2020-07-15 19:02:29.000000000 +0100
+++ OpenCPN-5.2.0/CMakeLists.txt	2020-07-17 17:40:55.875774787 +0100
@@ -756,9 +756,9 @@
   # looked for them. This is a nasty fix which might fail miserably. Assumption:
   # All builds using GTK uses unicode and wxWidgets 3.0
   if (GTK3_FOUND)
-    list(APPEND wxWidgets_LIBRARIES "-lwx_gtk3u_aui-3.0")
+    list(APPEND wxWidgets_LIBRARIES "-lwx_gtk3u_aui-3.1")
     if (OPENGL_FOUND)
-      list(APPEND wxWidgets_LIBRARIES "-lwx_gtk3u_gl-3.0")
+      list(APPEND wxWidgets_LIBRARIES "-lwx_gtk3u_gl-3.1")
     endif ()
   endif ()
```
The build log (as far as it gets) is here:
[http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/opencpn-5.2.4-1.mga8_bcj.src.rpm/build.0.20201026230243.log](url)

Any help appreciated!

Barry

leamas, 2020-10-28:
Hi,

Some points:
  - We are currently is a "patches welcome" state when it comes to other Linux distros than Debian/Ubuntu.
  - Have you tried the flatpak version? This should run fine on Mageia as well.
  - Even if you make it build on Mageia, there are no plugins available for this platform. Unless there is a strategy for handling this, one might want to reconsider the build attempt.
  - Unfortunately, the link to log above does not work.

leamas, 2020-10-28:
The failure stems CMakeLists, this part:

```
if (GTK3_FOUND)
  set( _VERSPROG [=[
    #include <stdio.h>
    #include <wx/version.h>
    int main(int argc, char**argv) {
      printf("%d\\n", wxMAJOR_VERSION * 10 + wxMINOR_VERSION);
    }
  ]=])
  set (TRY_COMPILE
    "echo '${_VERSPROG}' > wx-32.c; cc $(wx-config --cflags) wx-32.c; ./a.out")
  execute_process(
    COMMAND "/bin/sh" "-c" "${TRY_COMPILE}"
    WORKING_DIRECTORY ${CMAKE_BINARY_DIR}
    OUTPUT_VARIABLE _WX_VERSION
    ERROR_VARIABLE _WX_ERRORS
    OUTPUT_STRIP_TRAILING_WHITESPACE
  )
  if (NOT "${_WX_ERRORS}" STREQUAL "" OR "${_WX_VERSION}" STREQUAL "")
    message(FATAL_ERROR "Cannot run wxWidgets  version test (!)")
  elseif ("${_WX_VERSION}" LESS 32)
    add_definitions("-DNDEBUG")
    message(STATUS "Forcing -NDEBUG on gtk3 build (#1500).")
  endif ()

endif ()
```

that is, for some reason it is not possible to compile the test program using the flags provided by `wx-config --cflags`.

barjac, 2020-10-28:
Maybe I should have mentioned that I am the maintainer for opencpn packages in Mageia including the plugins.
The link to the log does work and is correct. It is github that replaces it with a link back to github - crazy! Just copy the text of the link and put it in a browser.
Not sure what changed between 5.2.0 and 5.2.4 as with the simple patch 5.2.0 build was OK.
Is the test essential?
Can you suggest a patch to fix the problem?

leamas, 2020-10-28:
Yes, the link worked, thanks.

You can try to just remove the complete `if (GTK3_FOUND) ... endif() ` above. Just make sure that you use -DNDEBUG when compiling anyway.

I note that the non-free plugins for example oesenc_pi is missing in Mageia.  This is one of the more useful ones, at least for European users, readily available in Flatpak. Also note that while the packaged plugins are fine, the new UX in opencpn is that users installs the plugins from within opencpn -- but this requires another kind of plugins known as "managed plugins" or tarballs.

Yes, we are trying to move non-debian linux users to flatpak. It's hard (using Fedora myself), but probably the best route over time. OTOH, that is not to say it is to happen immediately.

barjac, 2020-10-28:
Thanks, it's now building and I have almost packaged oesenc plugin, however it appears to be installing what looks like a library in /usr/bin!
/usr/bin/libsgllnx64-2.29.02.so
That can't be right. Can it?
Also the site says the current version is 4.0.10 but in git it's 4.2.17.1 - all very confusing ;)

rgleason, 2020-10-29:
I believe it is likely that the Downloads page on the website shows the version number for the non-managed version of the plugin and it is an earlier version.  This is the version that can be downloaded from the links for MacOS and Windows (I believe).

leamas, 2020-10-29:
The helper library is correct. Note that there is no source for this library, it's a binary blob used handle DRM. I don't know the specifics for Mageia, but most distributions does not allow packages which are not completely FOSS.

The new plugin loader walks around this by downloading the DRM stuff in runtime, similar to how browsers handle the DRM issue.

barjac, 2020-10-29:
OK thanks. Yes that rules out packaging oesenc plugin. I know nothing about flatpack (yet).
So in future all we will need to distribute is the main opencpn package and it will be able to install all plugins directly on line?
If these plugins will be in tarballs, where does flatpack come in?
Also when will this happen? I see the plugin catalogue updater already, but it finds nothing that is not already installed.

leamas, 2020-10-30:
In short, flatpak is a binary distribution format which works on all Linux distros. It's similar to snap. More: https://flathub.org. Basically, flatpak buys the platform compatibility to the price of a huge (~0.5GB) initial download.

As I see it, there are two strategies for Mageia opencpn packaging.

The first would be to leave things as they are, opencpn and some plugins packaged. While doing so, communicate to your users that only flatpak gives the full experience (all plugins, simple plugin management, access to latest plugin versions). Eventually users will move to flatpak, but this will take time and the native packages are still needed as a transition tool.

The second would be to actually make Mageia plugins available. This is nothing like normal packaging, for sure. The outline:

  - Set up CI build chains which build the plugins for Mageia, one for each plugin/platform/arch combination
  - The CI build chains also produces an xml metadata file, one for each plugin build. Metadata is similar to packaging metadata but also includes an URL where the plugin build is available
  - Store the built artifacts  on a public server (usually cloudsmith.io)
  - Make a PR against the github.com/OpenCPN/plugins  with the metadata included. When accepted, this makes the Mageia plugins visible in the GUI.

Even if it looks worse than it actually is, there are templates which could be used, I'd still advice against the second alternative. From an overall perspective users on "odd" linux distros like Arch, OpenSUSE, Fedora and Mageia would benefit from using and maintaining a common flatpak packaging rather than splitting scarce resources on multiple, small platforms.  

My own platform of choice is Fedora.

barjac, 2020-10-30:
Thanks for taking the time to explain our options.
We are getting close to version freeze for Mageia 8 so now is not the time for major changes. I will leave things as they are and re-evaluate in the next cycle. To me, from what you say, flatpacks sound a bit like static builds. I will do some reading.
If Fedora take the second option we could maybe collaborate on this, as we already do for some projects.

leamas, 2020-10-31:
It's not that you or anyone else has to do anything to run the opencpn flatpak. As of today, the simple setup  is described in https://flatpak.org/setup/Mageia/. After completing this, just point the browser to https://flathub.org/apps/details/org.opencpn.OpenCPN, here is an install button. You can test, in particular the plugin handling.

Fedora basically does nothing when it comes to native OpenCPN rpms.  That said,Fedora is one of the major stakeholders in the flatpak project, and Fedora has integrated flatpak builds in the build chain (koji can build also flatpak packages besides rpms). The same flatpak'ed Opencpn is used also in Fedora.

asy972, 2021-02-07:
> We patched 5.2.0 CMakeLists.txt as follows to allow build, but this patch does not fix 5.2.4:

As a note. I did the same when I built 5.2.x for ALT Linux and both versions was compiling. I didn't need any additional changes.

rgleason, 2021-04-22:
Time to close this? OP?

rgleason, 2021-04-23:
Won't support. Closing

closed 2021-04-23 by rgleason