OpenCPN/OpenCPN #2079: Add 'close' link to about screen.

mgrouch, 2020-11-06:
On some systems with touch screens you can't rely on window having icon decoration to close it (those you usually have in right up corner).

In my system once I visit 'About' dialog I can't even leave it.

Please add "Close' link on the bottom of the dialog.

Thanks

mgrouch, 2020-11-06:
Actually the problem seems to be wider spread.
I clicked on update plugin catalog without internet connection. It have me error window with no button to close it.
I have to kill OpenCPN and restart.

bdbcat, 2020-11-10:
I agree that a normal small "close" decoration can be difficult for touch screen use.
But the bigger question, to me, is: "what system/desktop manager does NOT show decorations?.. and why?"
On linux, GTK2, wx3.0, the "About" dialog has a "close" decoration.  Same for Windows/Macos.

mgrouch, 2020-11-10:
Gtk2 window managers do not show decorations for applications compiled with gtk3 because gtk3 expects client side decorations.

bdbcat, 2020-11-10:
Hmmm...
Wonder if wxWidgets will provide some sort of "compatibility" mode for this?
Seems like this breaks a lot of legacy apps, us included...

mgrouch, 2020-11-11:
wxWidgets provide options to compile against GTK2 or GTK3. This somewhat shields you from GTK2 to GTK3 migration.

bdbcat, 2020-11-11:
So, if you are using a GTK2 window manager, why do you compile OCPN for GTK3?

mgrouch, 2020-11-11:
Because I didn't know what I was doing

mgrouch, 2020-11-16:
Well I compiled using GTK2 and GTK3 on Ubuntu and on Debian arm32, arm 64
And this issue is in every build.
I used wxWidgets 3.1.*

The builds are here:

GTK2 Debian arm32/64, wxWidgets 3.1.*
https://cloudsmith.io/~bbn-projects/repos/bbn-gtk2-repo/packages/
(Click "Set me up")

GTK3 Debian arm32/64, wxWidgets 3.1.*
https://cloudsmith.io/~bbn-projects/repos/bbn-repo/packages/
(Click "Set me up")

GTK3 Ubuntu (all platforms), wxWidgets 3.1.*
https://launchpad.net/~bbn-projects/+archive/ubuntu/bbn-opencpn/+packages

apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 24A4598E769C8C51          # bbn-opencpn on launchpad

and put

```
deb http://ppa.launchpad.net/bbn-projects/bbn-opencpn/ubuntu bionic main
deb-src http://ppa.launchpad.net/bbn-projects/bbn-opencpn/ubuntu bionic main
```

into /etc/apt/sources.list.d/bbn.list

Also for complete OS image you can download 

lysmarine_0.9.1.2-raspios-arm64.img.zip from

https://cloudsmith.io/~bbn-projects/repos/bbn-repo/packages/?q=Lys

It showcases all the bugs I've reported recently.

Thanks,
--MG

mgrouch, 2021-04-10:
proper openbox config fixed it for me

closed 2021-04-10 by mgrouch