OpenCPN/OpenCPN #2080: Window to reset into safe mode appears with no text inside

mgrouch, 2020-11-07:
Window to reset into safe mode appears with no text inside.
The dialog prompting for safe mode appears tiny, shows no text inside and 
only 'Yes', 'No' buttons are visible.
This is with latest GTK and WxWidgets on arm rpi 4.
Thanks

bdbcat, 2020-11-10:
Thank would appear to be a wx3.2 regression problem.  Looks fine on all other platforms/distros.

mgrouch, 2020-11-16:
Yes it is related to GTK3/wxWidgets3.1
Is there any workaround?

bdbcat, 2021-04-29:
Could we see a screenshot, please?

bdbcat, 2021-05-03:
Still waiting for screenshot.
Tried to reproduce, and no success. Works fine on rpi4, 64 bits, raspbian OS, wx3.1.2 local build.
Please confirm the buildtime/runtime environment.

mgrouch, 2021-05-03:
It was with gtk3 and wxWidgets latest

bdbcat, 2021-05-03:
OK, I'll try again with wx3.1.5.  What OS were you using?

mgrouch, 2021-05-03:
Raspberry os.

bdbcat, 2021-05-04:
Cannot reproduce with wx3.1.5.
Really would like to see a screenshot.....

bdbcat, 2021-05-05:
Closing, cannot reproduce.
Please re-open with more detailed information, if necessary.

closed 2021-05-05 by bdbcat