OpenCPN/OpenCPN #2081: Clear guide to compile OpenCPN for Android

floviolleau, 2020-11-09:
Hi everybody,

Thanks for your tremendous work here!

I would like to compile the APK for android of OpenCPN but I don't see a clear and up-to-date guide.
I'm doing the compilation for now under Ubuntu Budgie up-to-date on a laptop.

I followed [this README](https://github.com/OpenCPN/OpenCPN/blob/master/buildandroid/README).

I downloaded all components and followed from steps A to E. I skipped E. I and E. II and I used master branch for OpenCPN at step F.

I'm stuck when I compile OpenCPN at step F. 5. I
What I tried:
- get `cross_build_android.cmake` from another branch / old commit without success
- use a different branch of opencpn without success
- when I'm doing the cmake command, some discovery are missing. So I used cmake-gui to see what is missing and what I need to fill in. Most of the time, paths to the libs and headers are missing so I filled it manually.

<details>
  <summary>>>>> Here are my notes (Click to expand!) <<<<</summary>

```
# Versions used:
- Ndk-r10e
- Qt5.1.1_android_x86_x64/5.1.1 (Qt 5.2.1 is not available anymore for download)
- cmake: 3.16.3

# QT install
sudo apt install libstdc++6:i386 libgcc1:i386 zlib1g:i386 libncurses5:i386

# OpenCPN
sudo apt install build-essential cmake cmake-gui gettext git-core gpsd gpsd-clients libgps-dev wx-common libwxgtk3.0-gtk3-dev libglu1-mesa-dev libgtk2.0-dev wx3.0-headers libbz2-dev libtinyxml-dev libportaudio2 portaudio19-dev libcurl4-openssl-dev libexpat1-dev libcairo2-dev libarchive-dev liblzma-dev libexif-dev libelf-dev libsqlite3-dev libsndfile1-dev libarchive-dev zlib1g-dev libbz2-dev libjpeg-dev liblz4-dev libwxgtk-webview3.0-gtk3-dev libgtk-3-dev

# configure NDK Build the required cross-compiler toolchain
/home/user/projects/Android/Ndk-r10e/build/tools/make-standalone-toolchain.sh --toolchain=arm-linux-androideabi-4.8 --platform=android-19 --install-dir=/home/user/projects/Android/Ndk-r10e/ndk-toolchain

# prepare wxWidgets
cd /home/user/projects/opencpn
git clone https://github.com/bdbcat/wxWidgets.git
cd wxWidgets
git checkout master
mkdir -p build_android
cd build_android

export PKG_CONFIG_PATH=/home/user/projects/opencpn/Qt5.1.1_android_x86_x64/5.1.1/android_armv7/lib/pkgconfig
export CPPFLAGS=-D__ANDROID__
export PATH=/home/user/projects/Android/Ndk-r10e/ndk-toolchain/bin:$PATH
export CC=arm-linux-androideabi-gcc
export CXX=arm-linux-androideabi-g++

# first tried
../configure --with-qt --build=x86_64-unknown-linux-gnu --host=arm-linux-androideabi --enable-compat28 --disable-shared --disable-arttango --enable-image --disable-dragimage --with-libtiff=no --disable-baseevtloop --disable-xrc --disable-cmdline --disable-miniframe --disable-mdi --disable-stc --disable-ribbon --disable-propgrid --disable-timepick --disable-datepick --disable-xlocale --disable-intl --with-opengl --enable-selectloop

# command I finally used
../configure --with-qt --build=x86_64-unknown-linux-gnu --host=arm-linux-androideabi --enable-compat28 --disable-shared --disable-arttango --enable-image --disable-dragimage --with-libtiff=no --disable-baseevtloop --disable-xrc --disable-cmdline --disable-miniframe --disable-mdi --disable-stc --disable-ribbon --disable-propgrid --disable-timepick --disable-datepick --enable-xlocale --enable-intl --with-opengl --enable-selectloop PKG_CONFIG_PATH=/home/user/projects/opencpn/Qt5.1.1_android_x86_x64/5.1.1/android_armv7/lib/pkgconfig

make -j4

sudo make install

# OpenCPN
cd /home/user/projects/opencpn
git clone https://github.com/OpenCPN/OpenCPN
git checkout master
mkdir -p build_android
cd build_android
cmake -D_wx_selected_config=androideabi-qt -DCMAKE_TOOLCHAIN_FILE=../buildandroid/build_android.cmake -DwxWidgets_INCLUDE_DIRS=/usr/local/include/wx-3.1 -DwxWidgets_LIBRARIES=/usr/lib/x86_64-linux-gnu/ .. 

cmake-gui .
# fill in all missing libs manually and configure > Generate

make -j4 => fail
```
</details>

It seems that there is an issued/mix between wx-3.0 and wx-3.1
So because it is a minor version between both, I did a `sudo ln -s /usr/local/include/wx-3.1/ /usr/local/include/wx-3.0` without luck.. :(

Can someone update the README or at least help me / give me hints to be able to continue?

Thanks for your time ;)

bdbcat, 2020-11-10:
Yes, Android build environment and guide are a mess.  Lots of errors, version inconsistencies, platform dependencies, etc.
We are working to improve this now.  Please be patient.
Thanks
Dave

floviolleau, 2020-11-10:
Ok cool. Thanks for the quick update!

I stay tuned on that issue.

Keep safe

Twillpower, 2024-12-11:
When I compile Android, compilation fails. I am using version 5.10.2 and my error is caused by the make -f Makefile.android step. chart1.h cannot be found when compiling ocpn_wrapper.cpp. I found that chart1.h was removed two years ago, but ocpn_wrapper.cpp was last modified 9 years ago. Do I need to modify ocpn_wrapper.cpp and how

bdbcat, 2024-12-11:
The canonical way to build OpenCPN for Android starts with building the OCPN core as a shared library.
For validation of the CI build process, this library is built by the CircleCI CI engines at every code commit.
Here is the most recent build:
https://app.circleci.com/pipelines/github/OpenCPN/OpenCPN/3583/workflows/bd521bb5-b0c4-4c15-9bc6-ce9005307f67/jobs/11545
Other build workflows are certainly possible, but are not supported due to resource constraints.
Good Luck.

stone1706, 2024-12-12:
> The canonical way to build OpenCPN for Android starts with building the OCPN core as a shared library. For validation of the CI build process, this library is built by the CircleCI CI engines at [...]

Thanks for your relpy !
For now ,The library is built by the CircleCI CI .  there are lots of output files after job is done, but I cann't find APK file, I think these files just are libs file . What is next step I can follow to build APK file ?

Thanks again.
stone

stone1706, 2024-12-13:
Hi bdbcat,
the library is built by the CircleCI CI is 32bit ,not for 64bit ,
when I installed APK on my panel  and run it, then I got error message like below:

"java.lang.UnsatisfiedLinkError: dlopen failed: "/data/app/..../lib/arm64/libgorp.so" is 32-bit
...

bdbcat, 2024-12-13:
By reading the source, especially the shell scripts in directory "cmake", you should be able to replicate the CI build locally, and then modify to produce an AARCH64 build.

Twillpower, 2024-12-17:
I successfully build and run the Android version, but I can't find the import plugin function, the desktop version has this function, is the Android version does not have this function or I have a problem compiling? How do I import other plugins on Android

sebastien-rosset, 2024-12-20:
@Twillpower , do you have reproducible build instructions for Android? I have started #4272 , but right now I'm just at the stage of being able to run `make`.

Twillpower, 2024-12-20:
Did you only use the documentation in buildandroid under opencpn?

sebastien-rosset, 2024-12-20:
> Did you only use the documentation in buildandroid under opencpn?

I read other cmske files as well and used some trial and error. I just completed a successful `make`. Now I need to build the APK.

Twillpower, 2024-12-20:
The opencpn project only provides a linked library; to compile an apk, you need to use the oepncpn-android project

sebastien-rosset, 2024-12-20:
> The opencpn project only provides a linked library; to compile an apk, you need to use the oepncpn-android project

Ok. The reason I ask if you have step by step instructions is because I made a few enhancements in the VDR plugin. I tested on Linux Windows and Mac but not on Android. @rgleason pointed out the VDR plugin is needed to test other capabilities in Android. Hence I'm trying to get debug builds both for OpenCPN and the VDR plugin. Then I'll be able to test on Android, and fix issues if necessary.
As was pointed out, the build instructions are severely outdated, so if anyone has specific and reproducible steps to build and test for android, that would be great. Otherwise I'll keep trying.

Twillpower, 2024-12-20:
Yes, Android builds are tricky, I will try to document it later and commit it in my branch

rgleason, 2024-12-20:
Perhaps using Testplugin Frontend2 Template would help, this of course is for plugins and OpenCPN might be somewhat different.

If you got to VDR_pi github  https://github.com/rgleason/vdr_pi  click on green check, select the link for one of the androids

https://app.circleci.com/pipelines/github/rgleason/vdr_pi/396/workflows/c9a05909-5689-4ef7-ad39-164f7924b4c8/jobs/4046

https://app.circleci.com/pipelines/github/rgleason/vdr_pi/396/workflows/c9a05909-5689-4ef7-ad39-164f7924b4c8/jobs/4052

Then you can see how circleci builds for android.

The other way to learn more about building for android is to look at the Testplugin Frontend Template which we use to build all the TP based plugins.

I would look at the circleci/config.yml   and at the ci/ folder builds for android.
Perhaps that will help you figure out how to build for android.
https://github.com/jongough/testplugin_pi/blob/master/.circleci/config.yml

ci/ builds for android https://github.com/jongough/testplugin_pi/tree/master/ci
Also cmake folder https://github.com/jongough/testplugin_pi/tree/master/cmake

Its not simple, but perhaps this will help?