OpenCPN/OpenCPN #2085: OCPNTemplate-Recovery.conf should not exist

mgrouch, 2020-11-11:
Why would you have and maintain this file?
Defaults  in the code should be the ones which are safe.

Thanks

Hakansv, 2020-11-27:
What I know that's exactly what that file supports. If the user has changed things out of control and want to get back to defaults wo emptying charts list or similar as a deleted config would do. Just use the default template.

mgrouch, 2020-11-28:
Why would you trust user who broke the original configuration editing safe configuration too?

Hakansv, 2020-11-28:
My English is not at all native so it may be I misunderstand your message. But anyhow:
All users have different technical knowledge and level of interests. Most just want a good navigation tool for a safe and pleasant cruising. As they achieve more knowledge about OCPN's waste amount of options they may try to reach a for them suitable set up. During that process it's not strange if someone has forget about earlier changes and don't remember how to "come back". 
It can also be someone on board has set up a user template that won't fit another crew. Then the default template is a handy "way back".

mgrouch, 2020-11-28:
Exactly my point. Safe template shouldn't be up to a user. It's should be default built in into software.

rgleason, 2020-11-28:
It is to allow a user to go back to another configuration that works. BCN needed it to help new users trying to install charts and it was understood to be useful for other situations too.

closed 2021-04-10 by mgrouch