OpenCPN/OpenCPN #2088: ObenCPN 5.2.4 does not start on Mac

krokodil66, 2020-11-15:
Tried to update from 5.0.1 to 5.2. Tried OpenCPN_5.2.0+1.4d956e1 and OpenCPN_5.2.4+1.6b314e6. 
But both versions end up with the error: "Failed to initialise the user interface."

Even if I delete all the preferences etc. before, the same error occurs. The log file just contains a zero :-//.
Use macOS Catalina 10.15.7 incl. supplemental update. 
Hope there is a fix for that, because like OpenCPM very much ;-)). 

Regards, Armin

CarCode, 2020-11-15:
The download page promise that OpenCPN works with macOS 10.7 to 10.15.
However all OpenCPN versions with file extension *.pkg have problems or does not work.
Go back to version 5.0.1 until it might be fixed.
I doubt however OpenCPN will ever work with macOS 11.0.1 Big Sur.

krokodil66, 2020-11-15:
Installed OpenCPN half an hour before on another Mac with BigSur to see what's happen. OpenCPN started seamless ;-)). However did not do any testing to checks...

rgleason, 2021-04-23:
This is pretty old. Is it still a problem?

closed 2021-04-23 by nohal