OpenCPN/OpenCPN #2095: oeRNC and Flatpak

tai-fun, 2020-11-16:
Maybe not quite the right place....
There is a problem with the oeRNC plugin, using the plugin-manager, on Flatpak.
I have tested on both Kubuntu 20.04 and  Linux Mint 20. oeRNC does not show up as an option in either.

leamas, 2020-11-16:
Ack, I can reproduce this. Don't really have time to look into it right now, though.

leamas, 2020-11-16:
Still, just cannot avoid having a quick look. The problem seems to be that the oeRNC metadata contains the nowadays wrong value `<target>flatpak</target>` which should be `<target>flatpak-x86_64</target>`

The radar plugin suffers from the same problem. These seems to be  the only ones. 

May I suggest that you file downstream tickets at the oeRNC and radar plugins' repos? 

Note that this metadata will fail validation against ocpn-plugins.xsd (in the plugins project) for this reason.

rgleason, 2020-11-16:
leamas, I checked my version of testplugin_pi and  cmake/PluginSetup.cmake  has

https://github.com/rgleason/testplugin_pi/blob/bld-dep-yml/cmake/PluginSetup.cmake
line 45
```
message(STATUS "${CMLOC}OCPN_FLATPAK_CONFIG: ${OCPN_FLATPAK_CONFIG}, UNIX: ${UNIX}")
if(OCPN_FLATPAK_CONFIG OR OCPN_FLATPAK_BUILD)
    set(PKG_TARGET "flatpak")
    set(PKG_TARGET_VERSION "18.08") # As of flatpak/*yaml
elseif(MINGW)
    set(PKG_TARGET "mingw-x86_64")
    if(CMAKE_SYSTEM_VERSION)
        set(PKG_TARGET_VERSION ${CMAKE_SYSTEM_VERSION})
    else()
        set(PKG_TARGET_VERSION 10)
    endif()
elseif(MSVC)
    set(PKG_TARGET "msvc")
    if(CMAKE_SYSTEM_VERSION)
        set(PKG_TARGET_VERSION ${CMAKE_SYSTEM_VERSION})
    elseif(CMAKE_VS_WINDOWS_TARGET_PLATFORM_VERSION)
        set(PKG_TARGET_VERSION ${CMAKE_VS_WINDOWS_TARGET_PLATFORM_VERSION})
    else()
        set(PKG_TARGET_VERSION 10)
    endif()
```

Shouldn't Line 46 be
    set(PKG_TARGET "flatpak-x64_64")  ?

Note that line 50 has
    set(PKG_TARGET "mingw-x86_64")

rgleason, 2020-11-16:
What other recent changes are needed?

leamas, 2020-11-16:
BTW: As a walk-around to be able test oeRNC and radar plugins the following oneliner does the job:

    sed -i  '/target>flatpak</s/flatpak/flatpak-x86_64/' $HOME.var/app/org.opencpn.OpenCPN/config/opencpn/ocpn-plugins.xml

leamas, 2020-11-16:
@rgleason: please file a bug for testplugin at Jon's repo, and attach a PR with just this simple fix. Yes, it should be _flatpak-x86_64_

rgleason, 2020-11-16:
Leamas,  I have not heard from Jon for awhile, but he is making his way back to homeport.
I have cleaned up my master branch which largely matchs Jon's and made a PR. I have also cleaned up my new branch bld-dep-yml and am getting ready to merge it to my master with appropriate changes.

rgleason, 2020-11-16:
Is there anything else you know of other than stripping mingw symbols?

bdbcat, 2020-11-17:
@tai-fun 
oeRNC_PI for flatpak should be corrected by Version 1.2.3 currently in master catalog.

tai-fun, 2020-11-17:
@dsr <bdbcat@yahoo.com>
 Confirming that version 1.2.3 works on flatpak. Tested on Kubuntu 20.04
and Mint 20.

leamas, 2020-11-17:
Downstream bugs:

  - radar_pi: https://github.com/opencpn-radar-pi/radar_pi/issues/153
  - testplugin: https://github.com/jongough/testplugin_pi/issues/112

From my perspective: with these bugs filed and oeRNC working we could close this bug.

closed 2020-11-17 by tai-fun

leamas, 2020-11-17:
@tai-fun : BTW: Thanks for reporting, good catch!