OpenCPN/OpenCPN #2102: Plugin Manager  Improvements

rgleason, 2020-11-21:
FS#2710 - Plug-in manager on Windows: use ~AppData 
https://opencpn.org/flyspray/index.php?do=details&task_id=2710
I think this might be a good suggestion.

```
Use the same logic on WIndows than on Linux:

$HOME/.local/lib

$HOME/.local/share

$HOME/.opencpnplugins

by replacing $HOME with $HOME/AppData/Local

Not only being compliant with multi-user configuration control it would allow also user and the third party software to acces the files created along the way. For example, Docker cannot bind a file from \c\ProgramData but it can access any file in $HOME.

```

Also some other things to consider.
https://github.com/canne/dashboard_tactics_pi/issues/130#issuecomment-731645851

rgleason, 2020-11-25:
Not really being a programmer, I guess suggesting this is pretty bold.

Is there way to solve some of the issues regarding installation of plugins in various environments that would allow two different xsd validity checks to coexist?  Perhaps we could have the existing schema and a new schema informed by experience (benefited by looking back) with discussion and  analysis, ending with successful testing for each environment?  

The goal should be to have an easier time validating, installing and integrating properly working plugins into OpenCPN, in many different environments.  Alec has done noble job getting this system working and has been much like Peter and the dike fixing problems that have cropped up. In building the current system we've been learning a lot.  Is it possible to find some flexibility to be able to make some appropriate adjustments now?  

This effort would require some talented programmers to discuss and work together on plugin code, plugin frontend code, Plugin Manager and  perhaps even OpenCPN.

It  might solve some of the current recognized restrictions and problems, and would allow the plugins to migrate gradually over to a new schema when the new system is verified to be working.  It would also give us good method to transition the schema as necessary in the future!

I don't know enough about computer science theory and techniques to provide a name for this, so I that I could do a google search to learn more, but perhaps this "Issue" can be used to discuss and identify what could be done about this, and identify a way forward. 

I invite constructive thoughts, comments, and ideas about this.  And  I hope to learn more, so that we can improve the OpenCPN family of programs and the Plugin Installer.

bdbcat, 2020-11-25:
We could start by listing, in a bullet fashion, the problems with the current implementation.
We will then break this down into bite-sized pieces for discussion.

rgleason, 2020-12-25:
Completed by Jon and Dave

closed 2020-12-25 by rgleason