OpenCPN/OpenCPN #2103: Feature Request: Enable web services

rgleason, 2020-11-22:
FS#2715 - Enable web-services

CarCode, 2020-11-22:
I guess there are other things which are more importand. If OpenCPN should still be used for navigation there is a strong need to rewrite the GPS functions. Actually only the prefix $GP is supported (the US GNSS) however we have now $GA (the Galileo positioning system), $GB (the Beidou sytem), $GL (the Glonass system) and also $GN (a mixed GPS and Glonass).
Next: 
It is foreseen that OpenGL will not supported anymore by newer hardware. You know it was very bad supported by Windows. So it is time to think of a switch for something else. A candidat might be Vulkan.
Furthermore with the last updates of Apple (macOS Big Sur, M1 Arm Chip) OpenCPN will not run with newer Apple machines anymore.
I suggest starting soon otherwise it will be too late...

bc38, 2020-11-23:
I support!
In addition, realize smooth movement of the ship position using 5-10hz GNSS

CarCode, 2020-11-23:
> I support!
> In addition, realize smooth movement of the ship position using 5-10hz GNSS

Correct; my GPS device can be switched between 1Hz and 10Hz. The latter is actually not supported by OpenCPN.

rgleason, 2021-04-10:
Hakan has worked on this

closed 2021-04-10 by rgleason