OpenCPN/OpenCPN #2105: FreeBSD patches

dropenmind, 2020-11-24:
Hi,
I made patches in order Opencpn work completly on FreeBSD
How to pull them into the offical repo ?
Regards

leamas, 2020-11-29:
Well, the way is to make a one ore more pull requests to this repo.

That said, supporting FreeBSD is outside our current scope. This means that in order to have something useful for FreeBSD users someone (you?) needs to make a FreeBSD package of some kind, incorporating your patches. You would also need to consider the need for plugins, some of which handling DRM; these are basically impossible to build for FreeBSD today.

OTOH, opencpn is perfectly usable without the DRM-dependent plugins, especially in the US.  So if you make a downstream FreeBSD package you could start the process of upstreaming patches. I'd suggest that you start with small pull requests which you deem as generally useful rather than a long patch series.

rgleason, 2021-04-14:
Can this be closed?

rgleason, 2021-04-25:
Outside current scope

closed 2021-04-25 by rgleason