OpenCPN/OpenCPN #2106: Zoom stranges in non-quilt mode while auto follow is active

Hakansv, 2020-11-24:
The following refer to tests on Win10 and Debian Buster Openplotter (RPi4)
OpenGL on/off no difference
oeSENC or Raster is no difference.

When quilt mode is off and Auto follow is active the zoom behavior is not correct.
If I repeatedly  zoom out the canvass scale is only changed to value just below the actual chart scale. If a chart scale equals e.g. 1:50000 the zoom out stops at about a canvas scale 1:75000. For further zooming out nothing happens for a while. To zoom in again I've to repeatedly  press i.e. the "+" key a number of times. Probably counted to the same amount of strokes I earlier used to pass the "no reaction" limit.
If I continue to zoom out beyond the "no reaction" limit, by keys or mouse wheel, neither the canvas scale nor the view point are changed but eventually the chart view is drifting far sideways in an indefinite direction. The canvas scale is changed to a "random" value. 
To come back to a normal view I can for example toggle the Auto follow.

When repeatedly  zooming out at a non Auto follow mode the chart view is as normal i.e. the single chart is step by step contracted to a smaller square while the view center is not moved.

Hakansv, 2020-12-10:
Having looked further into this I understand it's a feature and not a bug.
When in AutoFollow mode the the single chart zoom function should not make the scale to small for the actual chart. Although I found the scale rules to conservative. Now a 1:50 000 chart admits a canvas scale of around 1:75 000. Two times would fit me more.
Comments before closure?

bdbcat, 2020-12-11:
The reason to bound the display scale to something reasonable is to conform to a wide range of PC performance capabilities.  No-one likes a sluggish system, attempting to render a chart at so small a scale as to be unusable. And especially critical in auto-follow mode, where one assumes we are underway in real navigation conditions.
It might be worthwhile to investigate the coupling this scale boundary with some variation of the options->chart zoom scale factor, so that it could be user adjustable.

Hakansv, 2020-12-11:
The intention to keep a usable scale while autofollow is fine. No doubts. The function to hinder zooming out beyond a certain point is then understood. But the scenario I've seen that zooming back in to a smaller scale is not possible, or troubled, after several attempts to zoom out to far. It's sometimes also locked to one specific not changeable scale. To break out from that a autofollow toggle will do. 
All this in single BSB chart mode and a offset position like "look ahead" only so probably seldom used. 
I'm trying to follow the somewhat complex DoCanvasUpdate() function and all calls to understand what's happening but it will take time for me. If you get a clear view where to check or fix please do or advise.
Thanks
Håkan

bdbcat, 2021-05-02:
I cannot reproduce this in the current github code base.  Using simple NOAA BSB charts, VDR file, auto-follow "on" with offset, and single chart mode.
Please confirm.

Hakansv, 2021-05-02:
I can't reproduce the "hanging" of the zoom function. Instead is the chart automatically zoomed back to a detailed level if I try to zoom out to much while in auto follow off set or look ahead mode.
But I think we should not care for further work on this. If someone would like to see e.g. the chart border the auto follow mode is not relevant so if further discussed the advice would be switch auto follow off. 
Also I haven't seen or heard any user comments about this so; thanks for your check. I'll close this.

closed 2021-05-02 by Hakansv