OpenCPN/OpenCPN #2110: GPS: Extend and rewrite functions for GN etc

rgleason, 2020-11-27:
https://github.com/OpenCPN/OpenCPN/issues#issuecomment-734657611

Hakansv, 2020-11-27:
re: "Actually only the prefix $GP is supported "
From what code is that statement formed?
Although I wouldn't call me a coder I can't verify that. When parsing NMEA we're not asking for any specific sender just the message ID.
One example:
if( m_NMEA0183.PreParse() )
    {
        wxString IDs[] = {_T("RMC"), _T("HDT"), _T("HDG"), _T("HDM"),
                          _T("VTG"), _T("GSV"), _T("GGA"), _T("GLL")};

CarCode, 2020-11-27:
@Hakansv 
Correct and that is just the problem.

Hakansv, 2020-11-27:
Hmm.. Confused!
What's the problem?  If we don't care about the sender everyone of the mention senders, GP/GA/GN/GB/GL are parsed if received.

rgleason, 2020-11-27:
I don't have a GPS   GA, GN, GB, or GL  device, but perhaps it should be 

_T("GG(some wildcard here in lieu of A"),

transmitterdan, 2020-11-27:
I think the “problem” is that the filter function cannot filter based on a particular sender code. So, if two senders with different ID strings send the same message type, there is no way to favor one sender over another.

This has been requested as an enhancement for some time. The general request is to be able to set a priority on which device ID should be used for ship’s position. Then, if the top priority ID stops sending then the next highest priority should be used and so on. 

This is harder to code than it sounds. Not impossible, but not trivial either. It requires a lot of rework to the NMEA multiplexer I suspect. 

Best regards,
Dan

Hakansv, 2020-11-27:
Dan
Then we're on another topic. Most interesting! 
So if we filter out RMC from a prio 2 connection and trust the RMC from a prio 3 connection we've to manually take out the filter if the prio 3 fails. Yes, that's an old problem. And probably not a simple solution. I agree.
Now I'm pondering..... What if we make the filter conditional? Like a condition for using the filter on the prio 2 connection. The condition would be that a (RMC)watchdog for prio 3 is not timed out. If so the (RMC) rejecting filter on the prio 2 connection will not be used.
This would be a challenged UI but may be a trace to a doable programming?

Hakansv, 2020-11-27:
Or... Make a watchdog on the RMC that has been active due to higher priority and reset the priority list if it disappears. It's may be better that to involve filters?

CarCode, 2020-11-28:
I propose to read the sources of gpsd a little bit. It seems to me the gpsd source is some more uptodate than the nmea/gps code in ocpn. For example you can get RMC not only from 2 physically different connections (devices) but also from one connection sending with different talker IDs like $GP, $GL, $GA and so on.

rgleason, 2020-11-28:
Thanks CarCode.

rgleason, 2020-12-01:
Regarding GPS functions, just for clarity, does OpenCPN support 
$GA (the Galileo positioning system),
$GB (the Beidou sytem),
$GL (the Glonass system) and also
$GN (a mixed GPS and Glonass).

or not?   I believe Carcode believes not.

CarCode, 2020-12-02:
According to NMEA 4.11 there are these talker IDs available:
|================================================================
|AB     | Independent AIS Base Station
|AD     | Dependent AIS Base Station
|AG     | Autopilot - General
|AI     | Mobile AIS Station
|AN     | AIS Aid to Navigation
|AP     | Autopilot - Magnetic
|AR     | AIS Receiving Station
|AT     | AIS Transmitting Station
|AX     | AIS Simplex Repeater
|BD     | BeiDou (China)
|BI     | Bilge System
|BN     | Bridge navigational watch alarm system
|CA     | Central Alarm
|CC     | Computer - Programmed Calculator (obsolete)
|CD     | Communications - Digital Selective Calling (DSC)
|CM     | Computer - Memory Data (obsolete)
|CR     | Data Receiver
|CS     | Communications - Satellite
|CT     | Communications - Radio-Telephone (MF/HF)
|CV     | Communications - Radio-Telephone (VHF)
|CX     | Communications - Scanning Receiver
|DE     | DECCA Navigation (obsolete)
|DF     | Direction Finder
|DM     | Velocity Sensor, Speed Log, Water, Magnetic
|DP     | Dynamiv Position
|DU     | Duplex repeater station
|EC     | Electronic Chart Display & Information System (ECDIS)
|EP     | Emergency Position Indicating Beacon (EPIRB)
|ER     | Engine Room Monitoring Systems
|FD     | Fire Door
|FS     | Fire Sprinkler
|GA     | Galileo Positioning System
|GB     | BeiDou (China)
|GI     | NavIC, IRNSS (India)
|GL     | GLONASS, according to IEIC 61162-1
|GN     | Combination of multiple satellite systems (NMEA 1083)
|GP     | Global Positioning System receiver
|GQ     | QZSS regional GPS augmentation system (Japan)
|HC     | Heading - Magnetic Compass
|HD     | Hull Door
|HE     | Heading - North Seeking Gyro
|HF     | Heading - Fluxgate
|HN     | Heading - Non North Seeking Gyro
|HS     | Hull Stress
|II     | Integrated Instrumentation
|IN     | Integrated Navigation
|JA     | Alarm and Monitoring
|JB     | Water Monitoring
|JC     | Power Management
|JD     | Propulsion Control
|JE     | Engine Control
|JF     | Propulsion Boiler
|JG     | Aux Boiler
|JH     | Engine Governor
|LA     | Loran A (obsolete)
|LC     | Loran C (obsolete)
|MP     | Microwave Positioning System (obsolete)
|MX     | Multiplexer
|NL     | Navigation light controller
|OM     | OMEGA Navigation System (obsolete)
|OS     | Distress Alarm System (obsolete)
|P      | Vendor specific
|QZ     | QZSS regional GPS augmentation system (Japan)
|RA     | RADAR and/or ARPA
|RB     | Record Book
|RC     | Propulsion Machinery
|RI     | Rudder Angle Indicator
|SA     | Physical Shore AUS Station
|SD     | Depth Sounder
|SG     | Steering Gear
|SN     | Electronic Positioning System, other/general
|SS     | Scanning Sounder
|ST     | Skytraq debug output
|TC     | Track Control
|TI     | Turn Rate Indicator
|TR     | TRANSIT Navigation System
|U#     | '#' is a digit 0 ... 9; User Configured
|UP     | Microprocessor controller
|VA     | VHF Data Exchange System (VDES), ASM
|VD     | Velocity Sensor, Doppler, other/general
|VM     | Velocity Sensor, Speed Log, Water, Magnetic
|VR     | Voyage Data recorder
|VS     | VHF Data Exchange System (VDES), Satellite
|VT     | VHF Data Exchange System (VDES), Terrestrial
|VW     | Velocity Sensor, Speed Log, Water, Mechanical
|WD     | Watertight Door
|WI     | Weather Instruments
|WL     | Water Level
|YC     | Transducer - Temperature (obsolete)
|YD     | Transducer - Displacement, Angular or Linear (obsolete)
|YF     | Transducer - Frequency (obsolete)
|YL     | Transducer - Level (obsolete)
|YP     | Transducer - Pressure (obsolete)
|YR     | Transducer - Flow Rate (obsolete)
|YT     | Transducer - Tachometer (obsolete)
|YV     | Transducer - Volume (obsolete)
|YX     | Transducer
|ZA     | Timekeeper - Atomic Clock
|ZC     | Timekeeper - Chronometer
|ZQ     | Timekeeper - Quartz
|ZV     | Timekeeper - Radio Update, WWV or WWVH
|================================================================

A lot of them are not supported by OpenCPN nor by the Dashboard plugin.

rgleason, 2020-12-02:
I don't have any of these non US gps, but if you had on you could simply use the nemaconverter to change the designation. That's not that familiar with users, but I believe it would work.

rgleason, 2021-03-05:
Should this be closed Hakan?

Hakansv, 2021-03-05:
In my opinion, yes!

The NMEA message GSV is from now on the only message where OCPN cares for the Talker Id. That's is to determine different GNSS system in view and used in the Dashboard instrument "GNSS Status".
The position fix as reported by e.g. RMC and GGA can be a mix of, "the best of", different GNSS system. According to the NMEA0183 standard then using the Talker Id "GN" but OCPN don't make any differences for either of used Talker Id.  GGA reports how many satellites are used for the position fix, i.e "Satellites in use". (Apart from the GSV "Satellites in view" divided per GNSS system.)

closed 2021-03-05 by rgleason