OpenCPN/OpenCPN #2114: Chart downloader should be a separate executable. Same is applicable to plugin downloader.

mgrouch, 2020-12-02:
Chart downloader should be a separate executable called by main OpenCPN program.
It should be made reusable in true Unix fashion and standalone documented command line program so 
that others could create scripts around it.

Same is applicable to plugin downloader.

Gui should just call those programs and analyze output.

bdbcat, 2020-12-02:
Thanks for the input.
Not really very practical in cross platform environment supported by OCPN.   By example, Windows, our largest user base, does not do this sort of thing very well.
But, as always, feel free to submit patches, forks, etc. exploring these ideas further.  We would be very interested in your implementation.

mgrouch, 2020-12-02:
What is the issue with Windows? What can't it do to support it?
Thanks

rgleason, 2020-12-02:
Things can be organized in many ways, there are at least a dozen way of doing any one thing. For you being able to run a script is paramount, perhaps others saw it another way. Because it is not configured the way you would have preferred does not necessarily mean it is wrong.  There are many points of view and many environments to support.

mgrouch, 2020-12-02:
And where did I say it’s wrong?
I only pointed areas of possible improvements

nohal, 2020-12-02:
I would think it is much easier to write scripts directly using the XML catalogs and `xmllint`/`xq` and `wget`/`curl` than implementing what you ask for (Which IMO is quite a gigantic effort for something that basically changes nothing for the vast majority of the users).

rgleason, 2020-12-02:
You did not say it was wrong, but sounded like a commandment...."should"

leamas, 2020-12-02:
>  Current implementation makes OpenCPN too monolithic and makes it harder to evolve further.

I think you have a point here. OTOH, I wonder if the idea to split it into separate executables is the right remedy. It might be too expensive compared to possible gains.

That said, the current codebase could certainly benefit from  creating more distinct interfaces between different parts. 

Bottom line: if someone makes patches which supports better, more defined interfaces between various parts it would be great. Still, evolution rather than revolution is perhaps a better bet in this case?

rgleason, 2020-12-02:
Chart Downloader is an internal plugin that integrates into the Opencpn interface. Pavel and TransmitterDan worked on it as a separate external plugin, how separate can you get?  You want a completely separate and distinct interface that is not integrated? What is the advantage I thought Pavel did a great job integrating it with existing chart features.

mgrouch, 2020-12-02:
As evolutionary approach what if 
OpenCPN could be started with some kind of —headless option as a daemon, without GUI and listen for
restful HTTP commands?

These commands could be 
loading charts, installing plugins,
reindexing charts, updating / backing up configs. 

Thanks

leamas, 2020-12-02:
> a separate external plugin, how separate can you get?

Given the plugin API : you could get much more separate. 

But agreed, chartdldr is probably not the best example on the overall situation with about 250 KLOC with minimal module isolation.

And while I see the problem, I still don't think splitting the executable is the way (and certainly not internal REST APIs; we need less complexity, not more). Plain old C++ object design can make things better without breaking too many eggs.

rgleason, 2020-12-02:
Yet there are many librarys that plugins also need to load. Bdbcat originally setup plugins to be much separate with a very small API.  The API has grown as necessary information to and from Opencpn is needed.

closed 2021-04-10 by mgrouch