OpenCPN/OpenCPN #2116: RPI 3B / Raspbian/64bits wrong architecture.

tai-fun, 2020-12-05:
A user raised this problem here, https://www.cruisersforum.com/forums/f134/rpi-no-plugin-to-install-243202.html#post3287816

From the log
13:42:18.621 ERROR dlunix.cpp:162 /home/pi/.local/lib/opencpn/libocpn_draw_pi.so: mauvaise classe ELF�:�ELFCLASS64
13:42:18.622 MESSAGE pluginmanager.cpp:2291 PlugInManager: Cannot load library: /home/pi/.local/lib/opencpn/libocpn_draw_pi.so

Looks like a config problem in the PluginManager.

leamas, 2020-12-06:
There are  multiple problems reported against the 64-bit arm builds, both raspbian and stock Debain, many of which reported by @mgrouch.  For the time being, the simple message is that we don't support 64-bit builds on ARM.

Perhaps we should file a tracker bug for 64-bit builds on ARM.

rgleason, 2020-12-06:
This is the first report we have had about this problem. This happens because we do not have the resources to test every environment built, so users become the testers.

I believe the particular plugin illustrating this is[ Jon Gough's  Ocpn_Draw ](https://github.com/jongough/ocpn_draw_pi)
and the Cloudsmith Deployment Repository is here
https://cloudsmith.io/~jon-gough/repos/ocpn_draw_pi-prod/packages/
and this is his main CS reposiitory  https://cloudsmith.io/~jon-gough/repos/

I expect that there will be interest in chasing down this problem, because [Testplugin_pi
](https://github.com/jongough/testplugin_pi) is the template for Frontend2 which has been used for many plugins.

closed 2021-04-10 by tai-fun