OpenCPN/OpenCPN #2121: ci: macos builds fails.

leamas, 2020-12-07:
The ci macos builds fails. This is related to some horrible hacks to reset the homebrew installation. I have code at hand in the shipdriver plugin which works, but cannot test it due to lack of travis credits.

leamas, 2020-12-07:
hm... Among other problems the download links for the pre-built macos build deps  seems broken. The link used in current master is  http://opencpn.navnux.org/build_deps/wx312B_opencpn50_macos109.tar.xz. As obvious by clicking on this link, it doesn't work any more, it returns some plain HTML instead of an archive.

@nohal: @bdbcat: Thoughts?

leamas, 2020-12-07:
BTW: What is included in this tarball?

nohal, 2020-12-07:
We are just right now migrating the downloads from the CDN that is ceasing the sponsorship program, so broken downloads will likely appear as every remaining request to http://openspn.navnux.org is now redirected to https://opencpn.org
The archive contains just what the name suggests, prebuilt dependencies compatible with the old macOS 10.9, which are required because Homebrew already dropped compatibility with it long time ago.

leamas, 2020-12-07:
Right, I see.

The 10.9 version is a problem also in the CI builds, it is not supported be either Apple nor CircleCI. How bad is the idea to move to 11.3?

nohal, 2020-12-07:
11.3 version of what? The current macOS BigSur is 11.0.1. AFAICT it should still be reasonably doable maintaining support for 10.10, which cuts off just very old hardware (like Dave's old Mac mini, if I remember correctly)
FWIW right now I don't seem to be able to produce a build of OpenCPN that would not crash on illegal instruction shortly after start even myself on the latest BigSur and Xcode 12.2, so we will probably have to give this stuff a little love soon anyway.

leamas, 2020-12-07:
Sorry, my macos understanding is rudimentary. I meant xcode 11.3 , corresponding (if I get it right) to macOS 10.14 Mojave.

nohal, 2020-12-07:
Yeah, it's kind of a mess, the compatibility is not directly influenced by the Xcode version, but by the minimal SDK version supported, as set by `MACOSX_DEPLOYMENT_TARGET` and the associated `MAC_OS_X_VERSION_MIN_REQUIRED` macro (or `-mmacosx-version-min=`)
Toolchain from Xcode 11.3 should still support the 10.9 target, the current 12 does not anymore.

leamas, 2020-12-07:
So, using  Xcode 11.3 +` MACOSX_DEPLOYMENT_TARGET=10.9 `should be OK? I have the impression it should make things a little easier.

nohal, 2020-12-07:
Yes, it should work without any problem.

leamas, 2020-12-07:
OK, thanks.  Good night!

leamas, 2020-12-07:
hm... before I got to bed: will the archive be available on the same link as before, or something else? If it's a new link could you please announce it here?

nohal, 2020-12-07:
It is currently available at https://download.opencpn.org/s/rwoCNGzx6G34tbC/download, so `curl -o wx312B_opencpn50_macos109.tar.xz https://download.opencpn.org/s/rwoCNGzx6G34tbC/download` should do.

bdbcat, 2020-12-08:
Download link fixed, but still fails on homebrew install.

leamas, 2020-12-08:
> Download link fixed,

Great!

> but still fails on homebrew install.

I'm on it, PR under way.  There are multiple problems, the links is just of them.

leamas, 2020-12-08:
> Download link fixed,

The appveyor builds also fails lookig for  http://opencpn.navnux.org/build_deps/nsis-3.04-setup.exe, needs an update...

Rasbats, 2020-12-08:
And the link for wxWidgets-3.1.2.7z as well please. (Needed for Appveyor)

jongough, 2020-12-08:
The macos plugin builds also need http://opencpn.navnux.org/build_deps/Packages.dmg which is not available. Is there a new location for that file?

nohal, 2020-12-08:
https://download.opencpn.org/s/SneCR3z9XM3aRc6/download

leamas, 2020-12-09:
@jongough: You can also install packages from brew, see ci/generic-build-macos.sh here (main OpenCPN).

leamas, 2020-12-10:
Fixed in #2050. Closing

closed 2020-12-10 by leamas