OpenCPN/OpenCPN #2125: Download links for build dependencies broken/updated.

leamas, 2020-12-09:
The links to download pre-built build dependencies are broken/changed after CDN closing their sponsorship program. Discussion started in #2121, but this is only about macOS and is about to be closed. Current status (?):

  - wxWidgets for macOS, wx312B_opencpn50_macos109.tar.xz, available at   https://download.opencpn.org/s/rwoCNGzx6G34tbC/download.
  - Packages.dmg for macOS can be installed using brew, see ci/generic-build-macos.sh in main OpenCPN.
  - wxWidgets for msWindows, available at TBD.

@nohal: Do you have the Windows missing link?

nohal, 2020-12-09:
I sure do, which file exactly?If it is wxWidgets-3.1.2.7z, the link is https://download.opencpn.org/s/E2p4nLDzeqx4SdX/download

leamas, 2020-12-09:
Yes, exactly. Thanks. 

BTW: The links are up & running after some outage.

leamas, 2020-12-09:
@Rasbats: ^

rgleason, 2020-12-09:
Alec, we need to add NSIS file location for appveyor.

leamas, 2020-12-09:
> Alec, we need to add NSIS file location for appveyor

???

bdbcat, 2020-12-09:
NSIS for Windows appveyor build:

https://download.opencpn.org/s/54HsBDLNzRZLL6i/download

rgleason, 2020-12-11:
If there are other files found that are vital build files for any environment here is a place for developers to list them (and to correct them as needed).
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:plugins:plugin_build_files

leamas, 2020-12-28:
This bug seems basically fto be fixed, but keeping it open and visibler for yet some time.

leamas, 2021-01-29:
time to close.

closed 2021-01-29 by leamas