OpenCPN/OpenCPN #2138: Now that gtk2 is dead can you start compiling arm32/64 OpenCPN with gtk3?

mgrouch, 2020-12-22:
Gtk2 is EOL now. No new development and code is frozen. 
And gtk2 version of OpenCPN on arm both 32/64 bit is buggy with touchscreen. Can we have official OpenCPN builds with gtk3 for arm?

Thanks

rgleason, 2020-12-22:
I still have to catch all of Sean's plugins up with the previous cycle.... I cannot keep up with these changes. Can this change get included with the current testplugin status, before I start updating?

leamas, 2020-12-22:
There is already an official Debian package supporting arm which is gtk3 only.  The problem is the plugins, and building for new platforms is a pure manpower problem. I'm sure Dave would happily accept a PR against the plugins project with new, gtk3 builds for arm.

mgrouch, 2021-02-17:
Gtk4 is out too. Might be better skip gtk3 and go straight gtk4.

rgleason, 2021-02-17:
It keeps moving. No problem. I can wait to update all Sean's plugins again, but it will be some time period maybe a month before everything gets figured out.

mgrouch, 2021-02-17:
https://developer.gnome.org/gtk4/unstable/gtk-migrating-3-to-4.html

Migration guide.

rgleason, 2021-02-18:
Oh really? Look at that huge list.  Maybe we should get real about this?  Just my opinion.

nohal, 2021-02-18:
@rgleason the above is completely irrelevant for you, ignore.

bdbcat, 2021-02-18:
@mgrouch 
The issue is this:
Our many, many production users depend on loading wxWidgets from a well known and trusted repository.  AFAIK, today there are wxWidgets builds for rPI available for gtk2 only.  No "official" wxWidgets for gtk3 that I can see, and certainly none for gtk4.
So far, we have resisted packaging (and maintaining) custom wxWidgets builds for OCPN on debian and derivatives.  I don't think that we want to start this now.
However, public forks of OCPN containing custom wxWidgets libraries (such as yours) are welcome, for the bleeding edge users who want to experiment.
Dave

leamas, 2021-02-18:
>  AFAIK, today there are wxWidgets builds for rPI available for gtk2 only

"rPI" is a bit ambiguous in this context.  I suppose you mean raspbian, which is used by most users (these days, standard Debian runs on rPI, and so does Ubuntu. But that's another story)

Current Raspbian is based on Buster which certainly has gtk3 packaged since long:

```
$ apt search libwxgtk3.0-gtk3-dev
libwxgtk3.0-gtk3-dev/stable,now 3.0.4+dfsg-8 armhf [installed]
  wxWidgets Cross-platform C++ GUI toolkit (GTK+ 3 development)
```

leamas, 2021-02-18:
The other part is  that we don't know where the rPI  ecosystem is heading. So far, Raspbian has had a firm grip on the market, due to historical reasons, graphics driver etc.

However, since the standard linux kernel now supports rPI there is now a different situation. Here is also a move  towards a full 64-bit solution (rPI 4 comes with 8 Gb), and we see this as attempts by @mgrouch and others to port OpenCPN to 64 bit Arm using standard Debian. I see the signs that we mightneed to start to build 64-bit plugins for some OS on Arm for some Linux distro. Hopefully just one, with Debian Bullseye as most likely alternative (?)

rgleason, 2021-02-18:
Pavel, you make a point, but some of these changes shuffle down to plugin builds, particularly gtk and wiring inevitably has to change for each plugin.

bdbcat, 2021-02-18:
`Current Raspbian is based on Buster which certainly has wxWidgets-gtk3 packaged since long:`
I stand corrected.
Still, the issue stands.  There is much fragmentation of linux bases used by the rPi community.  Difficult to know where to apply our limited resources.
And let us be clear (once again) that the technical point is this:  OCPN does not really depend on gtk itself.  OCPN depends on wxWidgets, which on linux depends on a particular gtk version.  So if the wxWidgets libs are available on the target system, then O will link and run.  Our "business" problem is that we need to support many legacy linux systems, with unknown wxWidgets versions available.  So a hard cut-over to a new wxWidgets  build is always problematic.

leamas, 2021-02-18:
Indeed. But  we will still need to move to gtk3 since any upcoming Raspbian version based on Bullseye (probably next) will drop support for gtk2. So we have no choice, this change is inevitable. Since Buster has been around for around 18 months it might possibly make sense to declare that next OpenCPN release on Arm will be using gtk3, thus creating an ABI break.

The 64-bit issue is of  the same kind, but more diffuse. Which will be the dominating 64-bit OS on rPI? Raspbian (64bit version is in beta), Debian, Ubuntu, ... We will have to make a decision, but it might be too early.

One way would of course be to enable flatpak aarch64 builds, and build plugins for this. Perhaps the most reasonable 64-bit option today. Using flatpak is the way to avoid fragmentation, for sure.

mgrouch, 2021-02-18:
Or might be have flatpack builds for supported platforms and use these flatpacks as the source to create native packages .deb .rpm, etc. That should be fast as no compilation is involved.

leamas, 2021-02-18:
> Or might be have flatpack builds for supported platforms and use these flatpacks as the source to create native packages .deb .rpm, etc. That should be fast as no compilation is involved.

Nope, wont work. Neither technical (different runtime env) nor administrative. 

The reason to use flatpak would  be to have one single package for all 64-bit aarch64 machines. If we still wan't to compile for different os'es we are back on square one. 

BTW: It's spelled flatpak :)

mgrouch, 2021-02-20:
I think circleCI is not a good place to build anything arm based.
TravisCI has builders for arm which are native. Without emulation.
What do you think of building arm stuff natively?

leamas, 2021-02-20:
Problem is that Travis is not free for open-source projects anymore, so it's kind of dead end. I have been looking at Shippable as an alternative, but nothing really done -- help appreciated!

mgrouch, 2021-02-20:
Launchpad supports daily automatic builds. Launchpad builds for arm are quite fast too. 

https://help.launchpad.net/Packaging/SourceBuilds

leamas, 2021-02-20:
But it doesn't work. Launchpad builds .deb packages, we need to build arbitrary tarballs.

If we could find a way to use the Debian/Launchpad/COPR/OBS whatever build chains it would be really interesting. I have thought about it,but  inconclusive.

mgrouch, 2021-02-20:
.deb packages can be converted somewhere else into tarballs

leamas, 2021-02-20:
yes... but that is a complicated way, creatiung a debian packaging for each and every plugin, add  the tarball generation to that, post-build jobs which install/downloads the debs, stores the tarball on something like cloudsmith, generates the metadata, builds the catalog PR. 

It *is* probably doable, but a complicated mess compared to the current solution where we just builds tarballs, uploads  and generates metadata on the fly.  My initial thought is that it's just not worth it.

mgrouch, 2021-02-20:
But launchpad can build on so many platforms and for so many Debian flavors. And it’s fast for arm builds. 
One can re-use Debian packaging files (most of O plugins had it) in the LaunchPad recipe. Daily recipe will trigger builds on changes in upstream. Once this setup it should be low maintenance.

leamas, 2021-02-20:
Well, as they say: patches welcome.  When it comes to Linux, the fragmentation is about Debian, so using the Ubuntu build farms makes sense (exception: non-free plugins like oesenc). 

Note that launchpad does not build for Debian, only Ubuntu. So there is one catch.

mgrouch, 2021-02-21:
I’ve played a bit with daily builds recipes on launchpad. It not that difficult to set up. The recipe is just 4 lines. the git repository with OpenCPN needs to be imported into launchpad (I did it as git but there is also bazaar option)
Then another repository with just Debian packaging files needs to be imported into launchpad. 
Launchpad will perform automatic syncs every 5 hours from those 2 repositories on GitHub. 

in the recipe you will have one 
merge statement 
And another nest-part to fetch debian directory. 

and then after importing the recipe into launchpad you can have daily builds for all Ubuntu flavors on all supported platforms. 

the maintenance will be only maintaining proper Debian packaging project for OpenCPN.

in launchpad you could check daily logs. If builds are successful it will automatically put results into daily PPA.

leamas, 2021-02-21:
Yes, I'm not not that surprised that these parts works without too much hassle. Now, the remaining work on this path if I get it right, out of the top of my head:

  - Create a Debian packaging which matches some kind of template plugin -- my favourite is [shipdriver](https://github.com/Rasbats/shipdriver_pi)
  -  Modify the packaging so it installs a tarball and metadata under some fixed locations like /usr/share/opencpn/tarballs, /usr/share/opencpn/metadata.
  - Add some kind of scripting which pulls the created .deb packages, unpacks them and retrieves tarball + metadata.
  - With the tarball and metadata available, jack into current workflows to publish the tarball and create a catalogue PR.

Everything here is certainly doable. The upside is the well-maintained Ubuntu build farm,  with a more or less complete Ubuntu versions / hardware matrix. The downside is that this will be a complicated setup which also creates long delays between pushing a commit and checking the result. Also, it doesn't really apply to Flatpak, Debian or non-free plugins like oesenc. And, not least, it would force plugin developers to sign the Ubuntu  Contributor Agreement -- not all want to do this.

Bottom line: I see the possibilities but I'm not convinced the result is worth the costs. That is certainly not to say I'm certain.

BTW: this is indeed interesting.

leamas, 2021-02-21:
Just for the sake of discussion: I don't really believe in the nightly  jobs  approach. It'sprobably better to stick to the traditional PPA workflow where developers creates a source package and uploads it. Developer knows when they want to build (typically not every night), and when it's time they don't want to wait for a nightly job. But, this is a detail in the bigger picture.

mgrouch, 2021-02-21:
The correct approach is to have continuous build (build triggered by every commit). This one is daily build which doesn’t get triggered if there was no changes (afaik). About 5 hours polling interval which is not that bad. 

Developers uploading into launchpad is too much effort for them. There are too many steps. With this all they need it to push into GitHub.

leamas, 2021-02-21:
Perhaps, perhaps not. But these are details. The bigger question is if the upside motivates the costs. I'm not convinced...

mgrouch, 2021-02-21:
With uploading to launch pad you have to repeat-yourself. Which violates principle of do-not-repeat-yourself. You have to upload a package for each distribution with manually edited Debian changes file. 
Here your Debian packaging work completes after a commit. 
And recipe does it for all Ubuntu distros on all platforms. 
Developer I do not think they need to sign Ubuntu agreement. Only the person who sets up recipe.

rgleason, 2021-02-21:
@mgrouch
@jongough
For all plugins that use Jon Gough's  frontend2, generally there are deb (linux), exe (windows), and macos packages created.   For example doing several queries of the Cloudsmith  OpenCPN Organization repository,
OpenCPN Organization  https://cloudsmith.io/~opencpn/repos/
Ocpn-Draw_pi https://cloudsmith.io/~opencpn/repos/ocpn_draw-prod/packages/?q=1.8.5.2-debian
Climatology_pi  https://cloudsmith.io/~opencpn/repos/climatology-prod/packages/?q=1.4.23.0-debian

If you want to setup launchpad with debian plugins, perhaps you can create a script to use these.  I might be useful to get a free opensource Cloudsmith account using your git login.

nohal, 2021-02-21:
Just my 2 cents, after I have created hundreds of PPA builds for both OpenCPN and many plugins over a period of several years:

1. Launchpad sucks, it is one of the most unintuitive tools I have used, ever.
2. Scripting automatic creation of the "debian directory" and upload to Launchpad is a not an issue (has been done like that the whole time I have been maintaining the OpenCPN PPA), adds next to no work to the developer once done and I have never felt violated by doing that except in the moments when hitting
3. "Debian directory" sucks (Fortunately usually just in the specific moments when the Debian maintainers "improve" things it contains and the underlying toolchain does and enforces for the new release) and having to maintain it just to fool Launchpad to produce something I can later repackage in a completely different way is the last thing I will ever do again.

rgleason, 2021-02-21:
Pavel, all your great efforts were hugely appreciated by many unknowing users!

mgrouch, 2021-02-21:
You haven’t even tried making recipe on launchpad. I agree that manual launchpad process if flawed. I’m talking about automatic one

nohal, 2021-02-21:
> You haven’t even tried making recipe on launchpad. I agree that manual launchpad process if flawed. I’m talking about automatic one

Oh yes, I did and it even did somehow work sometimes. Complete joke compared to any other CI system out there though.

But don't take me wrong, when I see a pull request implementing what you are suggesting and does actually work, I will of course give you big thumbs up and hit that merge button myself.

mgrouch, 2021-02-21:
Pull request not going to create a project on launchpad under OpenCPN. It’s not going to import GitHub repositories into launchpad either. recipe itself is just 4 lines. Once you set it up in launchpad (Right now it’s just ppa) I can make a pull request with those 4 lines if that helps.

nohal, 2021-02-21:
I'm personally afraid this is not how it can work. What on Earth is prohibiting you from demonstrating the (whole, doing something that actually solves a problem of course, not 4 lines of recipe "code") workflow on your clone as everybody else does?

But of course just show me the 4 lines if you are convinced they are enough to actually do all of it.

rgleason, 2021-02-21:
You could create a script that downloads the deb files to wherever, similar too this one for windows
https://github.com/OpenCPN/plugins/blob/master/download_xml_bash.sh
or this one for linux
https://github.com/OpenCPN/plugins/blob/master/download_xml.sh

These are just examples.

rgleason, 2021-02-22:
mgrouch   "The correct approach is to have continuous build (build triggered by every commit)."

Not so sure about that. For example I do not build locally for every environment, just windows. Thus other builds may fail and have to be worked on.  You may not want Launchpad to process everything all the time.

mgrouch, 2021-02-22:
> I'm personally afraid this is not how it can work. What on Earth is prohibiting you from demonstrating the (whole, doing something that actually solves a problem of course, not 4 lines of recipe [...]

Here is the demonstration. I've set up recipe in launchpad now.

https://code.launchpad.net/~bbn-projects/+recipe/bareboat-necessities-projects-daily

The recipe (to build OpenCPN trunk into .deb packages for all current Ubuntu versions on all platforms):

# git-build-recipe format 0.4 deb-version {debupstream}-0~{revtime}
lp:~bbn-projects/bareboat-necessities-projects/+git/OpenCPN master
nest-part debian lp:bareboat-necessities-projects debian debian debian/sid

Where 
lp:~bbn-projects/bareboat-necessities-projects/+git/OpenCPN master
is a periodic import of the Git repository at https://github.com/OpenCPN/OpenCPN

and lp:bareboat-necessities-projects  debian/sid
is a periodic import of the Git repository at the Git repository at https://gitlab.com/leamas/opencpn

Only debian directory is taken from it

The builds are failing now for the reasons of dependencies not specified correctly in 
https://gitlab.com/leamas/opencpn/debian

So this gives leamas a view of why the builds are failing and chance to correct debian packaging.

If there is some other code in git where I can pick up debian packaging files for opencpn I can try those as well.

mgrouch, 2021-02-22:
For example:

Bionic build
Missing build dependencies: debhelper-compat (= 12)
Clearly an issue debian packaging files. Does it really need debhelper 12?

Focal build
The following packages have unmet dependencies:
 builddeps:opencpn : Depends: libwxgtk-webview3.1-gtk3-dev but it is not installable

Groovy build (same as previous)
The following packages have unmet dependencies:
 builddeps:opencpn : Depends: libwxgtk-webview3.1-gtk3-dev but it is not installable

Hirsute (same issue):
The following packages have unmet dependencies:
 builddeps:opencpn : Depends: libwxgtk-webview3.1-gtk3-dev but it is not installable

nohal, 2021-02-22:
So now all we need is a few, I suppose trivial, pull requests fixing your findings. And then what?
BTW, there are OpenCPN DEB packages for both Bionic and Focal available from the official Launchpad PPA (= they were built on Launchpad and they do work, which is what we do expect from them), so I wonder why your recipe is not working for them (and anything newer with the same supposed issue).

Again, this is not saying that what you suggest is total crap, I just want to see it aligned with reality we live in.

And again, I am much more interested in seeing pull requests fixing issues than list of "this is broken, fix it somehow" items.

nohal, 2021-02-22:
To be more of a nitpicker, the "bugs" you are pointing out here all refer to https://gitlab.com/leamas/opencpn which is not even part of this project but a single Linux distribution, not being Ubuntu, specific packaging related repository.
I am more than convinced that it is *impossible* to both make the "debian directory" acceptable for Debian (sure not more than one specific release of it) and many Ubuntu releases at the same time, so problems like what you outline are as far as I'm concerned absolutely expected and probably not even resolvable by a trivial recipe I see.

nohal, 2021-02-22:
If you are interested how the PPA was maintained when it was done by me, you may have a look at https://github.com/nohal/launchpad
What I can tell about it is it sure was not perfect and I would have done it differently doing it now with the current CI tools available, but the amount of work it generated was still minutes per month and thus completely acceptable.

The "debian directory" of course can't be taken from there and applied to absolutely any DEB based distro and expected to be usable neither.

mgrouch, 2021-02-22:
So there is need for several branches for Debian packaging. These can be a separate project without OpenCPN code base. And one recipe per one of those branches. 

The end product which is for Debian based distributions is .deb package should be testable after each commit into OpenCPN git ideally. To be testable it at least need to exist (be built). Daily build doesn’t solve each commit validation but at least it solves issue of validation of “buildability” on all target Debian platforms*distros of one day of work done in OpenCPN.

mgrouch, 2021-02-22:
Where can I find most appropriate Debian directory with which OpenCPN master branch would build right now on most distros?
I’d like to set it up with recipes that it builds on all platforms/distros in launchpad.

mgrouch, 2021-02-22:
That Debian packaging needs to refer to latest released wxWidgets which is 3.0 (am I right?) and wxWidgets need to be setup to use latest available gtk3. There are some third party libraries which are copied into OpenCPN codebase (I guess for a reason) and it needs to use those to compile statically instead of pulling other versions of those. Is that how OpenCPN is built?

nohal, 2021-02-22:
There are at least two completely different beasts:
1. DEB acceptable in Debian
2. DEB working "everywhere" we want to (This is a unicorn that at times does not even exist), which is most likely totally unacceptable for Debian maintainers at any moment

For OpenCPN, we must be VERY conservative about any ABI changes as they will trigger the need to rebuild, repackage and redistribute all the plugins. And we have absolutley no control over the plugins.

We simply have no "take this, build it, it works" you keep asking for. We have "You change anything, you might break a lot of stuff". I don't know what "libraries copied into OpenCPN codebase" relevant to Linux you talk about - as far as I can recall everything we can expect to be packaged on Linux we detect and link dynamicaly if it is available.
On other platforms like Windows, macOS and Android, all of which we do support and have more users on them than on Linux, there are libraries that we must provide, build and link ourselves as they are not otherwise available there.

We have absolutely no control over which wxWidgets versions are available on which Linux distribution and it's version and what version of Gtk they are built against. And we do not care, if there is a problem that is reported, widespread and severe enough, we try to work it around. The workaround is NEVER providing our own wxWidgets or Gtk build.

mgrouch, 2021-02-22:
Looking at this part in CMake.txt
If you find gtk2 even if there is gtk3 on the system 
you would link against gtk2. It doesn’t look
right to me, given gtk2 is basically obsolete 
and you do support gtk3. 

if (NOT WIN32 AND NOT APPLE AND NOT QT_ANDROID)
  find_library(GTK2U NAMES wx_gtk2u_core)
  if (GTK2U)
    option(OCPN_FORCE_GTK3 "Force the build to use GTK3 [OFF]" "OFF")
  else ()
    option(OCPN_FORCE_GTK3 "Force the build to use GTK3 [ON]" "ON")
  endif ()
endif ()

nohal, 2021-02-22:
What does this break and on what platform? The purpose of this particular build option (and we just set a default for it here, still can be modified by the user running the build) was to work around a (really existing) problem on Linux systems where both wxWidgets linked against Gtk2 and Gtk3 could be present based on user choice and the result being a totally unpredictable mess. I *think* we could probably already remove it nowadays as that mess is not present in any current distro version.

It never ever disabled support for Gtk3 if it was present on the system, less when it was the only Gtk version present.

To be honest I would much prefer to talk about problems that do exist and need to be solved.

leamas, 2021-02-22:
hm... here are many open issues, at least from my point of view:

  - Is this about compiling main opencpn and/or the plugins for different platforms?
  - Where does the fact that the official 5.2.4 packaging is trickling down into upcoming Debian Bullseye and Ubuntu Hirsute fit into the picture?
  - Is this about creating packages which are accepted into Debian (more work, and not always possible) or just building them in a PPA (much  easier, but Ubuntu only)

My personal view is that building main opencpn is not really an issue. We have a well working CI setup which builds each and every change. And from the upcoming Debian/Ubuntu releases OpenCPN will be built for all Debian [[1]](https://packages.debian.org/bullseye/opencpn) and Ubuntu [[2]](https://launchpad.net/ubuntu/+source/opencpn/5.2.4+dfsg-1) platforms.
 
The bottleneck is the need for each new platform to maintain plugins. Today we use use CI setups, mostly on CircleCI for this. The idea to use Launchpad *could* fit into this puzzle, by building many but not all plugins for most Ubuntu version/hardware platform combinations. To repeat myself, this would benefit from the well maintained Ubuntu build farm,  but with some problems like overall complexity, non-free plugins (oesenc etc), Debian (doesn't work) and developer ideology + paperwork (Ubuntu Contribute Agreement, registered gpg keys etc.).

In the end, the upsides might not justify the costs. OTOH, like @nohal says:  a PR which makes it possible for a plugin like [shipdriver](https://github.com/Rasbats/shipdriver_pi) to use the Launchpad infrastructure to build for Ubuntu platforms would certainly  be interesting.

leamas, 2021-02-22:
But then again, the core issue here is the Linux fragmentation. When we now are about to enter the aarch64 Arm world, the most important piece is, again, to enable the main OpenCPN Flatpak aarch 64 build (simple) and add a plugin build chains for it.  This would ensure we have a working solution on all aarch64 platforms, and actually fight the fragmentation instead of just accepting it.

transmitterdan, 2021-02-22:
> Looking at this part in CMake.txt If you find gtk2 even if there is gtk3 on the system you would link against gtk2. It doesn’t look right to me, given gtk2 is basically obsolete and you do support [...]

We don't care about GTK flavors. We care about which GTK version does the system wxWidgets libraries use.  And wxWidgets supports both GTK2 and GTK+ to coexist on the same platform.  The builder has to manually run wx-config to set the desired toolkit (gtk2 or gtk3).  So far, CMake does not do that part so it is a manual step done by the person building before running cmake.

leamas, 2021-02-22:
@transmitterdan: the rest of story seem to be that, for Debian wxWidgets:

  - Bullseye only supports gtk3, gtk2 is dropped
  - Buster is the transition phase, supports both gtk2 and gtk3.
  - Stretch and before only supported gtk2.
 
These changes has trickled down to Ubuntu over time. Bionic supports gtk2 + gtk3, Focal is gtk3 only.

I still think building main opencpn for more platforms besides just enabling for example Flatpak/aarch64 is a non-issue.

transmitterdan, 2021-02-22:
@leamas,  You are right, but we should not drop GTK2 from CMakeLists.txt as that might break builds on some of these older platforms (e.g. buster, focal).  The way CMakeLists.txt is now is a bit of a hack but it allows building on most of the Ubuntu flavors.  Someday we can forget about gtk2.  But not for perhaps a year or more.

I also notice a lot more problems with gtk3 flavors of wxWidgets.  I can create random crashes with gtk3 on focal amd64 and buster Arm builds, but gtk2 has proven remarkably stable.  I hope gtk3 matures rapidly.  Also, performance under gtk3 is noticeably slower on focal.

leamas, 2021-02-22:
@transmitterdan: there are lot's of applications out there on gtk3 which are neither slow nor unstable. So the missing point is probably the wxWidgets gtk3 adaptation.

Re-tracking to the core issue here: there is no need to build to  create new builds of main OpenCPN using gtk3. They will be upon us with Ubuntu Hirsute and Debian Bullseye for all possible platforms. Also, the focal builds already uses gtk3, it's the only option there.

The problem is the Linux fragmentation and the corresponding need for multiple plugin builds.

leamas, 2021-02-22:
@transmitterdan: would of course be interesting if you could make some tests with the updated opencpn build in #2174, to see if we could have some faith in  upcoming wx 3.2. No plugins, though.

closed 2021-04-10 by mgrouch