OpenCPN/OpenCPN #2140: No Xmas greetings

leamas, 2020-12-23:
Project suffers from missing Xmas greetings. 

MERRY X-MAS, all of you out there!

Here, navigation is somewhat odd these days...
[image]

rgleason, 2020-12-23:
Merry Christmas Alec, Dave, Kathi, Hakan, Jon, Sean, Mike, Steven, Pavel and all the rest of our friends using, supporting and building OpenCPN.  Here in Mystic, CT we can see the famous Mystic Seaport across the river, and the weather preparing for a big blow on Thursday. Snow has been melting and we can now get out of the driveway.

leamas, 2020-12-25:
Greetings has been exchanged in private messages (thanks!). and Chrismas is now officially upon us.   Closing bug :)

closed 2020-12-25 by leamas