OpenCPN/OpenCPN #2142: Betat catalog not accesssible

leamas, 2020-12-27:
As heading says: the beta catalog is not accessible from the GUI which just says *Cannot download data from URL* whne trying to update the beta catalog.

The culprit seems to be the the code uses a link like https://raw.githubusercontent.com/OpenCPN/plugins/beta/ocpn-plugins.xml. The correct link should replace beta -> Beta

Hakansv, 2020-12-28:
Fixed by #2143

rgleason, 2020-12-28:
I thought Dave had disabled Beta early on to simplify thing.

Hakansv, 2020-12-28:
Apparently not. But access to other catalogs then master needs a config tweak, Slightly excessive if you ask me. But there are more people to ask.

leamas, 2020-12-28:
>  Slightly excessive if you ask me. 

Agrred. But this is a decision Dave has made, and I really appreciate when things like these occasionally get a definite solution, stopping otherwise endless discussions.

leamas, 2020-12-28:
Anyway, the reported  problems is fixed. Closing.

closed 2020-12-28 by leamas