OpenCPN/OpenCPN #2145: Deployment solution: Requirements.

leamas, 2020-12-28:
The purpose of this bug is to try to discuss the need for some infrastructure to make the flatpak nightly  builds and the plugin downloader  (#1865) available.

The nightly builds are used in two contexts:

  - By bleeding edge end users, for example after filing a bug and trying to verify a fix in master branch.
  - By plugin build infrastructure. Plugins building a flatpak tarball links to the master branch by downloading the nightly build flatpak opencpn.

The plugin downloader needs a deployment solution which includes rebuilding the npm application locally since the server web address is a build parameter.

It is hard to deploy the nightly builds on current infrastructure on top of my wardrobe, creating problems described in [OpenCPN/Plugins#295](https://github.com/OpenCPN/plugins/issues/295#).  Likewise, the deployment issues described in #1865 has blocked the downloader.

Without going into implementation which if possible should be kept out of this issue:  any thoughts on the situation as described here, and the related requirements?

rgleason, 2020-12-30:
Lee Skillan has moved the Flatpak ahead
See  https://trello.com/c/BXjGt1BO/181-flatpak-packages  and please Vote!!

>  lskillen moved the card [Flatpak Packages](https://trello.com/c/BXjGt1BO/181-flatpak-packages) to Next Few Months on [Cloudsmith Roadmap](https://trello.com/b/IhvGx4aN/cloudsmith-roadmap)

Thank you Lee.

and vote!

lskillen, 2020-12-30:
You could alternatively just package as deb and rpm instead, which would work today. Flatpak (and snaps) are great, but there are advantages to keeping it simple too. If you don't want to worry about distribution infrastructure, then Cloudsmith would already have you covered, _today_. Do your users demand the vector in which the package is delivered to be flatpaks? It's completely upto you folks, though. We'll still consider prioritisation since our goal is universal compatibility with tooling. No matter what, good luck with it! :+1:

leamas, 2020-12-30:
well... Our goal is absolutely not "universal compatibility", it 's more about most bang for the bucks, bucks in the form of work, bang in the form of user and developer support. 

Supporting rpms means supporting at least fedora and probably some more. Fedora is hard given it's 6 months cycle. In other words, a lot of bucks for a quite little bang (Fedora is a smaller community).

The idea with the .deb/flatpak approach is that we have something for Debian/ubuntu, the largest Linux user community. And flatpak, which to some extent evolves around Fedora,  for the rest. In this context, flatpak is a lot pf bang for the bucks.

Also note that there are no problems to deploy released versions (or even betas) over flathub.org which is a very stable and user-friendly platform. Likewise, cloudsmith works just fine for distributing the flatpak plugins, which are ordinary tarballs (and *not* flatpak extensions, which is a different beast). The issue we discuss is how to make the nightly  development builds available for users.  I'm sure we can solve it somehow, we already have an unstable version up & running. 

Did I mention that Fedora is my primary platform?

rgleason, 2021-02-12:
Isn't this worth voting for?
https://trello.com/c/BXjGt1BO/181-flatpak-packages

leamas, 2021-04-23:
This should be moved to Discussions, but I cannot do that. If anyone else can, please do.

rgleason, 2021-04-23:
Move to discussions "Show & Tell" topic?

closed 2021-04-23 by bdbcat