OpenCPN/OpenCPN #2147: master.zip supplied to build android armhf & arm64 fails with arm64

jongough, 2021-01-03:
I am attempting to building ocpn_draw_pi for android arm64 but it fails in one of the provided modules in master.zip:
```
In file included from /home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include/wx/fileconf.h:22:
/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include/wx/filename.h:473:43: error: use of overloaded operator '[]' is ambiguous (with operand types 'wxString' and 'unsigned int')
        { return GetPathSeparators(format)[0u]; }
                 ~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include/wx/string.h:1440:18: note: candidate function
    wxUniCharRef operator[](int n)
                 ^
/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include/wx/string.h:1442:18: note: candidate function
    wxUniCharRef operator[](long n)
                 ^
/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include/wx/string.h:1444:18: note: candidate function
    wxUniCharRef operator[](size_t n)
                 ^
/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include/wx/string.h:1428:15: note: candidate function
    wxUniChar operator[](int n) const
              ^
/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include/wx/string.h:1430:15: note: candidate function
    wxUniChar operator[](long n) const
              ^
/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include/wx/string.h:1432:15: note: candidate function
    wxUniChar operator[](size_t n) const
              ^
/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include/wx/filename.h:473:43: note: built-in candidate operator[](const char *, long)
        { return GetPathSeparators(format)[0u]; }
                                          ^
```
Is there a solution to this issue?

leamas, 2021-01-03:
Not tested, but perhaps `GetPathSeparators(format).GetChar(0)` could make the trick.

bdbcat, 2021-01-03:
May we see a link to the build log?

leamas, 2021-01-03:
Oops, I missed that this is about wxWidgets core libraries. Forget what I wrote...

jongough, 2021-01-03:
[https://app.circleci.com/pipelines/github/jongough/ocpn_draw_pi/221/workflows/cf22ede1-597b-4a7b-899a-b20274d30a3e/jobs/2048](https://app.circleci.com/pipelines/github/jongough/ocpn_draw_pi/221/workflows/cf22ede1-597b-4a7b-899a-b20274d30a3e/jobs/2048)

leamas, 2021-01-03:
hm... That happens in /wx/filename.h after including it. I can see two possibilities:
  
  - The context (included from fileconf.h) is odd and not tested. Easily checked by including wx/filename.h before wx/fileconf.h.
  - If this also fails  wxWidgets  seems broken. What version is it?

leamas, 2021-01-03:
Perhaps a sidenote, but the build flags looks inconsistent:  `-DANDROID -DARMHF `. You are trying to build a 64-bot aarch64 build, right? Then ARMHF seems wrong since it's basically a 32-bit architecture

jongough, 2021-01-03:
I added wx/filename.h before wx/fileconf.h and changed the definition from -DARMHF to -DARM64 but it did not affect the outcome
```
/opt/android/android-ndk-r20/toolchains/llvm/prebuilt/linux-x86_64/bin/aarch64-linux-android21-clang++  -DANDROID -DARM64 -DOCPN_USE_WRAPPER -DQT_WIDGETS_LIB -DTIXML_USE_STL -DUSE_ANDROID_GLES2 -DUSE_GLSL -DUSE_GLU_TESS -DUSE_S57 -D__OCPN__ANDROID__ -D__WXQT__ -DocpnUSE_GL -DocpnUSE_GLES -Docpn_draw_pi_EXPORTS -I/home/circleci/project/ocpnsrc/bitmaps -I/home/circleci/project/ocpnsrc/lz4 -I/home/circleci/project/ocpninclude/GL -I/home/circleci/project/ocpninclude/wxcurl/include -I/home/circleci/project/ocpninclude/wxcurl -I/home/circleci/project/extinclude -I/home/circleci/project/extsrc/glshim/include -I/home/circleci/project/extinclude/android -I/home/circleci/project/home/circleci/project/extsrc/glshim/include/GLES -I/home/circleci/project/ocpnsrc/glu/include -I/home/circleci/project/ocpninclude -I/home/circleci/project/extsrc/mygdal -I/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include -I/home/circleci/project/build/CMakeFiles/include -I/home/circleci/project/include -I/home/circleci/project/src -I/home/circleci/project/extsrc/glshim/include/GLES -I/home/circleci/project/OCPNAndroidCommon-master/qt5/build_arm64_O3/qtbase/include -I/home/circleci/project/OCPNAndroidCommon-master/qt5/build_arm64_O3/qtbase/include/QtCore -I/home/circleci/project/OCPNAndroidCommon-master/qt5/build_arm64_O3/qtbase/include/QtWidgets -I/home/circleci/project/OCPNAndroidCommon-master/qt5/build_arm64_O3/qtbase/include/QtGui -I/home/circleci/project/OCPNAndroidCommon-master/qt5/build_arm64_O3/qtbase/include/QtOpenGL -I/home/circleci/project/OCPNAndroidCommon-master/qt5/build_arm64_O3/qtbase/include/QtTest -I/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/libarm64/wx/include/arm-linux-androideabi-qt-unicode-static-3.1 -I/home/circleci/project/extsrc/tinyxml  -pthread -fPIC -O2 -g   -Wall -Wno-unused -fexceptions -rdynamic -fvisibility=hidden  -fno-strict-aliasing  -O2 -s  -DPREFIX=\"/usr/local\"  -s -Wno-inconsistent-missing-override -Wno-potentially-evaluated-expression -Wno-inconsistent-missing-override -Wno-potentially-evaluated-expression -Wno-overloaded-virtual -Wno-unused-command-line-argument -Wno-unknown-pragmas -O3 -fPIC -o CMakeFiles/ocpn_draw_pi.dir/src/Boundary.cpp.o -c /home/circleci/project/src/Boundary.cpp
In file included from /home/circleci/project/src/Boundary.cpp:27:
In file included from /home/circleci/project/include/ocpn_draw_pi.h:152:
/home/circleci/project/OCPNAndroidCommon-master/wxWidgets/include/wx/filename.h:473:43: error: use of overloaded operator '[]' is ambiguous (with operand types 'wxString' and 'unsigned int')
        { return GetPathSeparators(format)[0u]; }
                 ~~~~~~~~~~~~~~~~~~~~~~~~~^~~
```

leamas, 2021-01-03:
Right... let's face it, building for arm64 isn't easy. We have not been able to close #2081 for regular 32-bit arm builds, and the 64-bit builds are certainly not easier. 

That said, not being able to include wx/filename.h or, more general, not being able to compile anything indicates some build setup problems. After all, it is certainly possible to include this  file in other builds.

It looks like the wxWidgets setup is hardcoded rather than retrieved using wx-config, right? This might the culprit, as well as other parts of the hardcoded setup.

Sorry to not be that helpful...

jongough, 2021-01-03:
I have just checked and oesenc_pi has the same code as is being used here so if oesenc_pi is building correctly for android arm64 I don't know what the difference is.

bdbcat, 2021-01-04:
jon...
My point exactly.  My plugins are building for Android/64 without error.
I'll try to study the logfile....

bdbcat, 2021-01-04:
This seems to a be a semi-known issue, though I am not sure why my builds do not reflect the problem.
See:  https://stackoverflow.com/questions/23600736/cant-build-wxwidgets-on-cygwin-ambiguous-overload-for-operator-basedll-ape
I have corrected the OCPNAndroidCommon repo as recommended.
Please retry your job.

jongough, 2021-01-04:
I am building locally and I thought I better try downloading master.zip again (I don't normally as my internet is too slow) in case mine was out of date. It was the new one is 4 chars bigger and with that I do not get the issue in the wxWidgets code. 

Thanks.

It has now started turning up in OD code which it didn't before, so I am applying the same 'fix'.

leamas, 2021-01-04:
> I have corrected the OCPNAndroidCommon repo as recommended.

Have you reported it upstream? The corresponding, old bug is considered fixed: http://trac.wxwidgets.org/ticket/16130

jongough, 2021-01-04:
I haven't reported it, been too busy trying to get things going. I would like to close this issue now, as for ODraw the android arm64 the issue is fixed. However, I am not sure how the contents of master.zip is maintained or by whom. I know Dave changed it to make it work, but is he the owner of it or is there another group?

leamas, 2021-01-04:
The wxWidgets sources are owned by the wxWidgets project, for sure. And they are in the final push for releasing 3.2, so it's a little bit urgent to upstream bugs to them (using patched upstream sources is bad).

OTOH, since Dave did the patch it seems reasonable that he takes care of that part.

bdbcat, 2021-01-04:
OCPNAndroidCommon is a custom build of wx and Qt specifically to support OCPN on Android.  It was created to support CI plugin builds for OCPN Android.  It is based on an older wx release, known to be relatively stable, and sufficiently modern for OCPN Android. I maintain it as required.
Soon, OCPN Android will transition to wx3.2..  This will be disruptive, but there is no particular hurry.

CarCode, 2021-01-04:
Huh? Another time using its own custom build? Not a good idea!

bdbcat, 2021-01-04:
There **is** no unmodified wx source build that supports Qt and Android.  AFAICT, we are the only production app that uses this environment. So we now, and probably always, will need a custom build of wx for Android.

CarCode, 2021-01-04:
So then something is wrong with your code...

jongough, 2021-01-05:
Which code is wrong? If you know how to fix it can you please provide a pull request to either OpenCPN or testplugin_pi with the changes needed for cmake to work correctly without the need for master.zip on Android armhf and arm64.

CarCode, 2021-01-05:
> Which code is wrong? If you know how to fix it can you please provide a pull request to either OpenCPN or testplugin_pi with the changes needed for cmake to work correctly without the need for master.zip on Android armhf and arm64.

You might not do or want to understand: If someone need to change wxWidgets making its own wxWidgets then either his code is wrong or he don't want to discuss his problem with the wxWidgets owners. But why?

leamas, 2021-01-05:
> You might not do or want to understand:

We don't have an explicit code of conduct in this project. But I'd say there is an implicit one.  Perhaps one central  part of the Fedora code states it in a reasonable way (other projects have similar rules):

>Be respectful. Not all of us will agree all the time, but disagreement is no excuse for poor behavior and poor manners. We might all experience some frustration now and then, but we cannot allow that frustration to turn into a personal attack. It’s important to remember that a community where people feel uncomfortable or threatened is not a productive one. Members of the Fedora community should be respectful when dealing with other contributors as well as with people outside the Fedora community and with users of Fedora.

I actually think you have a point, but your way making personal attacks obscures it.

jongough, 2021-01-05:
I still don't know another path to getting android working without using the provided master.zip. If someone can provide the cmake code to ensure all the correct bits of wxWidgets for the android platform are built/available then I can update my system to use that. 

ODraw builds fine, it is just that the clang compiler did not like one of the wxWidgets provided bits of code. So I think it is wxWidgets that needs to change to be fully compatible with the android platform. However, the issue only occurs some of the time as some other plugins build OK, including mine, but ODraw did something/used something that caused clang to issue the error. I have no idea what or why.

leamas, 2021-01-06:
> I still don't know another path to getting android working without using the provided master.zip

Fair enough. Since we are at the bleeding edge, it's natural that we find bugs in wxWidgets and applies patches to handle them. The discussion is (should be) about how to get these patches sent upstream so we don't diverge from wxWidgets development. Having to maintain a wxWidgets fork is a bad place to be in...

transmitterdan, 2021-01-06:
I work a little with the wxWidgets guys.  I have a current fork and can submit PRs to wxWidgets.  Can anyone spike out exactly the changes needed?  I'm happy to do all the tests they make you do before submitting a PR.

leamas, 2021-01-06:
@bdbcat : ^

bdbcat, 2021-01-06:
tdan...
We can try.
In OCPN branch "sysroot", you will find "buildandroid/patches/wxWidgets".  This is a set of patches to the current wxWidgets github source, as would be required for any Android app using the most current wxQt port.
Unfortunately, the patches required for the current OCPN/Android build are much more extensive, and have been overtaken by events in the wx sources.  So best to move ahead, and focus on the coming release of wx3.2. 
So, to reiterate, these patches are not used verbatim in the current OCPN Android build.  They are in preparation as we migrate OCPN for Android to wx3.2, whenever that happens.  Once they are accepted and applied, the OCPN base and plugins will hopefully build cleanly from the latest wxWidgets github source using a simple wxconfig invocation.

Thanks for taking this on.
Dave

bdbcat, 2021-01-06:
To amplify on the above, a little...
When wx3.2 is released, I plan to update the OCPN/Android build to use that version.
This will be disruptive.  All OA plugins will need rebuilding.  This will involve modifications to the CI build scripts, among other things.  We will cross that bridge when 3.2 arrives.
Meanwhile, the OA plugins are building with a custom wxWidgets/Qt blob (master.zip), which I maintain.  I'll fix bugs as the arise, as was the case in this thread.
So let us carry on as we are today with the OA plugins, and prepare for the migration in the background.

leamas, 2021-01-06:
On a sidenote: there is a need to update the flatpak runtime, also disruptive. I have been waiting to do this at the same time as we do the wx 3.2 update. For flatpak, this also means updating from  gtk2 -> gtk3.

transmitterdan, 2021-01-09:
@bdbcat 

I am reviewing all the patches now.  I have run all the CI builds and did not break any CI targets when patches were applied to the master branch.  So some good news there.

I have some observations/issues/questions in no certain order:

- Item #1:

`void wxMenuItem::SetFont(const wxFont& font)` - This is advertised in the docs as only available under wxMSW port.  I realize the value in adding it here but I don't think the wx people will accept this lone setter.

- Item #2:

In the function `bool wxGLApp::InitGLVisual(const int *attribList)` there are about a dozen blank lines.  I don't know what that is for but probably violates some coding standard.  I will delete if no objection.

- Item #3:

The new PanGesterRecognizer implements a "smarter" pan recognizer.  What is the possibility other qt platforms will not work with this implementation?  Also, I presume the added code in `wxQtScrollArea::event` is to support this pan recognizer, correct?

- Item #4:

The new destructor for `class wxGLCanvas` has some trailing commented out code.  I guess this is against the coding standard.  Is it ok to remove it altogether?

- Item #5:

In `wxGLCanvas::ConvertWXAttrsToQtGL` multisampling is turned on.  Is this appropriate for other non-Android qt targets?

bdbcat, 2021-01-10:
TDan...
Thanks for looking at this.
#1  wxMenuItem::Set Font() is trivial, and quite useful to the wxQt port, so why not do it?  Has nothing to do with any other port.  Because it is not documented (yet) for this port, or not available in other linux ports, is not a good reason not to do it, IMHO.
#2. OK.
#3. This patch acts as a "bridge", to connect the well understood Qt gesture system into the new wxWidgets gesture event management, as introduced in wx 3.1.1.  It has no specific Android dependency, so should work on all wxQt platforms using gesture capable hardware. Leverages the inherent Qt ability to replace gesture recognizers as desired.
#4. OK.
#5. Not sure.  But this patch, and upstream access to the function, could be improved by using the class wxGLAttributes, instead of the older int *wxattr construction.  Call that a TODO.

To provide more talking points, I make the following observations.
The wxqt port is is incomplete, missing large sections, and plain wrong in some of the classes that are implemented.  https://wiki.wxwidgets.org/WxQt/Status
Further, there seems to be very little general interest in the port, at least as gauged by google juice.  In its current state, it is, however, a reasonable bridge from a generic wx app to a functional Android equivalent.  And that is why OCPN/Android uses it.
We offer these patches in the spirit of a) making our wxqt/Android port work better, and b) moving the wxqt port ahead in terms of functionality and stability.
The patches are non-breaking to the wx build process.  If they break some existing wxqt apps (other than OCPN), then we shall all discover the wider level of interest in the wxqt port, and adjust accordingly.

Thanks again 
Dave

transmitterdan, 2021-01-14:
I have started the process with wxWidgets folks.  Will update here any progress or problems.

transmitterdan, 2021-01-17:
@bdbcat, I have made a PR targeting the upcoming 3.2 release. I got back some good comments and questions as expected.  I am handling all the style and formatting issues raised.  There are two questions that I cannot answer.  See if you have some comments re these issues raised by wx maintainer.

```
In src/qt/glcanvas.cpp:

> +    format.setSampleBuffers( true );

+    format.setSamples( 4 );

This is normally defined by the attributes used, why should we hard code this here?
```
and

```
In src/qt/window.cpp:

>  +        // Delete QWidget when control return to event loop (safer)

 +        m_qtWindow->deleteLater();

I'm really not sure about this change. It seems wrong to have to do it, normally we shouldn't get any events after the window is destroyed. Doing this probably means that the sequence of events is not the same when using wxQt and in all the other ports.

Are you sure this is the right solution? Do we have a simple example showing what goes wrong without this commit?

```

Also, I don't have a way to test the final PR so will come back to you and ask that you test my wx branch containing all the final approved fixes.  In other words, use my wxWidgets branch that will be the final PR in place of your patches.

bdbcat, 2021-01-18:
tDan...
I'm looking at these comments now.  They seem reasonable. Some of the patch contents are legacy cruft, no longer necessary for modern wxWidgets. I should have clean patches in a day or so, after some device testing.
I can test against your wx branch, no problem.

transmitterdan, 2021-01-18:
@bdbcat ...
There is a branch called qtPatches in https://github.com/transmitterdan/wxWidgets which has been offered as a PR. It has passed automated review against wxWidgets master.  The CI workflow does minimal testing on Qt and nothing for Android I think.  @vadz has made initial review and you can see the conversation @ https://github.com/wxWidgets/wxWidgets/pull/2168 ... I made quite a few formatting changes per his request.  I think that is done. He is willing to accept the PR if we can deal with the few open questions.

bdbcat, 2021-01-19:
@transmitterdan ...
I have forked your repo, and submitted a PR to qtPatches.  Some redundancies here, but should merge cleanly.
I corrected one issue, related to hard-coding of wxglCanvas attributes.  Also fixed a recent commit that broke the build on Android, related to GetHostByAddr(). Still working with the second issue involving wxWindow DTOR.  Otherwise, except for these three issues, the branch builds OK and runs fine on Android, minimally tested.
Let me know if you need anything else.

rgleason, 2021-04-22:
Does this work now?

jongough, 2021-04-26:
The current master.zip file works.

closed 2021-04-26 by jongough