OpenCPN/OpenCPN #2148: Mac version OpenCPN 5.2.4 (and versions before) crash during scrolling in maps

awbakker, 2021-01-06:
With latest and also previous release I experience full crash of OpenCPN during scrolling/moving/zooming through the maps. I use O-charts. Same happens with using free ENC charts from Dutch government. It happens at both my mac mini and my mac pro laptop. Especially when scrolling a bit fast crashes occur quicker.

What happens is the following:

1) OpenCPN 5.2.4 is starting normally 
2) The maps dialog is shown[image]

3) then after some time of scrolling zooming the following dialog appears
[image]

4) inspecting the security dialog indicates that OpenCPN should have access (I tried to remove and give access a number of times, no improvement)
[image]

5) OpenCPN then suddenly disappears without any dialog - crashes

Any help would be appreciated as this keeps me from using it and I have become very enthousiastic about the general setup. On a windows 10 install I have no issues at all.

CarCode, 2021-01-07:
Try to use for macOS an OpenCPN version before 5.x such as 4.8.8. The later ones are all broken somehow.

awbakker, 2021-01-07:
Thanks, where can I find this older version. On the opencpn website I cannot find it and the link to older versions does not work. Here I find only source code.

nohal, 2021-01-07:
The "suggestion" you got from @CarCode is a total nonsense, do not waste time on it.

nohal, 2021-01-07:
FWIW, I'm unable to reproduce your problem, so please try to find some minimal reproduction scenario starting with all the plugins disabled and only the Dutch IENC charts.

CarCode, 2021-01-07:
> The "suggestion" you got from @CarCode is a total nonsense, do not waste time on it.

Maybe you have forgotten your own error messages:
https://github.com/OpenCPN/OpenCPN/issues/2131

nohal, 2021-01-07:
Maybe you have forgotten to read. This issue talks about a problem with FUTURE version of wxWidgets and is in no way supporting your, as usually false, claims. Please leave and stop wasting everybody's time if yu don't have anything relevant to say.

CarCode, 2021-01-10:
> Maybe you have forgotten to read. This issue talks about a problem with FUTURE version of wxWidgets and is in no way supporting your, as usually false, claims. Please leave and stop wasting everybody's time if you don't have anything relevant to say.

Yes, of course, sorry, exuse me. It is all fake news and a waste of time. Please tell this to other macOS users too.
I wonder however why the flyspray seems to be abandoned and nobody cares about all the claims there.
There are folders of old OpenCPN versions up to 5.2.0 but not for the actual version 5.2.4.
In the forum there are claims to be found as well such as:
https://www.cruisersforum.com/forums/f134/cant-install-opencpn-5-2-on-mac-10-13-a-240701.html
and
https://opencpn.org/flyspray/index.php?do=details&task_id=2726
Of course that are all false claims as always.

nohal, 2021-01-10:
Again, if you don't have anything to say related to this particular issue, please leave it alone.