OpenCPN/OpenCPN #2149: Linux plugins excessive runtime deps: tracking bug.

leamas, 2021-01-13:
The discussion in #1594 reveals that some plugins on linux has runtime dependencies beyond what's provided by opencpn. We should consider this as a bug, since the plugin loader doesn't resolve package dependencies on plugins (and thus works as Windows and MacOS always have).

Known plugins:

  - photolayer: https://github.com/Rasbats/photolayer_pi/issues/22
  - wheatherfax: Not clear where to file issue, see https://github.com/OpenCPN/OpenCPN/issues/1594#issuecomment-759344172. @jongough  @rgleason

This issue is not aimed to have any general discussions on the merits of the new pluginloader, in particular when it comes to handling package dependencies. It's just about handling the existing implementation, and adapting plugins to it. 

The general discussion on the pluginloader and possible enhancements lives in #1594.

leamas, 2021-01-13:
I suggest that you enable the issue tracker so we have somewhere to file the issue

rgleason, 2021-01-13:
Done

leamas, 2021-01-13:
I don't see any issue tracker at https://github.com/rgleason/weatherfax_pi Delays? Misconceptions?

leamas, 2021-01-13:
I see that you have enabled the issue tarcker on watchdog_pi. This one also has rtlsdr in the build deps, but isn't this just a copy-paste error? Can the plugin be built without it (in that case the rtlsdr line should just be removed)?

rgleason, 2021-01-13:
Yes. It can be removed.

leamas, 2021-01-13:
Then do that. And please enable the weatherfax issue tracker

rgleason, 2021-01-13:
Issues is enabled for Wxfax now.

leamas, 2021-01-13:
@bdbat @nohal: For the time being, would it be possible to update the plugin pages on opencpn.org with snippets like

[weatherfax] : Known issues: On Debian, plugins requires the librtlsdr package to be installed, see  `https://github.com/rgleason/weatherfax_pi/issues/48`. Walk around using *sudo apt install librtlsdr*.

[photolayer]  Known issues: On Debian, plugins requires the libgeotiff package to be installed, see `https://github.com/Rasbats/photolayer_pi/issues/22`. Walk around using _sudo install libgeotiff2_.

Users can access this page using the "website" button on the plugin's panel in the installer, so it seems like the natural point.

bdbcat, 2021-01-14:
@leamas...
Sure, this is possible.
Shall we use the language above verbatim, or is it still under consideration/analysis?
Devs can always email me directly to make changes to the text as necessary.  Or we can continue to track it here, thus retaining history.  I have no preference.

leamas, 2021-01-14:
@bdbcat: when it comes to language... I'm a Swede. Anyone with English as their native language could probably phrase it better. But I think the basic information is correct, and will be for some time.

rgleason, 2021-01-14:
If this note is put in place would I still remove librtlsdr?
> [weatherfax] : Known issues: On Debian, plugin requires the librtlsdr package to be installed, see https://github.com/rgleason/weatherfax_pi/issues/48. [Work around] using sudo apt install librtlsdr.  This package allows the use of the rtlsdr dongle to receive radiofaxes directly.

leamas, 2021-01-14:
Your call,  but personally I don't think so. Users who actually want to use this plugin *can* do it, if they just find the info (perhaps with some help). After all, it's a simple work-around. And sooner or later we will fix this.

leamas, 2021-01-14:
> would I still remove librtlsd

I suppose you mean weatherfax?

rgleason, 2021-01-14:
Yes

bdbcat, 2021-01-14:
Done.
opencpn.org site updated as advised.

leamas, 2021-01-15:
@bdbcat: thanks!

More issues here in photolayer: https://github.com/Rasbats/photolayer_pi/pull/23

It seems like libGLU.so a k. a. libglu-dev is required on arm. If I understand this correctly (probably not...) it's pulled in by wxWidgets on raspbian. 

The simple solution might possibly be to let main opencpn depend on libglu-dev on Debian/arm, so the plugins can use this as it seems necessary lib. But I don't really understand this opengl library stuff, is there a better way. 

Thoughts

leamas, 2021-01-15:
The last issue is libtiff. I tend to think that also this should be pulled in by main opencpn. It's available "fir free" on flatpak, which i my eyes qualifies it as a commonly used.

We could  for now teach linux users to install a package like opencpn-plugins-support, common for all plugins This should pull in things like  libtiff and libGLU. Eventually, we'd let main opencpn depend on it using a weak Suggests: or Recommends:

rgleason, 2021-01-17:
I now find that I have at some time added
librtlsdr-dev,   to ci/control.

Why hasn't anyone commented on that?
- No problems?

leamas, 2021-01-17:
Adding  build-time dependency on librtlsdr-dev  could basically  mean two things. 

The first would be if it wasn't required and used.  Easily tested by removing it and rebuilding. If the plugin builds the dep is not used and should be removed.

The other means that the plugin links to the rtlsdr library. Practically, on Linux this means the users must install the librtlsdr pcakage for plugin to load. If not installed, the plugin has an unresolved runtime dependency, and Opencpn will refuse to load it.

The situation can be diagnosed using ldd.

Bottom line: runtime deps like these are evil and should be avoided. As shown in the photolayer plugin, rebuilding libraries form source as part of the plugin process is fine, and also creates a solution which works on MacOS. In this particular case, I think it's a path that should be explored

leamas, 2021-01-18:
More experiments and thinking on photolayer: The easiest and most straight forward approach is probably to depend on the -dev package when building, an then just bundle the required runtime .so files in the plugin. Simpler then rebuilding from source. but should cover many linux cases just fine

leamas, 2021-01-18:
photolayer fixed. Perhaps this might become some kind of reference plugin when it comes to dependencies, it's hopefully as bad as it gets in this respect.

rgleason, 2021-01-20:
> More experiments and thinking on photolayer: The easiest and most straight forward approach is probably to depend on the -dev package when building, an then just bundle the required runtime .so files in the plugin. Simpler then rebuilding from source. but should cover many linux cases just fine

You'll have to explain this in more detail for me to understand the process, if I am to have any chance of fixing Sean's plugins.

leamas, 2021-01-20:
In photolayer_pi, you can have a look at libs/libglu/CMakeLists.txt. This one just bundles the library on raspbian;  easily modified to bundle on all Linuxes, though.it's 

You probably want to link to headers and library as well. This is standard cmake stuff, there are plenty of examples out there (some of which in main opencpn).

libs/libtiff/CMakeLists.txt is a more complete example which uses a pre-built, statically linked  binaries on windows and the system copy on LInux. The latter is bundled.

rgleason, 2021-01-20:
Thanks, I will look and study. Locally I was having trouble with the libs in watchdog after updating frontend and need to sort that out.
To summarize  librtlsdr should be compiled with the plugin and  I should remove  portaudio  and use

> A plugin in need to play a sound should use PlugInPlaySound() in the plugin API.

leamas, 2021-04-11:
I think the message is received by the plugins.  Closing

closed 2021-04-11 by leamas