OpenCPN/OpenCPN #2150: Opencpn 5.2.4 & oesenc 4.2.17.1 crash (debian 10)

poetsmeniet, 2021-01-16:
I compiled OpenCpn version 5.2.4 and installed oesenc 4.2.17.1 using the plugin installer. Then created system identifier file. All seems wel, up until:

Logging into oesenc (refresh chart list). I get a segmentation fault.

All on Debian, uname -a output:
Linux Julia-plotter-main 4.19.0-13-amd64 #1 SMP Debian 4.19.160-2 (2020-11-28) x86_64 GNU/Linux

poetsmeniet, 2021-01-16:
Runnning in gdb gets me this:
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x555500000000) at malloc.c:3093
3093	malloc.c: No such file or directory.

Can anyone tell me how to compile opencpn with debug symbols?

leamas, 2021-01-18:
You should get debug symbols in opencpn using -DCMAKE_BUILD_TYPE=Debug to cmake. You might also need the symbols for the system packages like wxWidgets, I don't have out the top of my head.

You might also need to debug symbols in the plugin file. Rebuilding it locally should make the trick. After building, overwrite the version installed by the plugin. It lives  under ~ /.local/lib/opencpn.

poetsmeniet, 2021-01-18:
> You should get debug symbols in opencpn using -DCMAKE_BUILD_TYPE=Debug to cmake. You might also need the symbols for the system packages like wxWidgets, I don't have out the top of my head.
>
> You [...]

Thanks mate, will check it out when I'm back on the boat.

olivier-france, 2021-03-22:
I have the same problem on debian 10 but with a compiled version "opencpn5.2.4.0-0~focal1" 
the crash is due to a "segmentation fault" 
i use the compiled version of opencpn installed from source : "http://ppa.launchpad.net/opencpn/opencpn/ubuntu focal main".
When i was used a "4" version on debian 9, I don't have this issue.

leamas, 2021-03-22:
Debian generally isn't binary compatible with Debian, so a crash is more or less expected when running a focal binary. Of course, it *might* work, but if so it's a sheer coincidence.

From Bullseye/11 opencpn is packaged on Debian, resolving this issue. But until then I guess you need  you need to build from source. It isn't that complicated, but requires basic programming skills.

leamas, 2021-03-22:
BTW: there is a opencpn package also in Debina 10. However, this is 5.0.0 and does not the new plugin model, which makes using plugins harder than elsewhere.

olivier-france, 2021-03-22:
Thanks you a lot for your quick response.
I have just make the migration to debian bullseye (it's was time, buster is near the dead line). It's works perfectly.
(I don't compile but install opencpn from package ".deb").

PS: I have even build some soft from source, effectively it's not difficult.

Thanks you so much.

poetsmeniet, 2021-04-25:
Sure

rgleason, 2021-04-25:
Ok

closed 2021-04-25 by rgleason