OpenCPN/OpenCPN #2151: MacOSX - plugin menus

Rasbats, 2021-01-21:
Many plugins use a menubar. On the Mac this presents a problem as the OpenCPN menubar remains in position. If a fix is made for the plugin menubar this can be shown, but the OpenCPN menubar is lost. Only 'Window' remains, alongside the plugin menu. To get the main menubar back it is necessary to close OpenCPN and open the program again.

Could the plugin API be extended to include 'SetOpencpnMenubar', 'GetOpencpnMenuBar'. With this facility we could capture the menubar for the main program and when the plugin is hidden/closed the main menubar reinstated. 

Not sure how practical this is but...

CarCode, 2021-01-22:
Before the changes in the 5.x versions of OpenCPN it has all worked very well. That means the plugin menu appears when the plugin window is activated and the main OpenCPN menu appears when the plugin window is deactivated or when this plugin is closed. So the code of OpenCPN has to be fixed for macOS (formerly called MacOSX). To install new functions when somethings does not work is the wrong way. We already have this so called "Spagetti" code enough in the sources.

We should strongly appeal to the developers not to publish new versions anymore when they are not tested. There are however more problems with the OpenCPN 5.x versions for macOS which leads to crashes or even does not install at all. So I tell macOS users to use the old 4.8.8 version which is stable.

leamas, 2021-01-22:
> There are however more problems with the OpenCPN 5.x versions for macOS which leads to crashes or even does not install at all

I might certainly miss something here. However, searching for "macos" issues and skimming the headers I only find two issues: #2148 and #2088. None of these seems relevant in the context of the comment above.

Still, I agree that adding more functions to the plugin interface should be a last resort. 

@Rasbats: At which point should the main opencpn interface be restored, really?  When the plugin closes it's preferences dialog, closing some other dialog, some other point...?  A screenshot or two could perhaps clarify the situation (?)

leamas, 2021-01-22:
BTW: Using a concrete example from a specific plugin might also be helpful, screenshots or not.

rgleason, 2021-01-22:
Incidentally, the Webpage pLugin Download links that are reliant on navnux are failing.  All download links that use navnux website should be changed to a download link that works. This comes from several reports by users.

CarCode, 2021-01-22:
@leamas: It is a little bit arrogant what you write here. Most users are users only and therefore use the forum for their claims and not github and even not the flyspray because it seems no one of the developers is responsible or read the flyspray. So to see the real world please check these sources first.
Second it is often said already that OpenCPN does not work as before for macOS. So check e.g. version 4.8.8 how it should work and then check the same behavior with 5.2.4 when you are able to install and run 5.2.4 with macOS together with the claimed behavior of these plugins.
Furthermore people wanting to compile OCPN themselfes have problems because of libarchive.dylib. I don't need that lib.

@rgleason: All download links are broken and it is not possible anymore to download older versions of OpenCPN. Do you know who is responsible for the flyspray? It should taken offline also when nobody read it.

Rasbats, 2021-01-22:
Ok...

Any of Sean's plugins that use a menubar will fail on the Mac. Weatherfax picture first with MacOS and then Windows. In Windows and Linux the menus are useable. On the MacOS they have disappeared. A bug in wxWidgets according to Gerhard.

[image]
[image]

What I did with PhotoLayer was SetMenubar(... using two menubars. Used the blank menubar when PL was hidden and the proper menubar when shown. But the OpenCPN menubar is lost. If this could be accessed/shown/hidden?

[image]
[image]

leamas, 2021-01-22:
@CarCode:

>  Most users are users only and therefore use the forum for their claim

Fair enough.  Could you please share some relevant links?

That said, hijacking this bug for general "MacOS does not work"  claims is not productive. I have probably nothing more to say on this,.

nohal, 2021-01-22:
FWIW, nothing described here with the menubars happens on my Mac, everything works as expected and the menu is switched between the main app an plugin as the windows get activated. 

[image]
[image]

leamas, 2021-01-22:
> A bug in wxWidgets according to Gerhard.

hm... any chance Gerhard knows which bug? If we knew, we could perhaps patch the version of wxWidgets we use on MacOS (until 3.2 is out). 

Patching wxWidgets isn't nice, for sure. Could perhaps be motivated in  case like this though, at least of there is an upstream patch available,.

leamas, 2021-01-22:
Mid-air collision... let's focus on  Pavel's findings. What might be the difference between Pavel's setup and Mike's?

Rasbats, 2021-01-22:
Kicking off...

Mac Catalina 10.15.7 with the latest updates

OpenCPN 5.2.4
Weatherfax 1.9.14.0 from the Master catalog

nohal, 2021-01-22:
The only difference here is the up to date Big Sur, but the menubar worked like this as far back as I can remember.

Rasbats, 2021-01-22:
My Mac Mini has an ancient Intel HD Graphics 4000 1536mb graphics card. Does this have anything to do with it?

The chart outlines hamburger menu is not available here.

Rasbats, 2021-01-22:
OpenCPN 5.0.2 works properly but without the plugin manager.

nohal, 2021-01-22:
I highly doubt it has anything to do with hardware. What exactly does the "hamburger menu missing" mean? How different from this is it on your system?

[image]

nohal, 2021-01-22:
One more thing - are you absolutely sure you are not running against some local wxWidgets build (Which sure may contain different set of bugs than the one we distribute in the bundle)?

CarCode, 2021-01-22:
> Any of Sean's plugins that use a menubar will fail on the Mac. Weatherfax picture first with MacOS and then Windows. In Windows and Linux the menus are useable. On the MacOS they have disappeared. A bug in wxWidgets according to Gerhard.

No, I have only said the menu item "Window" in wxWidgets for macOS might have a bug. I have not found a way to localizing it neither its submenus.

Changing and switching the menus between OCPN and plugins always work for me and has nothing to do with graphic hardware. However several people with macOS told me the public OCPN version 5.x crashes certain Mac machines with a dual core cpu.

CarCode, 2021-01-22:
Once again the switching of menus work:

[image]
[image]

Rasbats, 2021-01-22:
Pavel...

'Chart Panel Options' is just a grey rectangle.

At one stage I did compile my own wxWidgets 3.1.2 on the Mac. 

Gerhard...

Which version of OpenCPN are you using in your screenshots above?

leamas, 2021-01-22:
Same for me on a reasonably clean  Catalina VM - menus switch as expected (I think).

OpenCPN downloaded from OpenCPN.org, current release (5.2.4+6b314e6).

Ergo: this is probably about Mike's setup. @Rasbats: Are you sure that you are loading the plugin from the master catalog (i. e., you are not using something you have installed using Import Plugin for testing purposes).

Rasbats, 2021-01-22:
@all ... Give me about 40 minutes. I am reinstalling Catalina.

CarCode, 2021-01-22:
> Gerhard...
> 
> Which version of OpenCPN are you using in your screenshots above?
That is my own version based on the official 5.2.4 sources but without buggy things I don't need such as SignalK, mbtiles or plugin manager and such things.

Rasbats, 2021-01-22:
Catalina reinstalled. Also OpenCPN 5.2.4 from the . Also Weatherfax from the Master Catalog. The same issues as before.

leamas, 2021-01-22:
From the log on my catalina: 

```
PlugInManager:  /Users/leamas/Library/Application Support/OpenCPN/Contents/PlugIns/libweatherfax_pi.dylib
18:12:26.120    INFO pluginmanager.cpp:2404         Plugin common name: WeatherFax
18:12:26.120    INFO pluginmanager.cpp:2406         API Version detected: 116
18:12:26.120    INFO pluginmanager.cpp:2407         PlugIn Version detected: 1.9
eatherfax_pi.dylib
18:12:26.120    INFO pluginmanager.cpp:2404         Plugin common name: WeatherFax
18:12:26.120    INFO pluginmanager.cpp:2406         API Version detected: 116
18:12:26.120    INFO pluginmanager.cpp:2407         PlugIn Version detected: 1.9

```

Do have the same/similar?

Rasbats, 2021-01-22:
20:31:06.882 MESSAGE pluginmanager.cpp:1174 Checking plugin candidate: /Users/mikerossiter/Library/Application Support/OpenCPN/Contents/PlugIns/libweatherfax_pi.dylib
20:31:06.882 MESSAGE pluginmanager.cpp:2169 Plugin is compatible: true
20:31:06.882 MESSAGE pluginmanager.cpp:2232 PlugInManager: Loading PlugIn: /Users/mikerossiter/Library/Application Support/OpenCPN/Contents/PlugIns/libweatherfax_pi.dylib
20:31:06.894    INFO pluginmanager.cpp:2403 PlugInManager:  /Users/mikerossiter/Library/Application Support/OpenCPN/Contents/PlugIns/libweatherfax_pi.dylib
20:31:06.894    INFO pluginmanager.cpp:2404         Plugin common name: WeatherFax
20:31:06.894    INFO pluginmanager.cpp:2406         API Version detected: 116
20:31:06.894    INFO pluginmanager.cpp:2407         PlugIn Version detected: 1.9

Rasbats, 2021-01-22:
```
20:48:19.822 MESSAGE glChartCanvas.cpp:1211 OpenGL-> Renderer String: Intel HD Graphics 4000 OpenGL Engine
20:48:19.822 MESSAGE glChartCanvas.cpp:1220 OpenGL-> Version reported:  2.1 INTEL-14.7.8
20:48:19.822 MESSAGE glChartCanvas.cpp:1271 OpenGL-> Detected Intel renderer, disabling stencil buffer
20:48:19.822 MESSAGE glChartCanvas.cpp:1314 OpenGL-> Texture rectangle format: de1
20:48:19.823 MESSAGE glChartCanvas.cpp:1359 OpenGL-> Using Vertexbuffer Objects
20:48:19.823 MESSAGE glChartCanvas.cpp:1177 OpenGL Framebuffer OK, size = 1358
20:48:19.823 MESSAGE glChartCanvas.cpp:1440 OpenGL-> Using Framebuffer Objects
20:48:19.823 MESSAGE glChartCanvas.cpp:1443 OpenGL-> Using FBO Stencil buffer
20:48:19.823 MESSAGE glChartCanvas.cpp:1451 OpenGL-> Using Depth buffer clipping
20:48:19.823 MESSAGE glChartCanvas.cpp:1570 OpenGL-> Using s3tc dxt1 compression
20:48:19.823 MESSAGE glChartCanvas.cpp:1600 OpenGL-> Compressed tile size: 128kb (8:1)
20:48:19.823 MESSAGE glChartCanvas.cpp:1464 OpenGL-> Minimum cartographic line widt
h:  1.0
```

rgleason, 2021-01-23:
If you are using the most recent version on my repository https://github.com/rgleason/weather_routing_pi
https://github.com/rgleason/weather_routing_pi/blob/master/ci/control
has portaudio19-dev listed, but no librtlsdr listed.
It does compile for all builds and all of them were deployed to opencpn/plugins, I believe.

I'm trying to understand Carcode's statements which don't lead very far towards solving the problem.  I don't have a Mac and I am not going to get one, so I can't test. Mac users are going to have to step up and commit to work with me closely, testing each and every plugin in a methodic and careful way, or they will soon disappear to Mac Users.  This is not a threat, it is just a fact.

bdbcat, 2021-01-23:
First, let me backup leamas:
This issue is about one thing only:  MacOS menubar trouble with some plugins which create new menubars dynamically.
Please read the title. "MacOSX (sic) - plugin menus"
This issue is not a platform to complain about other unspecified bugs in OCPN on MacOs.  
@rick:  "Incidentally, (and then proceed to talk about other issues)" is not helpful here.
Complaining about Flyspray management is not helpful.
If we are to use the github Issues feature to create actionable work items, we need tighter focus.
Please help the devs by resisting the urge to pontificate when you think someone may be listening.
Comments relating to the issue are, of course, welcome.
Thanks.

bdbcat, 2021-01-23:
Menubars with WeatherFax plugin tested on MacMini 10.14.5.
They work fine, switching depending on which window has mouse focus.
AFICT, only outlier so far is @Rasbats....

Rasbats, 2021-01-23:
@All... This morning I removed all files related to OpenCPN (Library etc, etc). Reinstalled OpenCPN 5.2.4.

Weatherfax is working correctly for me on MacOS :) 

Apologies for wasting time on this. My only excuse is that a lot of development work has been done on this machine. 

BTW... For future reference I was led to the solution by changing the language of OpenCPN. This showed a dialog with the version 5.0.2, which I had used recently.

closed 2021-01-23 by Rasbats

rgleason, 2021-01-23:
@bdbcat   

> @rick: "Incidentally, (and then proceed to talk about other issues)" is not helpful here.
> Complaining about Flyspray management is not helpful.

I did not refer to or complain about flyspray,  that is someone else.
The bad links are about the Plugin Downloads page and are a cause for the flurry of complaints about the plugin by others.

Thank you for testing weatherfax in this confusing enviroment.

rgleason, 2021-01-23:
@rasbats  I have noticed similar problems with plugins, Right now I believe I have to start fresh on a plugin I have been working on. Something changes the files and things don't work afterwards. Cloning from the remote fixes the problem usually.  I don't believe it is Norton because I don't find files in the quarantined area.

rick, 2021-01-25:
[image]