OpenCPN/OpenCPN #2153: macos: Development/runtime  library mismatch

leamas, 2021-01-24:
Steps to reproduce:

  1. Build a plugin with 'stc' listed in ` find_package(wxWidgets REQUIRED  base core ... stc)`
  2. Transfer plugin to mac
  3. Run otool -L on plugin dylib: otool -L build/app/ShipDriver-2.5-darwin-10.13.6/OpenCPN.app/Contents/PlugIns/libShipDriver_pi.dylib | grep stc. This shows a dependency on something like `/Applications/OpenCPN.app/Contents/Frameworks/libwx_osx_cocoau_stc-3.1.dylib`
  4. Said file does not exist, OpenCPN does rightfully refuse to load the plugin, having an unresolved runtime dep.

The core issue here is that the list of libraries in the development version at opencpn.org does not match the list actually available in runtime  at `/Applications/OpenCPN.app/Contents/Frameworks`. This is macOS only, other platforms have the complete set of  wxWidgets library files available.

For now, this could be avoided on the plugin side by not linking to stc. This is the correct solution, since the functions here are not used. However, in the long run I think macOS should have the complete set of libraries available -- plugins should be able to use all of wxWidgets  libs, and the platforms should be the same w r t this.

leamas, 2021-01-24:
```
ls /Applications/OpenCPN.app/Contents/Frameworks
libarchive.13.dylib
libcairo.2.dylib
libexif.12.dylib
libfontconfig.1.dylib
libfreetype.6.dylib
libintl.8.dylib
liblzma.5.dylib
libpixman-1.0.dylib
libpng16.16.dylib
libwx_baseu-3.1.2.0.0.dylib
libwx_baseu-3.1.dylib
libwx_baseu_net-3.1.2.0.0.dylib
libwx_baseu_net-3.1.dylib
libwx_baseu_xml-3.1.2.0.0.dylib
libwx_baseu_xml-3.1.dylib
libwx_osx_cocoau_aui-3.1.2.0.0.dylib
libwx_osx_cocoau_aui-3.1.dylib
libwx_osx_cocoau_core-3.1.2.0.0.dylib
libwx_osx_cocoau_core-3.1.dylib
libwx_osx_cocoau_gl-3.1.2.0.0.dylib
libwx_osx_cocoau_gl-3.1.dylib
libwx_osx_cocoau_html-3.1.2.0.0.dylib
libwx_osx_cocoau_html-3.1.dylib
libwx_osx_cocoau_webview-3.1.2.0.0.dylib
libwx_osx_cocoau_webview-3.1.dylib

```

leamas, 2021-01-24:
```
ls /tmp/wx312B_opencpn50_macos109/lib
libwx_baseu-3.1.2.0.0.dylib
libwx_baseu-3.1.2.dylib
libwx_baseu-3.1.dylib
libwx_baseu_net-3.1.2.0.0.dylib
libwx_baseu_net-3.1.2.dylib
libwx_baseu_net-3.1.dylib
libwx_baseu_xml-3.1.2.0.0.dylib
libwx_baseu_xml-3.1.2.dylib
libwx_baseu_xml-3.1.dylib
libwx_osx_cocoau_adv-3.1.2.0.0.dylib
libwx_osx_cocoau_adv-3.1.2.dylib
libwx_osx_cocoau_adv-3.1.dylib
libwx_osx_cocoau_aui-3.1.2.0.0.dylib
libwx_osx_cocoau_aui-3.1.2.dylib
libwx_osx_cocoau_aui-3.1.dylib
libwx_osx_cocoau_core-3.1.2.0.0.dylib
libwx_osx_cocoau_core-3.1.2.dylib
libwx_osx_cocoau_core-3.1.dylib
libwx_osx_cocoau_gl-3.1.2.0.0.dylib
libwx_osx_cocoau_gl-3.1.2.dylib
libwx_osx_cocoau_gl-3.1.dylib
libwx_osx_cocoau_html-3.1.2.0.0.dylib
libwx_osx_cocoau_html-3.1.2.dylib
libwx_osx_cocoau_html-3.1.dylib
libwx_osx_cocoau_media-3.1.2.0.0.dylib
libwx_osx_cocoau_media-3.1.2.dylib
libwx_osx_cocoau_media-3.1.dylib
libwx_osx_cocoau_propgrid-3.1.2.0.0.dylib
libwx_osx_cocoau_propgrid-3.1.2.dylib
libwx_osx_cocoau_propgrid-3.1.dylib
libwx_osx_cocoau_qa-3.1.2.0.0.dylib
libwx_osx_cocoau_qa-3.1.2.dylib
libwx_osx_cocoau_qa-3.1.dylib
libwx_osx_cocoau_ribbon-3.1.2.0.0.dylib
libwx_osx_cocoau_ribbon-3.1.2.dylib
libwx_osx_cocoau_ribbon-3.1.dylib
libwx_osx_cocoau_richtext-3.1.2.0.0.dylib
libwx_osx_cocoau_richtext-3.1.2.dylib
libwx_osx_cocoau_richtext-3.1.dylib
libwx_osx_cocoau_stc-3.1.2.0.0.dylib
libwx_osx_cocoau_stc-3.1.2.dylib
libwx_osx_cocoau_stc-3.1.dylib
libwx_osx_cocoau_webview-3.1.2.0.0.dylib
libwx_osx_cocoau_webview-3.1.2.dylib
libwx_osx_cocoau_webview-3.1.dylib
libwx_osx_cocoau_xrc-3.1.2.0.0.dylib
libwx_osx_cocoau_xrc-3.1.2.dylib
libwx_osx_cocoau_xrc-3.1.dylib
wx

```

Rasbats, 2021-01-24:
stc is used by @antipole for Javascript_pi. His main platform is MacOS.

leamas, 2021-01-24:
Well, if he actually uses the functions in wx/stc/stc.h he must somehow link to the wxwidgets stc library. Which is tricky on Macos, the library isn't part of the opencpn installation. 

He could of course install wxwidgets separately and link to that. Or bundle the stc lib.

leamas, 2021-01-24:
@nohal: thoughts?

nohal, 2021-01-24:
AFAICT the only way is to bundle all the libs that may be used by the plugins and ship them with the application so that everything is correctly linked against the same dylibs and uses the ones under `@executable_path/../Frameworks` (the linking info for all the binaries is modified when the bundle is created by [fixup_bundle](https://cmake.org/cmake/help/latest/module/BundleUtilities.html))

leamas, 2021-01-24:
Mike's plugins doesn't use fixup_bundle, just a simple [script](https://github.com/Rasbats/shipdriver_pi/blob/master/cmake/fix-macos-libs.sh) which runs install_name_tool on the libs. The net result is that the plugin references wxWidgets libs in the OpenCPN bundle.

Seems the we agree that all of wxWidgets should be bundled with OpenCPN on macos, not just the parts used by OpenCPN. Right?

nohal, 2021-01-24:
I don't see any other way that would work without calling major headache.

nohal, 2021-01-24:
BTW, the above script is a perfect example - while it works for the "normal" situation, the result will explode and burn as soon as the app bundle is not sitting in `/Applications/OpenCPN.app`, which is a case for example when one has more than one version of OpenCPN installed at the same time.

leamas, 2021-01-25:
OK, this is a can of worms...

>  the result will explode and burn as soon as the app bundle is not sitting in /Applications/OpenCPN.app,

Good catch... yet another issue. Is a simple fix using for example @executable_path  as the starting point in the script's RUNTIME_PATH possible? (Need to read up on the macos runtime loader...)

nohal, 2021-01-25:
Yes. There is a good short overview of this magic for example at https://wincent.com/wiki/@executable_path,_@load_path_and_@rpath

leamas, 2021-01-25:
OK, thanks, will fix.

Another thing here is of course the non-wx libraries. Plugins tend to use brew to install various libraries to build the plugin.  But unless they are static, there are two cases:

  - The library exists in OpenCPN framework, for example libarchive. In which case the path to the lib should be patched in the same way as the wx libs.
  - The library does not exist in the OpenCPN bundle. It must then be bundled with the plugin, and have the path patched accordingly.

Messy. Is static linking an alternative on Mac? Seems that the brew libraries typically comes with a static library. (On linux, static linking doesn't work since the official libfoo.a files typically are compiled without -fPIC and thus cannot be linked into a shared library).

nohal, 2021-01-25:
As long as we keep support for the 10.9 target, libraries from Homebrew are problematic (read unusable, depending on which ones and on which actual version of macOS the thing is being built) anyway and as such linking them statically is not a problem.
IMO, OpenCPN app bundle should contain a sane set of common libraries (like for example complete wxWidgets as the additional effort to do so is next to none), but we should not care about any additional ones and leave it on plugin developers if they need them - we can't anticipate them all anyway.

And we should drop support for the prehistoric macOS deployment target as first thing when upgrading to wx 3.2 of course. (As we should drop support for XP on Windows).

leamas, 2021-01-25:
> And we should drop support for the prehistoric macOS deployment target as first thing when upgrading to wx 3.2 of course. (As we should drop support for XP on Windows).

Amen. And we should drop trusty....

Otherwise I'm thinking in similar lines. The devil is in the details, again.  I'll try to handle a few plugins as they should to set an example..

leamas, 2021-01-26:
>  OpenCPN app bundle should contain a sane set of common libraries

Looking at what's there today [https://github.com/OpenCPN/OpenCPN/issues/2153#issuecomment-766361851]  here are some libs which might be usable in plugins (libarchive comes to mind). However, these are not included in the build time bundle. 

Is it worth the effort to make these libs which are available in runtime also available in build time by adding libs + headers to the bundle? If so, we should open another issue for that so we can let #2156 close this one.

nohal, 2021-01-26:
I don't think so, all we need to do is deliver them to the final user and that we do already.

leamas, 2021-01-26:
But even if we do, how is building supposed to take place? In build time there are neither headers or libraries to link to available. Using libarchive as an example, one could of course install it form other sources. But that means all sorts of risks for versions mismatches for both headers and lib. 

Or?

nohal, 2021-01-26:
I'm probably missing the problem - libarchive, as all the other libs we us, are in Homebrew and the plugins can use them exactly the same way the core application does at build time.
Our problem was with plugins needing libs not present on the end user system at runtime and that did apply just to the now resolved subset of wxWidgets AFAICT.

leamas, 2021-01-26:
Still, you have a point I didn't think of. All libraries comes from brew. Which means that what might happen is that the homebrew library is updated after the OpenCPN release. Even if downwards compatible, it might add new functionality not available in runtime.

"scratches my head". 

It's late, going to bed. Let's sleep on it.

nohal, 2021-01-26:
If it happens, it will either deserve a new OpenCPN release or we will have to resolve it somehow for the core app as well, so I would simply let it aside until that happens.

closed 2021-01-27 by bdbcat

leamas, 2021-01-27:
> If it happens, it will either deserve a new OpenCPN release or we will have to resolve it somehow for the core app as well

Probably not. But in any case, this is a minor issue compared to how  brew libraries are used in general in some plugins. I'll focus on that, possibly returning to the last thread  here in a new issue.

In a distant future...

PR #2156 (2021-01-27): macos: cmake: Bundle all of wxWidgets for plugins to use (#2153).
The wxWidgets libraries bundled with OpenCPN are the only ones
available for plugins. Bundling all of them means that the runtime
libs matches the ones used in build time.

Closes: #2153