OpenCPN/OpenCPN #2157: Chart downloader performance on Windows

transmitterdan, 2021-02-01:
The chart downloader has poor performance on Windows due to excessive use of DWM (Desktop Window Manager) resources.  Symptoms are akin to a system hang when number of charts in a catalog number >1000+.

I have hacked together a test version of the downloader that uses wxDataViewListCtrl. This is what the user sees.

[image]

Performance under Windows is at least 1,000 times faster (maybe 10,000 times).  Therefore, I propose to rewrite that part of the downloader to use this control for the list of charts. 

Is there any supported platform that does not support wxDataViewListCtrl?

Is there any other reason not to use wxDataViewListCtrl?

bdbcat, 2021-02-01:
Dan
I'm not sure that wxDataViewListCtrl is supported on wxQt/Android.
Let's conditional build for this class under Windows only.  That way, we can test under all other platforms by simple one line code change, and rebuild.  If it checks out OK everywhere, we can then remove the conditional, and the old code.
Nice work.
Dave

transmitterdan, 2021-02-02:
Dave,

I will work on it.  There is already a lot of conditional "spaghetti" in there with the old unused wx checked list control.  Can we get rid of the code related to the old wx checked list control now and just keep the "new" list control and the "new new" wxDataViewListCtrl variants?

bdbcat, 2021-02-02:
Dan...
Feel free to refactor/cleanup.  Just try to leave the Android code path alone, if practical.  I recall that this was a pain to implement on the very small displays.

transmitterdan, 2021-02-04:
Dave...

If you like, please trial a test version that works with Linux (GTK3) and Windows. Not tested with GTK2 nor MacOS.

Where to get code: https://github.com/transmitterdan/OpenCPN
branch: chartDldOpt
tag: ChartList-0.1

One of the classes has 2 entirely different versions depending on a CMake variable.

To build the "original" NEW_LIST variant:

cmake .. -DCHART_LIST=OFF

That is the default and should work with Android but I cannot test.

Or to try the new chart listing:

cmake .. -DCHART_LIST=ON

transmitterdan, 2021-02-06:
Dave,

Still making progress.  I want to clean up a few more things but so far it works on Windows, Ubuntu focal and also on Raspberry Pi buster.  For now it only builds for Windows but with CMake command line switch can ask to build for anything besides wxQT platforms.

Raspberry Pi 4:
[image]

Windows Ubuntu-focal

[image]

transmitterdan, 2021-02-07:
@bdbcat, et. al.

I think the revision is stable now.  Would appreciate if anyone from the team can look at the code and make comments/suggestions.  https://github.com/transmitterdan/OpenCPN/tree/ChartdldrUpgrade

Installers for Windows, raspian/buster and ubuntu/focal are available here:

[raspian](https://cloudsmith.io/~dan-dickey/repos/opencpn-unstable-testing/packages/detail/deb/opencpn/5.2.0/a=armhf;d=raspbian%252Fbuster;t=binary/)

[ubuntu/focal](https://cloudsmith.io/~dan-dickey/repos/opencpn-unstable-testing/packages/detail/deb/opencpn/5.2.0/a=x86_64;d=ubuntu%252Ffocal;t=binary/)

[Windows](https://cloudsmith.io/~dan-dickey/repos/opencpn-unstable-testing/packages/detail/opencpn-msvc_5202696583192_setupexe-Wyk/)

Linux installers should use /usr/local as default install location.

Hakansv, 2021-02-08:
TDan..
Your Windows package tested. No issues detected.
I added some Chart catalogs. Netherlands France, Germany and GSHHG. The Update process is extremely fast. Only a short glimpse of the dialog before finalized and closed. 
The download process is as good as before. Also deletion and edit of a catalog works as it should.
Many thanks
Håkan

transmitterdan, 2021-02-15:
@bdbcat & @Hakansv 
Thank you for the initial review.  After testing on several platforms (Ubuntu 20.04, Arm Pi, Windows) I think this new chart lister is ready for a final review.  I rebased it against a recent master commit.  You can find it here:

https://github.com/transmitterdan/OpenCPN/tree/dldrTest

If desired I can make this into a PR but probably you should look at it first.  Here is a comparison to master link:

https://github.com/OpenCPN/OpenCPN/compare/master...transmitterdan:dldrTest

I somehow deleted a bunch of trailing spaces.  Must be my editor configuration.