OpenCPN/OpenCPN #2158: Scroll-to-zoom is unusable on smooth-scolling devices

emilecantin, 2021-02-01:
Version: 5.2.4 (but it's been present for "a while", 1+ year)
Platform: macOS
Hardware: Apple Magic Mouse, and MacBook's built-in trackpad

Steps to reproduce:
1- Open OpenCPN on any chart, main UI (no menus or anything, just the main chart display)
2- Scroll with the mouse to zoom in / out

Expected behaviour:
- The chart zooms in / out ONE level

Actual behaviour:
- The chart zooms in / out a crazy amount; moving my finger ~2mm on my mouse zooms like 10 levels in / out

I also tried a mouse with a regular scroll-wheel, and it behaves normally (1 click on the wheel equals one zoom level), so it's really related to the smooth-scrolling behaviour of Apple devices.

I suspect (this is pure speculation on my part) that smooth-scrolling devices send a large number of tiny scroll events, instead of a small number of large scroll events. OpenCPN behaves as if one event = one level.

nohal, 2021-02-01:
[image]

emilecantin, 2021-02-01:
@nohal Oh wow, I can't believe I missed this. Thanks, that does solve it!

closed 2021-02-01 by emilecantin