OpenCPN/OpenCPN #2161: Chart Pannel Option does not display any more (OCPN 5.2.4 Linux OpenSUSE)

dominig, 2021-02-03:
The Chart Panel Option accessible from the the right bottom corner is not displayed anymore.
OpenSUSE uses wx 3.2 based on GTK3.
Bug is checked on all OpenSUSE release 15.0 15.1 and 15.2.
It seems to be a layer ordering issue.
- The sliding can be guessed from the top line of the canvas which is visible.
- some time (rarely) it works after playing with F4 (I guess the pen display triggers a layer change).
Only work around is to display the menu to access Currents en Tide display button.
A real pain.

[image]

leamas, 2021-02-03:
hm... Cannot reproduce this on flatpak, the panel is fully visible. But  that is gtk2, not that relevant.

Although the OpenSUSE wxW packages are called 3.2 they are are actually built from the 3.1 series. They have to, since wxW 3.2 isn't released yet. Why are these packages used? Other distros builds using 3.0 + gtk3.

wxWidgets uses a  tick-tock release cycle. The 3.1 series is as far as I understand by definition unstable and for this reason not packaged in many (most?) distros.  OTOH, windows + macos builds against 3.1, so... That said, a question is if this bug also is visible if the opencpn packages are built against wxW 3.0 instead.

dominig, 2021-02-03:
Do not ask me why OpenSUSE calls them 3.2 as wx is at 3.1.4.
I can find bugs on internet around similar issues in wx mailing list but the main one seems to have been corrected in time to enter 3.1.4 (and is in OpenSUSE wx release).

What I can see on the console let thinks of a wx/GTK3 issue.
IN particular the error :
 **Gtk-WARNING **: 21:36:37.013: Negative content height -8 (allocation 2, extents 5x5) while allocating gadget (node scrollbar, owner GtkScrollbar)**
```
opencpn:22275): Gtk-CRITICAL **: 21:36:37.012: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkCheckButton
(opencpn:22275): Gtk-CRITICAL **: 21:36:37.012: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkCheckButton
(opencpn:22275): Gtk-CRITICAL **: 21:36:37.012: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkCheckButton
(opencpn:22275): Gtk-CRITICAL **: 21:36:37.012: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkCheckButton
(opencpn:22275): Gtk-CRITICAL **: 21:36:37.012: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkCheckButton
(opencpn:22275): Gtk-CRITICAL **: 21:36:37.012: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkCheckButton
(opencpn:22275): Gtk-WARNING **: 21:36:37.013: Negative content height -8 (allocation 2, extents 5x5) while allocating gadget (node scrollbar, owner GtkScrollbar)
(opencpn:22275): Gtk-CRITICAL **: 21:36:37.013: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
(opencpn:22275): Gtk-CRITICAL **: 21:36:37.015: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkRadioButton
(opencpn:22275): Gtk-CRITICAL **: 21:36:37.015: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkRadioButton
(opencpn:22275): Gtk-CRITICAL **: 21:36:37.015: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkRadioButton
(opencpn:22275): Gtk-CRITICAL **: 21:36:37.016: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkCheckButton

```

leamas, 2021-02-03:
It is perhaps significant, perhaps not. Have you any reason to believe these messages are related to the actual bug? wxW is known to have all sorts of assertion failures of this kind.

Basic question is still open: Given that opencpn is tested using wxW 3.0 on Linux, what is the reason you think this is an upstream opencpn bug and not a downstream, packaging problem? Perhaps caused by using the untested (on linux) 3.1.4 wxW packages?

transmitterdan, 2021-02-04:
When did this behavior begin?  Can you identify a commit where this behavior began using git bisect or some similar search?

The GTK warnings are not related to this issue I think.  I suspect this is a focus problem along with some WX + GTK specific bug.  I have noticed that on some flavors of GTK the focus of the slide out window gets lost after it appears.

leamas, 2021-02-04:
Just google around for those messages. You will find tons of discussion, with few exceptions inconclusive.

> When did this behaviour begin? 

No idea

> Can you identify a commit where this behavior began using git bisect or some similar search?

Perhaps, but I'm just not motivated. Spent too many hours trying to isolate, report and fix a single similar issue in wxWidgets upstream. 

The situation as I see it:

  - We are basically focusing on our basic platforms: Windows, macOS, Flatpak and Debian/Ubuntu. When it comes to other platforms we are basically in a "patches welcome" state, without spending that much on investigating platform-specific bugs.
  - This particular issue needs to be tested using the regular wxW 3.0 packages to sort out if it depends on the 3.1.4 packages (an obvious suspicion) or something else

leamas, 2021-02-04:
@transmitterdan: I thought I replied to dominiq... that said, my conclusions are the same.

Besides the context of this very bug, there is  of course a need to get more input from using up-to-date 3.1 wxW packages.  It would be nice if someone made pre-release packages available on a PPA. However, this is a non-trivial packaging task...

leamas, 2021-02-04:
>  It would be nice if someone made pre-release packages available on a PPA.

https://wiki.codelite.org/pmwiki.php/Main/WxWidgets31Binaries#toc2

transmitterdan, 2021-02-05:
It's not ready for anybody besides Dave ATM.  Need to check if it breaks Android.

Sindbad2p0, 2021-02-08:
Hello All,
I have the same issue: the option panel not showing. However, this is only happening in full screen mode. In non-fullscreen mode all is ok. I also found that the "About OpenCPN" dialog box (from the Help menu) is not showing in full screen mode, and the dashboard panel which is docked in the right hand side of the screen does not always dock properly when toggling back and forth between full screen and non-full screen.
All this issues are not present in -non-full-screen mode.

I had this issue on two systems:
1x on an Odroid, running 64bit Ubuntu Mate 20 and OpenCPN 5.2.4 just downloaded from the PPA
1x on a Raspberry Pi 4B, also running 64bit Ubuntu Mate 20, same OpenCPN version 5.2.4.

The same Raspberry, running on 32bit Raspbian, and OpenCPN version 5.2.4 does not have this issue, it renders perfectly.

I hope that this message adds something usefull, and that the cause of it can be traced.

Best regards,

Mark

leamas, 2021-02-08:
@Sindbad2p0: I did some checks on my Debian test vm which is an updated sid. I can not reproduce this if  building from current master or from the v5.2.4 tag. So, something differs. 

First question: are there any plugins active when you did this test?  There is a -safe_mode command line option to disable all plugins if it helps...

Sindbad2p0, 2021-02-09:
@leamas : In safe mode the behavior is exactly the same. I have just a freshly installed OpenCPN fro the Ubuntu ppa, so not build from source.
Best regards,
Mark

dominig, 2021-02-09:
I believe that have localised the issue in the program OpenCPN/src/MUIBar.cpp lines 1106-1108 where the animation is created.
Unfortunately, I am not an WX expert. As the code is 2 years old, it must come from a side effect of a change either GTK or Wayland.
But as other animation works, a workaround should be possible.
Would I know more of wx, I would have already tried, but the learning curve is steep and I still have a job :-(
Any hint for testing after removing the animation would be nice.

leamas, 2021-02-09:
hm... Ubuntu Mate certainly isn't Ubuntu.  Is it some kind of incompatibility? "Scratches my head"

What happens if you run the flatpak version instead? Same?

leamas, 2021-02-09:
OK, I fired up an Ubuntu Mates vm running 20.04. Here I can reproduce the problems described, both with the PPA builds and using flatpak.

The basic conclusion for now is just that OpenCPN doesn't work on Ubuntu Mate 20.04, and that there is some kind of incompatibility, most likely in the GUI parts. I'm personally not that surprised about the PPA packages, but disturbed that flatpak shows similar symptoms. Flatpak bundles a lot of system libraries, so seems to be about to very core.

As I understand it Mates 20.04 is not running Wayland (no WAYLAND_DISPLAY in environment).  Given that the current gtk development is focused on running Wayland, this is a factor to consider.

leamas, 2021-02-09:
Next step would be to rebuild the packages on Ubuntu Mates.  Dont't have the time right now.

Sindbad2p0, 2021-02-09:
Hello All,
Thanks for the investigations so far. Apparrently there is no quick fix, nor me doing anything wrong. I will just try another flavor of Ubuntu.

leamas, 2021-02-09:
@Sindbad2p0 : sound reasonable. Please report back, we are all curious (?). I am, at least.

Sindbad2p0, 2021-02-10:
Hello, So I installed Ubuntu server 20.04 and Lubuntu graphic user display system. And after installing OpenCPN from ppa the problems with full screen are non-present anymore. So moving from Ubuntu Mate to Lubuntu solved my problem, and hopefully gives you developpers some more clues about the real cause of the problem.
Thanks for your replies,
Mark

transmitterdan, 2021-02-10:
There are issues I believe due to the way WxWidgets handles windows in multiple environments. On 20.04 with xfce I see an "error" when the chart options slides out it loses focus and the desktop widget at the bottom gains focus. I don't think O can do anything about this behavior. It is not even aware of desktop widgets.

leamas, 2021-02-10:
Perhaps we could at least document a reference setup os -version/window manager/display server which is known to work. 

@Sindbad2p0: Is there any particular reason you don't use vanilla Ubuntu?

Sindbad2p0, 2021-02-10:
#Is there any particular reason you don't use vanilla Ubuntu?
Sorry, but here my Linux knowledge runs short. Vanilla Ubuntu comes without display server, will OpenCPN run without a display manager at all?

transmitterdan, 2021-02-10:
OpenCPN uses X-Windows under Linux.  So no, it will not run without an X display device.  It will run with a remote X display device if the environment is set up. A long time ago I ran O on a headless Pi with a remote X server.  Not very practical but it was possible.

Ubuntu comes with a variety of desktop environments.  I think the culprit here is Mate.  I use xfce with Ubuntu focal and it is usable with OpenCPN.  So I don't think it is Ubuntu itself but rather the desktop environment that is the source of incompatibility.

transmitterdan, 2021-02-10:
For this particular issue I suggest we create a opencpn.conf variable to control animation of the chart options window.  I think that would allow users to turn that feature off and then we would be compatible with many environments.  The Android build does this except it is hard coded instead of a UI configurable item.

transmitterdan, 2021-02-10:
I will even volunteer to add this feature if it is deemed useful.

leamas, 2021-02-10:
I meant standard Ubuntu, gnome + X11 and soon Wayland.

@transmitterdan:  To me, it looks extremely useful. I will volunteer to  build packages If you come up with a patch (-set) -- testing on multiple platforms is easier using .debs

transmitterdan, 2021-02-11:
I think I have found something that might explain what is happening.  In MUIBar.cpp at about line 1220 this code exists:
```
#ifdef __WXOSX__
            if(m_pushPull == CO_PUSH)
                pcc->TriggerDeferredFocus();
#else
            pcc->TriggerDeferredFocus();
#endif
```

What this code does is send a message to the chart canvas window that it should take focus.  For some window managers they will probably interpret this to send the chart canvas options dialog behind chart canvas.  I will focus on this as an avenue for fixing the problem rather than worrying about the animation.  As I don't think the animation is the root problem.  If anyone disagrees let me know.

transmitterdan, 2021-02-11:
@dominig I cannot exactly duplicate your issue but I think there is a way you can test a theory I have.  If you can build from latest master branch for your Mate platform then you could try this:

In file CanvasOptions.cpp at about line 86 you find this line:

```
    wxDialog::Create(parent, wxID_ANY, _T(""), wxDefaultPosition, wxDefaultSize, mstyle);
```
I suggest changing the parameter 'parent' to 'NULL'. This tells wxWidgets to keep this dialog as a child to the main application.  I think this may be a way to force this dialog to stay on top of the various application windows thus preserving the correct Z order.  If you can still test this on Mate would be helpful to me in diagnosing this issue.

dominig, 2021-02-11:
Just done the test. Does not fix the issue.
You can see the bug on a short video posted on YouTube and visible with this link.
The small top out line of the menu canvas coming (with the animation) when the drop down menu zone is clicked is very visible (coming in and out). It's like if the background grabbing done to enable the animation remains on top.
   https://youtu.be/FYKyMQpccLU

transmitterdan, 2021-02-11:
@dominig thanks for doing the test.  I may have found the root cause of the issue.  I can't test SUSE or Mate.  If you feel like trying something as a test case build a debug branch of my OpenCPN fork.

https://github.com/transmitterdan/OpenCPN/tree/canvasOptionsDebug

This tries to ensure that the pull-out dialog stays on top and in focus after the pull-out animation.  Then it tries to restore focus to the chart canvas when the dialog is destroyed.

leamas, 2021-02-12:
@transmitterdan : Mates report form a local make make install build from your branch: Basic issue seem to be resolved, but instead here is a new, interesting thing: In full-screen mode the toolbar has moved to the right. See attached screenshot

[image]

leamas, 2021-02-12:
While it's possible do drag around the top-left toolbar, it's *not* possible to move to left  towards the "normal" position. The whole area left of the toolbar is not accessible when trying to move into it.

transmitterdan, 2021-02-12:
Ok, I will have a look. Obviously I did not make changes to the toolbar.

leamas, 2021-02-12:
Indeed, the misplacement occurs also on current master. OTOH, it does *not* occur on Debian (Sid) which means that this also is a Mate incompatibility. Thus leaning to not filing it as a bug for now.

Any chance that you could rebase the patches on top your canvasOptionsDebug branch to the  v5.2.4 tag? I could then apply these to the flatpak build  and, with some luck,  we would have something working for Mate et. al. ...

transmitterdan, 2021-02-12:
I will work that in a day or two. We are having record severe weather in a few days and We have to prepare for that first.

This fix needs to be tested on all the most popular platforms. I cannot test MAC or Android.

leamas, 2021-02-12:
I  have a MacOS VM .  But I doubt anyone but Dave can make an Android test -- #2081 is still not resolved.

But then again, if you rebase it to v5.2.4 I can apply it as a patch to flatpak only, and test that one. It could be a starter.

dominig, 2021-02-12:
Dan,

I have checked your latest code in your test branch
   https://github.com/transmitterdan/OpenCPN/tree/canvasOptionsDebug
I am afraid that the behavior is not really what is expected but t does change things.

  - chart windows is not display (top bar is still coming and going)
  -  option bar disappears and come back when chart windows is called.
  - IN full screen behaviour is the same but OCPN exists full screen and Chart windows close action is called. 
You can see my video report here.
  https://youtu.be/mYIX-wKab8Q
If you have other test that you can propose to fix the issue, please let me know.
I am a "wx idiot", but I can build, test and reports.
Dominig

leamas, 2021-02-12:
@transmitterdan : Hope you securing your boat until now. I have done some more testing, and it looks like the flatpak version actually is fine. My other test was on another machine, but when I test i a fresh VM here all looks god besides the offset (toolbar cannot be moved to usual top-left position.)

The offset in full-screen mode isn't much to talk about, its basically cosmetic. Which means that we have something actually working on Ubuntu Mate.

@dominig : Have you tested flatpak? Installation instructions [here](https://www.opencpn.org/OpenCPN/info/downloadopencpn.html)

dominig, 2021-02-13:
No I did not test Flat pack. As the maintainer of the OpenSUSE official OpenCPN packages that is not what I am expected to deliver.

leamas, 2021-02-13:
But I did.  Flatpak does not exhibit this bug. Tested on a fresh Leap VM.

Which basically boils down to that for the two reported platformas with this problem (OpenSUSE, Ubuntu Mate) the flatpak packaging works just fine w r t this bug.   Given our basic strategy this means that we are in a "patches welcome" position here, without spending that much time on investigating the issue.

OTOH, "patches welcome" of course also includes possible patches from @transmitterdan,  should you feel motivated...

transmitterdan, 2021-02-13:
This issue is, to say the least, "complicated".  Here is a hack patch that works for the mentioned problem but still has other issues that go much deeper.

```
diff --git a/src/CanvasOptions.cpp b/src/CanvasOptions.cpp
index c0d2bf894..8cc2177fe 100644
--- a/src/CanvasOptions.cpp
+++ b/src/CanvasOptions.cpp
@@ -55,6 +55,7 @@
 //------------------------------------------------------------------------------
 //    External Static Storage
 //------------------------------------------------------------------------------
+extern MyFrame* gFrame;

 //  Helper utilities
@@ -83,6 +84,8 @@ CanvasOptions::CanvasOptions( wxWindow *parent)

     //SetBackgroundStyle( wxBG_STYLE_TRANSPARENT );
     long mstyle = wxNO_BORDER | wxFRAME_NO_TASKBAR | wxFRAME_SHAPED;
+    if (  gFrame->IsFullScreen() )
+        mstyle = mstyle | wxSTAY_ON_TOP;
     wxDialog::Create(parent, wxID_ANY, _T(""), wxDefaultPosition, wxDefaultSize, mstyle);

     wxBoxSizer *topsizer = new wxBoxSizer( wxVERTICAL );
```

This patch will detect full screen mode.  In that case it will force the canvas options slide-out menu to stay on top of all other windows.  Clicking anywhere in the chart window will hide the slide out window.

For some unknown reason, the toolbar and MUI bar disappear when the canvas options menu gains focus.  It is as if these other frames are intentionally submerged but I cannot find any code in O that does that.  It is as if wxWindows stops obeying its own rules and misapplies the Z-order by surfacing the parent frame above it's child frames which it should not do.

Therefore, I think this is a wxWidgets bug but the hack above will make things functional.  At least it does for me.  If this works for anyone else we can discuss including this in a more detailed patch specific to certain platforms that need it.  Windows and Android do not need it AFAIK.  It seems likely that a newer wxWidgets version will also not need this hack.

leamas, 2021-02-13:
Thanks for great investigation!

That said, for various reasons this patch is not required in neither Debian/Ubuntu, Windows, MacOS nor Flatpak i. e., all our supported platforms. When it comes to them, we should thus most likely just sit in the boat (Swedish saying) until wx 3.2 is out.

When it comes to suse, perhaps the logical solution is to apply this patch downstream?

For Ubuntu Mate I tend to say like for Mint: if it doesn't work, why don't just use Flatpak, which works?  Or, like @Sindbad2p0, use another distro.

We could of course try to include the patch upstream, but it carries obvious risks, is hard to test and likely obsoleted by upcoming wx 3.2. It has much against it...

Sindbad2p0, 2021-02-14:
Hello, just to fiddle a bit I tried installing OpenCPN on Ubuntu Mate, Raspberry Pi 4 by using Flatpak, as per instructions: https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:flatpak:background
However, I get the following error:
error: No such ref 'app/org.opencpn.OpenCPN/aarch64/stable' in remote flathub

Does this error make sense, is it because Raspberry (ARM) is not (yet) supported by Flatpak? Or is it me doing anything wrong?

leamas, 2021-02-14:
> is it because Raspberry (ARM) is not (yet) supported by Flatpak?

Yes, sort of. Flatpak as such supports arm (at least aarch64 IIRC), but we have disabled these builds since there are no plugins available. For now, flatpak is x86_64 only.

leamas, 2021-02-15:
This one actually bugged me. Found some time, and 
  - Made wxWidgets 3.1 packages for Debian (Sid) and installed them
  - Rebuilt wxsvg against these packages
  - Rebuilt opencpn against wxWidgets and wxsvg
  - Removed **all** wx 3.0 packages
  - Added GDK_BACKEND=x11 to environment
  
... and did some intitial tests. Obviously, plugins must be disabled (they are linked to wx3.0). The result was basically that when using wxWidgets built from the 3.1.4 tag things was unstable, notably  the position of the top-left toolbar could be outside the main window. OTOH, when using wxWidgets packages built from current master things looks sane,  at a glance similar to flatpak.

Ergo: building opencpn against wx 3.1.4 on linux carries all sorts of risks.

Some day I'll  try to make these packages available for others.  Some day...

dominig, 2021-02-20:
I have progressed on that issue but is raised an other one :-( (should I open a nw bug ??)
Config:
-----------
  with OpenCPN 5.2.x branch
  build with wx-3.0.5/GTK3 (I have made available those special packages)
BUG:
------
The chart windows display and hides without any issue :-)
BUT the plugin setting windows crashes.

Video of the bug
----------------------
https://youtu.be/LDo2kD8xcuw

leamas, 2021-02-20:
This is on OpenSUSE, right? In that case, please feel free to file a bug, but don't expect it to get that much attention. After all, we just don't  support Opensuse in any other way than Flatpak. OTOH, if you or someone else comes up with a patch it should IMHO absolutely be considered

dominig, 2021-02-21:
I found a way to fix the issue by deactivating some optimisation.
Not nice but work reliably in that mode and that what is the most important when going at sea with OCPN as main navigation tool.
I am afraid that in the future handling upto date version of wx will b e required as wx 3.0 will not be there for long.
At least building wx 3.0 against gtk3 works and provides a bit a breeze on supporting coming distro releases.
Have fun at sea (at least when this COVID let us sail).
Dominig

closed 2021-02-21 by dominig

leamas, 2021-02-21:
This is great. If possible, could you share your patch?

wx 3.0 will be around until wx 3.2 is released -- the odd-numbered 3.1  is just a development version, not a stable release.

dominig, 2021-02-21:
You can easily see how I build wx3.0.5 against gtk3 here.
My general project config is here, I gave a special name to my package, so it does not conflict against general wx packages provided by the distro :
   https://build.opensuse.org/project/show/home:dominig:wxWidgets-3_0-nostl-sandbox
The details of my spec file can be seen here :
   https://build.opensuse.org/package/view_file/home:dominig:wxWidgets-3_0-nostl-sandbox/wxWidgets-3_0-nostl/wxWidgets-3_0.spec?expand=1

Then I just had to change the build default for OpenCPN to use my wx3.0.5-gtk3-nostl. The configuration is located here:
   https://build.opensuse.org/projects/home:dominig:opencpn-sandbox/meta

Activating the debug build option fixes my last issue. That can easily be done here:
   https://build.opensuse.org/repositories/home:dominig:opencpn-sandbox

I will keep doing some test, and I will push my changes to the official repo next week if everything keep running flawlessly.

Enjoy your Sunday evening.
Dominig