OpenCPN/OpenCPN #2163: dashboard_pi.cpp:540:1: error: control reaches end of non-void function

asy972, 2021-02-06:
Hello.

5.2.4 can't be build when -Werror=return-type used. The problem in plugins/dashboard_pi/src/dashboard_pi.cpp:540:1:

```
double GetJsonDouble(wxJSONValue &value) {
    double d_ret;
    if (value.IsDouble()) {
        return d_ret = value.AsDouble();
    }
    else if (value.IsInt()) {
        int i_ret = value.AsInt();
        return d_ret = i_ret;
    }
}
```

Hakansv, 2021-02-07:
Than one is corrected in master branch.  
Now changed to:
```
double GetJsonDouble(wxJSONValue &value) {
    double d_ret =0;
    if (value.IsDouble()) {
        d_ret = value.AsDouble();
    }
    else if (value.IsLong()) {
        int i_ret = value.AsLong();
        d_ret = i_ret;
    }
    return d_ret;
}
```

asy972, 2021-02-07:
Really. Thanks.

closed 2021-02-07 by asy972