OpenCPN/OpenCPN #2164: Error generating armhf packages

LennartG-Sve, 2021-02-08:
Trying to generate packages from https://github.com/LennartG-Sve/GPS-Odometer_pi and are getting errors on armhf-packages. The error is:
------------------------------------------------------------
Collecting pip
  Downloading https://files.pythonhosted.org/packages/fe/ef/60d7ba03b5c442309ef42e7d69959f73aacccd0d86008362a681c4698e83/pip-21.0.1-py3-none-any.whl (1.5MB)
    100% |████████████████████████████████| 1.5MB 986kB/s eta 0:00:0101
Installing collected packages: pip
  Found existing installation: pip 8.1.1
    Uninstalling pip-8.1.1:
      Successfully uninstalled pip-8.1.1
Successfully installed pip-21.0.1
+ python3 -m pip install -q cloudsmith-cli
Traceback (most recent call last):
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/site-packages/pip/__main__.py", line 21, in <module>
    from pip._internal.cli.main import main as _main
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/site-packages/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax

Exited with code exit status 1

CircleCI received exit code 1
-----------------------------------
Can't see it logged anywhere and it has been there for a few weeks now. Hope it is not something in my code but I have the impression it is some kind of docker problem. Same thing on both armhf-buster and armhf-stretch packages. All other packages generated ok (Android is not implemented).

/LennartG

leamas, 2021-02-08:
This is about python3. cloudsmith-cli requires  a recent pip package, which needs at least python 3.7+. It's  a mess. Currently working on this issue, trying to find a clean solution. I will most likely fail.

Exactly which plugin is it about? Can you link  to complete build logs?

lskillen, 2021-02-08:
If it helps: The `cloudsmith-cli` package doesn't require a recent `pip` package nor even a recent python3. It wouldn't be recommended, but it can also run within python2. It somewhat looks, in this case, like it is trying to install the latest version of pip, maybe via the `pip install wheel` command? I'd avoid installing/upgrading pip/wheel if possible.

LennartG-Sve, 2021-02-08:
Leamas,
thanks for the prompt reply.

The plugin is 'GPS-Odometer_pi' (see also the initial message) and I have packaged it before. Are using a 'butchered' version of testplugin v149 where I've pulled out json and odapi.

This is for 'build-armhf-buster' ('stretch' shows the same error) where the sections Spin up, Preparing env.vars, Checkout code, chmod and ci/circleci-build-raspbian-armhf.sh all checks out ok.
The error comes with 'ci/cloudsmith-upload.sh' , looks a bit in disorder but this is what I get:
-------------------------------------------------
#!/bin/bash -eo pipefail

ci/cloudsmith-upload.sh

api-16	      buildosx	cmake		include  po	    src
appveyor.yml  buildwin	CMakeLists.txt	LICENSE  README.md
build	      ci	data		mingw	 Resources
ar_SA.mo
bg_BG.mo
ca_ES.mo
cloudsmith-upload.sh
CMakeCache.txt
CMakeFiles
cmake_install.cmake
CPackConfig.cmake
_CPack_Packages
CPackSourceConfig.cmake
cs_CZ.mo
da_DK.mo
de_DE.mo
el_GR.mo
es_ES.mo
et_EE.mo
fi_FI.mo
fil_PH.mo
fr_FR.mo
gl_ES.mo
gpsodometer_pi-0.4.3.0-raspbian-10-buster-armhf.deb
gpsodometer_pi-0.4.3.0-raspbian-armhf-10-buster-armhf.tar.gz
gpsodometer_pi-0.4.3.0-raspbian-armhf-10-buster-armhf.xml
hi_IN.mo
hu_HU.mo
id_ID.mo
install_manifest.txt
is_IS.mo
it_IT.mo
ja_JP.mo
ko_KR.mo
libgpsodometer_pi.so
Makefile
mr_IN.mo
nb_NO.mo
nl_NL.mo
pkg_version.sh
pl_PL.mo
PluginCPackOptions.cmake
pt_BR.mo
pt_PT.mo
Resources
ro_RO.mo
ru_RU.mo
sv_SE.mo
th_TH.mo
tr_TR.mo
vi_VN.mo
zh_CN.mo
zh_TW.mo
+ PROD_REPO=opencpn/gps-odometer-prod
+ BETA_REPO=opencpn/gps-odometer-beta
+ ALPHA_REPO=opencpn/gps-odometer-alpha
+ LOCAL_BUILD=false
+ '[' true ']'
+ BUILD_ID=774
+ BUILD_DIR=/home/circleci/project/build
+ BUILD_BRANCH=master
+ BUILD_TAG=
+ PKG_EXT=deb
+ set +x
+ '[' true ']'
+ pyenv versions
+ pyenv versions
  system
* 2.7.12 (set by /opt/circleci/.pyenv/version)
  3.5.2
+ pyenv global 3.7.1
pyenv: version `3.7.1' not installed
+ pyenv global 3.7.0
pyenv: version `3.7.0' not installed
+ pyenv global 3.5.2
+ python3 -m pip install --upgrade pip
Collecting pip
  Downloading https://files.pythonhosted.org/packages/fe/ef/60d7ba03b5c442309ef42e7d69959f73aacccd0d86008362a681c4698e83/pip-21.0.1-py3-none-any.whl (1.5MB)
    100% |████████████████████████████████| 1.5MB 959kB/s eta 0:00:01
Installing collected packages: pip
  Found existing installation: pip 8.1.1
    Uninstalling pip-8.1.1:
      Successfully uninstalled pip-8.1.1
Successfully installed pip-21.0.1
+ python3 -m pip install -q cloudsmith-cli
Traceback (most recent call last):
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/site-packages/pip/__main__.py", line 21, in <module>
    from pip._internal.cli.main import main as _main
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/site-packages/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax

Exited with code exit status 1

CircleCI received exit code 1
-----------------------------------------------

LennartG-Sve, 2021-02-08:
Sorry for the formatting stuff, copy/paste adds stuff I can't see beforehand

leamas, 2021-02-08:
@LennartG-Sve : You can edit your post (thee dots top-right menu and enclose it in `ett två tre`. Or just select it and use the angle  brackets symbol in the tools field.

leamas, 2021-02-08:
Furthermore: only having python3.5.2 available basically means that you are on Ubuntu 14.04 Trusty or something of the similar age.  Is this necessary? Sticking to such old OS versions means all sorts of problems. Or?

lskillen, 2021-02-08:
Just to point out, it's this line that's causing the issue:
https://github.com/LennartG-Sve/GPS-Odometer_pi/blob/master/cmake/in-files/cloudsmith-upload.sh.in#L72

If you _,must_ install it, pin pip to `<20` if you're running an older version of Python3:
`pip install --upgrade 'pip<20'`

leamas, 2021-02-08:
@lskillen : indeed. 

But sticking to python 3.5.2 (or, horrors, python 2)  means all sorts of problems, not just cloudsmith-cli. So  a better approach might be to use a more recent OS.  IMHO we thus need to explore that path first

lskillen, 2021-02-08:
Completely agreed with you @leamas; although it isn't my place to say. Just wanted to point towards the right direction for the root cause. Good luck with fixing it, and if you do need any advice, happy to help!

leamas, 2021-02-08:
@LennartG-Sve Attaching a build file for trusty which works. This is not from  the testplugin context, I'm using anoher set of templates. But it might you a hint

[circleci-build-trusty.sh.gz](https://github.com/OpenCPN/OpenCPN/files/5945876/circleci-build-trusty.sh.gz)

LennartG-Sve, 2021-02-09:
Leamas, Replaced my version with the above one but it made no difference, still getting the same error. 
Quote:
#!/bin/bash -eo pipefail
ci/cloudsmith-upload.sh

api-16	      buildosx	cmake		include  po	    src
appveyor.yml  buildwin	CMakeLists.txt	LICENSE  README.md
build	      ci	data		mingw	 Resources
ar_SA.mo
bg_BG.mo
ca_ES.mo
cloudsmith-upload.sh
CMakeCache.txt
CMakeFiles
cmake_install.cmake
CPackConfig.cmake
_CPack_Packages
CPackSourceConfig.cmake
cs_CZ.mo
da_DK.mo
de_DE.mo
el_GR.mo
es_ES.mo
et_EE.mo
fi_FI.mo
fil_PH.mo
fr_FR.mo
gl_ES.mo
gpsodometer_pi-0.4.4.0-raspbian-9.4-stretch-armhf.deb
gpsodometer_pi-0.4.4.0-raspbian-armhf-9.4-stretch-armhf.tar.gz
gpsodometer_pi-0.4.4.0-raspbian-armhf-9.4-stretch-armhf.xml
hi_IN.mo
hu_HU.mo
id_ID.mo
install_manifest.txt
is_IS.mo
it_IT.mo
ja_JP.mo
ko_KR.mo
libgpsodometer_pi.so
Makefile
mr_IN.mo
nb_NO.mo
nl_NL.mo
pkg_version.sh
pl_PL.mo
PluginCPackOptions.cmake
pt_BR.mo
pt_PT.mo
Resources
ro_RO.mo
ru_RU.mo
sv_SE.mo
th_TH.mo
tr_TR.mo
vi_VN.mo
zh_CN.mo
zh_TW.mo
+ PROD_REPO=opencpn/gps-odometer-prod
+ BETA_REPO=opencpn/gps-odometer-beta
+ ALPHA_REPO=opencpn/gps-odometer-alpha
+ LOCAL_BUILD=false
+ '[' true ']'
+ BUILD_ID=782
+ BUILD_DIR=/home/circleci/project/build
+ BUILD_BRANCH=master
+ BUILD_TAG=
+ PKG_EXT=deb
+ set +x
+ '[' true ']'
+ pyenv versions
+ pyenv versions
  system
* 2.7.12 (set by /opt/circleci/.pyenv/version)
  3.5.2
+ pyenv global 3.7.1
pyenv: version `3.7.1' not installed
+ pyenv global 3.7.0
pyenv: version `3.7.0' not installed
+ pyenv global 3.5.2
+ python3 -m pip install --upgrade pip
Collecting pip
  Downloading https://files.pythonhosted.org/packages/fe/ef/60d7ba03b5c442309ef42e7d69959f73aacccd0d86008362a681c4698e83/pip-21.0.1-py3-none-any.whl (1.5MB)
    100% |████████████████████████████████| 1.5MB 977kB/s eta 0:00:01
Installing collected packages: pip
  Found existing installation: pip 8.1.1
    Uninstalling pip-8.1.1:
      Successfully uninstalled pip-8.1.1
Successfully installed pip-21.0.1
+ python3 -m pip install -q cloudsmith-cli
Traceback (most recent call last):
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/site-packages/pip/__main__.py", line 21, in <module>
    from pip._internal.cli.main import main as _main
  File "/opt/circleci/.pyenv/versions/3.5.2/lib/python3.5/site-packages/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax

Exited with code exit status 1

CircleCI received exit code 1
Endquote

Note that the above is from build-armhf-stretch and build-armhf- buster looks similar.
This one also fils on trusty with this message:

Quote:
Processing triggers for sgml-base (1.26+nmu4ubuntu1) ...
Processing triggers for ureadahead (0.100.0-16) ...
+ mk-build-deps build-deps/control
W: Unable to locate package build-deps/control
mk-build-deps: Unable to find package name in `apt-cache showsrc build-deps/control'

Exited with code exit status 29

CircleCI received exit code 29
Unquote

Sorry

leamas, 2021-02-09:
@LennartG-Sve 
 
Please edit your post, enclose the log in backticks \` like these \` so it's readable. You can check the result (Preview tab) before updating. Another way is to select the log part and use the angle brackets tool ` <>` in the tools pane above the post.

I said that the new file was using another set of templates and thus cannot be used as-is. You might be able to get some inspiration so to speak, but that's it. One example is the reference to build-deps/control which most likely doesn't exist in your setup. It might be ci/control, but that's just a guess.

In the end, trusty is IMHO just not worth it. I don't think anyone besides Dave is using it, and the most reasonable way to handle this error might be to just drop the trusty build. That's what we did with @Rasbats's plugins, shipddriver et. al.

LennartG-Sve, 2021-02-09:
Leamas, I've edited the posts to make them more readable, hope I managed.

Not really sure where or why Trusty came into this, the problem is with armhf (both versions). 
I've reverted back to the original and working Trusty and will until this is fixed leave both armhf-versions out of my packaging.

Editing or modifying these scripts is far above my knowledge level, I'm just trying to contribute with a GPS Odometer plugin. I'll follow what happens here and with testplugin to see if this problem gets fixed.

bdbcat, 2021-02-11:
If I understand correctly, the ARM builds are now succeeding.
What changes were required, please?

LennartG-Sve, 2021-02-11:
When I first generated this report the latest testplugin, that I use as framework, was at v149. Yesterday, about a week later, it was at v155 and that is working ok. Only Jon can specify the related changes, I'm just pulling them in all together.
Will do some minor changes in my program before next try and prod commit.

leamas, 2021-02-11:
@bdbcat: There is also https://github.com/bdbcat/oesenc_pi/pull/105, although this is a major overhaul rather than a point fix.

LennartG-Sve, 2021-02-11:
Tried to generate a prod release (gps-odometer-prod) but only ubuntu bionic was generated in prod while all of them are available in beta. Something strange still going on.

rgleason, 2021-04-14:
LennartG  
I believe you need to update to Jon's current version as things have changed causing the problem.
I think this should be closed.

LennartG-Sve, 2021-04-14:
My bad, thought bdcat should close as he reopened the case. Closing it now.

closed 2021-04-14 by LennartG-Sve

rgleason, 2021-04-14:
It is OK Lennart, we learn the process by osmosis.
Jon is going to be updating his Testplugin at some point. I would wait til then.