OpenCPN/OpenCPN #2168: OpenCPN toolbar hamburger icon doesn't react properly on V5.2

jserack, 2021-02-16:
I am having the same problem as OpenCPN toolbar hamburger icon doesn't react properly on touch on raspberry pi arm64 or armhf #2097"on a Windows (Vista) hardened touch screen tablet when moving from OPENCPN Version 5.0 to 5.2. It makes it almost unusable. With a mouse plugged into the tablet I can select the hamburger stack and it will open up but with either a finger or stylus on the touch screen there is a bit of dither in position and the hamburger stack moves around versus opening the toolbar.

transmitterdan, 2021-02-16:
If you turn off the auto-hide feature does the toolbar work then?

jserack, 2021-02-16:
Well - sort of - with or without autohide once the toolbar is open you can select tools but not the menu button (for example to hide it if it is open the toolbar drags). If you go over the menu it drags when pressed by either of the wacom stylus or touch screen finger but responds to a USB mouse click. The USB mouse click and hold also results in a drag.

I have done a compare of the source code for toolbar.cpp from version 5.0.x to the current master and see there is some code that has been added (line 2444->2473) that is used to move the toolbar on the screen when event.Dragging is received - so that explains the behavior in 5.2.4 (and 5.2.0) is different than 5.0.0 just not why event.dragging is being received.

I have found some information I think is relevant but I am not deep enough into the inners of OPENCPN to track it to a code change. My understanding: Opencpn uses GTK for the graphical interaction. (although I don't see where to look for the version used in 5.2.4 vs 5.0.0)  searching the web there are other uses of the GTK toolkit where people have documented similar problems and different versions of the toolkit interpret the user input slightly differently.  My simple understanding is that different pointing devices will send hover and then tap (such as a mouse ... the cursor is over the menu icon and then there is a tap) which triggers the opening of the menu - versus a drag event. 
With OPENCPN using the wacom stylus or the resistive touch screen if I am very careful to slide the pointer from the chart (motion) over the hamburger icon and touch it will open, or very carefully avoiding motion tap the wacom stylus exactly where it has been hovering it will open.

So bottom line - maybe the issue with drag event was always there but was previously interpreted to mean to open the toolbar - with the new functionality in 5.2.4 it results in the toolbar moving instead of opening. Perhaps changing state ie open if closed and then drag would solve my issue. Otherwise it seems it will be something in the event "detection" in the gtk ...

bdbcat, 2021-04-27:
jserack...
Corrected in current github master.
Can you test this from the pre-Beta repo, for whatever platform you are using?
https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/

jserack, 2021-04-27:
bdbcat 
I tried opencpn-msvc_5.5.0+2671.ea788bd_setup.exe and did the upgrade preserving previous settings. It responds differently but mostly still follows one's finger about. I have tried long touches (follows finger all the time) and quick stabling touches for which 4 in 20 opened the menu and the rest just moved the hamburger stack.

jserack, 2021-04-27:
@bdbcat When I was looking at this code I was hopeful that I would find some config option that would allow me to fix or lock the location of the toolbar menu - I didn't find it. Is there a way? I don't think I would need to float the  menu very often and would prefer it stay in one location.

bdbcat, 2021-04-27:
@jserack...
Have you enabled "Settings->UserInterface->Touch interface"?  As you may see in the code, if "touch mode" is active, dragging is not permitted at all.
```
        if (!g_btouch && event.Dragging()) {
            s_drag = true;
            wxPoint pos_old = GetScreenPosition();
            wxPoint pos_new = pos_old;
            
            pos_new.x += pos_m.x - s_pos_m_old.x;
            pos_new.y += pos_m.y - s_pos_m_old.y;
            
            ocpnFloatingToolbarDialog * parentFloatingToolBar = dynamic_cast<ocpnFloatingToolbarDialog*>(GetParent());
            parentFloatingToolBar->MoveDialogInScreenCoords(pos_new, pos_old);
            s_pos_m_old = pos_m;
            return;            
        }

```

jserack, 2021-04-28:
@bdbcat SUCCESS - Thank you very much. It works perfectly now with the "touch interface" turned on. I have tested it with both a wacom touch stylus and a capacitive finger touch. Works as it should now - from my perspective the bug is solved. (Thanks for pointing out the "Settings->UserInterface->Touch interface" in my trashing around I must have turned it off and forgotten and of course the upgrade respected the prior configuration.

jserack, 2021-04-28:
From my perspective this bug is solved - tested on opencpn-msvc_5.5.0+2671.ea788bd on a GD-IX350 hardened outdoor tablet running windows business vista.

closed 2021-04-28 by jserack