OpenCPN/OpenCPN #2169: Watchdog update master 2.4.23 url error made the plugin to disappear.

Hakansv, 2021-02-18:
Issued here since I can't judge if it's a plugin manager or a WatchDog plugin issue.
Win10
When installing the master catalog update the url is not found, see pict1, but what's most important is that the previous version is uninstalled and the plugin is no longer available. Pict2
A user would accept a update link is broken but not to be left alone without any working plugin version. I suppose the plugin manager can check for a working update before uninstalling the present version to revert to?

From the log file:
```
`09:24:30.937 MESSAGE update_mgr.cpp:246 Selected update: Watchdog
09:24:30.953 MESSAGE pluginmanager.cpp:448 Installing Watchdog
09:24:31.179 MESSAGE Downloader.cpp:134 filesize https://dl.cloudsmith.io/public/opencpn/watchdog-alpha/raw/names/watchdog_pi-2.4.23.0-msvc-x86_64-10.0.14393-MSVC-tarball/versions/2.4.23.0+179.8dfb00b/watchdog_pi-2.4.23.0-msvc-x86_64-10.0.14393-MSVC.tar.gz: 324 bytes

09:24:31.600 WARNING Downloader.cpp:84 Failed to get 'https://dl.cloudsmith.io/public/opencpn/watchdog-alpha/raw/names/watchdog_pi-2.4.23.0-msvc-x86_64-10.0.14393-MSVC-tarball/versions/2.4.23.0+179.8dfb00b/watchdog_pi-2.4.23.0-msvc-x86_64-10.0.14393-MSVC.tar.gz' [The requested URL returned error: 404 Not Found]

09:24:33.604 MESSAGE pluginmanager.cpp:465 Installation of Watchdog failed
09:24:33.605   ERROR file.cpp:231 can't open file 'C:\ProgramData\opencpn\\plugins\install_data\watchdog.files' (error 2: The system cannot find the file specified.)
````

[image]

[image]

Hakansv, 2021-02-18:
BTW: The Watchdog beta catalog update, 2.4.22, is working though.

leamas, 2021-02-18:
Indeed. The code is in src/download_mgr.cpp around line 215. The short story is that existing plugin is removed before checking that the download succeeds.Clearly, this is a bug which should be fixed. 

Here are also other aspects: how come that here is an unaccessible url in the master catalogue? It shouldn't happen, there are some automatic checks in the plugins project to prevent that, at least when the catalogue is updated. Do we need to add some nightly job which checks that urls in the master catalogue is still accessible?

There is also #2046 which touches the same code. From my egoistic perspective I would thus prefer that this was merged before we dealt with this issue (which should be a simple fix).

leamas, 2021-02-18:
hm... perhaps I could add a fix also for this to #2046, to avoid conflicts. We'll see...

rgleason, 2021-02-18:
Sorry, I just saw all this.  Isn't src/download_mgr.cpp in Opencpn, I don't find it in WD src?  So that's not something I need to fix.
The main problem is the url is not correct which is weird because Actions is supposed to check that.
I will make a PR for 2.4..23.2 but there are still some things that need repair. Like playing the warning sound files in  data/warning.

rgleason, 2021-02-18:
The cloudsmith repos is  https://cloudsmith.io/~opencpn/repos/watchdog-alpha/packages/
and  the version is 2.4.23.2

leamas, 2021-02-18:
If you are using that repo, seemingly for alpha, for stable releases that might be a problem depending on the retention policy. Was the current url correct when the PS was merged?

rgleason, 2021-02-18:
Good point Alec. We set Alpha to be stricter. I will check. It might have been 30 days, but I think it is a storage limit, like 100 packages.

rgleason, 2021-02-18:
I did a query and the files seem to be there. Will check the url itself later.

leamas, 2021-02-18:
> We set Alpha to be stricter. I will check. It might have been 30 days, but I think it is a storage limit, like 100 packages.

In any case, packages in this repo should obviously not be used in master branch of o/plugins.  Such packages should be tagged and live in a repo which basically never cleans things up except for manual actions.

rgleason, 2021-02-18:
Alec that is not the way package retainage works.  Manual operations are repetitive and impractical the way it is 
implemented.  I have made some requests for improvement, but support discussed it and decided not to do anything.

See https://help.cloudsmith.io/docs/retention-lifecycle
Also in the Opencpn Cloudsmith  Organization repository  https://cloudsmith.io/repos/
Enter a plugin repository, then  Hit the recycling icon in the second horizontal row of icons for the retainage policies set. Generally they are all set as follows (I have not checked recent additions):
[Alpha Repositories](https://cloudsmith.io/~opencpn/repos/watchdog-alpha/retention/) = Limit by days 36 , Limit by Count  233 packages,  Limit by size 0 (disabled)  disjunctive
[Beta  Repositories](https://cloudsmith.io/~opencpn/repos/watchdog-beta/retention/) = Limit by days 0 (disabled),  Limit by Count  208 packages, Limit by size 0 (disabled)
[Prod Repositories](https://cloudsmith.io/~opencpn/repos/watchdog-prod/retention/)= Same as Beta

leamas, 2021-02-18:
Well, the basic promise for a package in master branch is that it will not be removed from the repo while it's still in the catalogue.  You will need to fix this.

The workflow we  are using in shipdriver is that any version which  goes into the master branch is tagged. And we don't clean up tagged versions, never.  This was the initial setup, and I still think it's sound.

rgleason, 2021-02-18:
Prod repositories could certainly be frozen and retained forever. Does this mean we are going to support all the past releases?

rgleason, 2021-02-18:
The main question is why did the 2.4.23.0 version fail?  Bad metadata?  Missing file? etc.
The answer is that someone has deleted the tarball and xml msvc files.
The reason I know this is that m
y plugins master has the original file still and the paths are correct.
I will next try to determine who deleted these two files in cloudsmith.
The metadata file found in my plugins master 

is zipped and attached
[watchdog_pi-2.4.23.0-msvc-x86_64-10.0.14393-MSVC.zip](https://github.com/OpenCPN/OpenCPN/files/6006081/watchdog_pi-2.4.23.0-msvc-x86_64-10.0.14393-MSVC.zip)

rgleason, 2021-02-18:
Well, I was wrong, nobody deleted these files, a query results in three msvc files  (exe, metadata, tarball
https://cloudsmith.io/~opencpn/repos/watchdog-beta/packages/?q=2.4.23.0-msvc

The xml uses this path:
https://dl.cloudsmith.io/public/opencpn/watchdog-alpha/raw/names/watchdog_pi-2.4.23.0-msvc-x86_64-10.0.14393-MSVC-tarball/versions/2.4.23.0**_+179.8dfb00b_**/watchdog_pi-2.4.23.0-msvc-x86_64-10.0.14393-MSVC.tar.gz

However the path to the tar.gz file to download is
https://dl.cloudsmith.io/public/opencpn/watchdog-beta/raw/names/watchdog_pi-2.4.23.0-msvc-x86_64-10.0.14393-MSVC-tarball/versions/v2.4.23.0/watchdog_pi-2.4.23.0-msvc-x86_64-10.0.14393-MSVC.tar.gz

leamas, 2021-02-18:
> Prod repositories could certainly be frozen and retained forever. Does this mean we are going to support all the past releases?

For some time, yes, Just because we publish a new catalogue it doesn't mean that all users immediately pushes the "Update Catalog" button. So old catalogues will linger out there, using their old URLs to download stuff.

There is a limit on how old a catalogue could be while still expected to work, but these are things we never really have settled.

rgleason, 2021-02-18:
PROD is currently set at Limit by 200 packages. How many packages is enough?

rgleason, 2021-02-19:
Hakan wrote: 

`BTW: The Watchdog beta catalog update, 2.4.22, is working though.`
Yes, that used the master branch.

rgleason, 2021-02-19:
Have pushed 2.4.23.2 to plugins. I think it should work, it works locally.

rgleason, 2021-02-19:
```
For some time, yes, Just because we publish a new catalogue it doesn't mean that all users immediately pushes the "Update Catalog" button. So old catalogues will linger out there, using their old URLs to download stuff.

There is a limit on how old a catalogue could be while still expected to work, but these are things we never really have settled.
```
To take a different tack here Alec, you and Bdbcat made the plugins easy to update, why shouldn't users be encouraged to Update the catalog, setting themselves up to update all older plugins easily. Their local machine has the cache which you have placed all previous versions, so keeping endless copies of older plugins is not really needed. 

Of course this assumes newer plugins are always better, which is not necessarily true, but the cache does carry previous versions which are easily installed.

rgleason, 2021-02-19:
Thinking a bit more about this, what is needed to keep is the last plugin for a given version of OpenCPN when changing wxwidgets etc.   So perhaps just keeping everything is easiest?

leamas, 2021-02-21:
> So perhaps just keeping everything is easiest?

[indeed](https://github.com/OpenCPN/OpenCPN/issues/2169#issuecomment-781646226)

leamas, 2021-02-21:
To summarize: this brilliant bug report  reveals no less than three bugs:

  - On update, the existing plugin is removed before a successful download is done. This obviously breaks badly when download fails.
  - The URL in the catalog was [wrong from the beginning](https://github.com/OpenCPN/OpenCPN/issues/2169#issuecomment-781659210) (how come?)
  - The url used a repository with [questionable retention policy](https://github.com/OpenCPN/OpenCPN/issues/2169#issuecomment-781713024).

We need to sort out that all issues are fixed before closing this bug. I think I might find some time to handle the first bullet, but it will certainly not be this week,

rgleason, 2021-02-21:
Item #2  In summary, I have investigated this and have not found the answer.  Check remote remote plugin metadata and cloudsmith and it appeared there were extra characters in the metadata that were not necessary.  See above notes.
Subsequent versions do not exhibit this problem.
 
Item #3  This can be changed, and it will have to be changed for all Opencpn Organization plugins, but I am not going to do anything without a full vetting and clear instructions.

bdbcat, 2021-02-22:
On # 3:
A valid point, although something of a red-herring in the context of this bug.
If the PROD channel retention policy promises to retain the latest 200 packages uploaded, and assuming that there are less than 50 packages per release (reasonable?), then this should ensure that at least 4 previous versions are available to support catalogs not updated "in the wild".  I think we can declare that this is enough.
Worst case:  User wants to re-install a plugin, and has somehow trashed her local cache, and also refuses to update the master catalog.  She gets a "404".  After slight panic, she updates the master catalog, and all is well.
So, I suggest that #3 is not an issue.  Let's focus on #1, and chalk up #2 to a glitch in the ever-evolving Frontend processes.
I'll take a look at #1.

rgleason, 2021-02-22:
#2 Dave your merge of https://github.com/OpenCPN/plugins/pull/339#event-4357004984 and the previous watchdog 2.4.23.2 should fix this.
#3 Thanks, Dave, just for good measure I will increase the retainage to 250 packages..

Hakansv, 2021-02-22:
#2 
Fix confirmed. Catalog master version 2.4.23.7 installs as it should on Windows and RPi Buster.
#3
I've no real insight on this but a 250 downloads limit, if so, sounds like someone have to regularly watch it so it's not passing the edge?
Thanks.

bdbcat, 2021-02-22:
@Hakansv 
#3.  The packages are automatically purged as necessary, oldest first.  So the contents of the repo will be the freshest, always.

leamas, 2021-02-22:
> If the PROD channel retention policy promises to retain the latest 200 packages uploaded, and assuming that there are less than 50 packages per release (reasonable?),

We currently typically build 13 platforms:

      $ ls metadata/weatherfax_pi-1.9.14.0-* | wc
      13      13     840

and something like 29 plugins:
     
      $ ls metadata/*focal* | wc
      29      29    1931

And we have all in all 455 plugin builds defined:

    $ ls metadata | wc
    455     455   22898

Which might  indicate that keeping the last 200 (or 250) isn't enough. However, not all plugins uses the PROD repository. But still...

leamas, 2021-02-22:
>  then this should ensure that at least 4 previous versions are available to support catalogs not updated "in the wild". I think we can declare that this is enough.

Right. In other words, the repositories storing plugins should store everything listed in the four last versions of the master catalog. We could stick to this compute the storage needed.

bdbcat, 2021-02-22:
'However, not all plugins uses the PROD repository.'
The current CLoudsmith storage arrangement stores each plugin in a separate repo. Or at least most do(Rick, Jon, Dan), only mine are not quite playing nicely with their peers.  On my TODO list.
So, per repo, the math gives 13 tarballs, 13 metadata files, and up to 13 debs, or exe, or pkg.  Yielding maximum 39 files per release, with the current platform set.  Thus, 200 ought to do it.

leamas, 2021-02-22:
Right, I stand corrected.  

Still, the "four last releases" requirement might be something to document so we don't loose it.

rgleason, 2021-02-22:
OK, this is very helpful.  I believe all OpenCPN Organization  Plugin PROD Repositories are set to 200 packages now. I will check the recently added repositories to be sure.  Also I will add a note in the Dev Manual about this about retainage and retaining the four last releases, with an example for the math.  Thanks all.

rgleason, 2021-02-22:
Please see Placeholder Note: https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:pi_installer_procedure#placeholder_note

Currently each plugin has between 12-15 environments, each environment deploys a tarball and metadata file and may deploy one additional file (for deb linux, exe windows or pkg macos file). 15 environments x 3 files x 4 releases = 180 packages. 
I wonder if we should increase the number of packages to  260 to be sure, in case a developer pushes the same version number twice or something..

Note: Mike [Rasbats] and I agree that the Dev Manual - Plugin Manager sections need rewrite and improvement as the process has improved and we need better instructions for plugin developers.  Some of these sections were written as the Plugin Manager process was being developed and need to be revised and rewritten.

rgleason, 2021-02-22:
Hakan wrote:

> I've no real insight on this but a 250 downloads limit, if so, sounds like someone have to regularly watch it so it's not passing the edge?

How many environments are we going to try to build? That is the determinant.  There is a practical limit. We have dozens of plugins that have not been fully tested but do build and are in Plugin Manager.  

Some user feedback "Use Count" per environment would help. If a user uses a plugin in a certain environment it would be nice to know that in a counter or something.  That would also show how many users are using that environment.  I imagine that would also help for Opencpn too..

leamas, 2021-02-22:
> Some user feedback "Use Count" per environment would help.

Indeed. However, note the privacy concerns.

rgleason, 2021-02-22:
All I know is that Dokuwiki does this with their plugins and it is very helpful. I have no idea what it would take to implement.  No private info would be needed, just the use count.

leamas, 2021-02-22:
In our case it would mean that  the application would connect to some kind of data base to update usage etc. This is by definition a privacy breach, and is subject to all sorts of discussions. We might run into some serious trouble if we implement this.

I would probably have to patch away any such functionality in the downstream Debian package to make it pass the privacy concerns there. There might be loopholes if there is a prominent dialogue where users approves the data collection.

Dokuwike, being a web service, can of course do this since it already stores all sorts of user info in a database.

rgleason, 2021-02-22:
Good points.  I guess that's no so good an idea.  

I think Cloudsmith keeps download data, which might help us understand what is being used and what is not, so we can trim the number of environments used. 

https://help.cloudsmith.io/docs/download-logs
The repository download logs show you a detailed list of every file that was downloaded from the repository: 
Counting Docker image downloads is a little different.

For example here is aisradar PROD repository
https://cloudsmith.io/~opencpn/repos/aisradar-prod/groups/

- Windows  2192 
- raspian armhf buster  754
- MacOS       257
- Flatpak        74
- Ubuntu 18.04 bionic  72
- raspian armhf stretch 15

The rest are ones and twos!
Is this the same case for other plugins
Perhaps we should focus this down a bit and support the more important environments to simplify this?
Maybe users should be forced to update Ubuntu more frequently, to be honest about this, that is the one that has the most environments.

rgleason, 2021-02-22:
Logbookkonni

https://cloudsmith.io/~opencpn/repos/logbookkonni-prod/groups/

- Windows  1750
- raspian armhf 10 buster 508
- Darwin  255
- ubuntu 20.04  gtk3  158
- ubuntu 18.04 bionic   102
- raspian armhf 9.4   21
-

rgleason, 2021-02-22:
I think further study of this will lead us to conclude that we should have at most 8-9 environments.
This would simplify the issues found building everything (12-14 environments) for each plugin and reduce time, effort and cloudsmith storage.

rgleason, 2021-02-22:
Perhaps someone should set and announce the plugin environments policy to users after some more consideration and discussion, before we do anything about this.

bdbcat, 2021-02-23:
And why do we want to trim the number of environments?

rgleason, 2021-02-23:
1. Simplify and ease the number of environments that have to be fixed when they stop building.
2. Save time during maintenance and updates of plugins.
3. Be able to maintain and update more plugins within a reasonable time, to provide better plugins to the majority of users.
4. Focus plugin build efforts on the environments that are used most.
5. Reduce the number of packages and space used in Cloudsmith.
6. Trimming the number of environments would probably affect the number of Ubuntu environments the most because some of these seem to have a lower number of downloads (at first glance).

Also it seems that there are more Windows and MacOS users, along with certain versions of Ubuntu, and we certainly want to support Android if possible.  Unfortunately getting Android to build and getting the plugin actually working in Android takes skill and work, that not many have.

There is an intermediate alternative that might help. Try to build all of the 14 environments and if they build, include them in Plugin Manager, but if they are not a primary environment and don't build, don't waste time trying to fix them, but spend the time on the more important builds that are not working.

rgleason, 2021-02-23:
https://cloudsmith.io/~david-register/repos/opencpn-unstable/groups/?page=1

bdbcat, 2021-02-23:
Rick...
It looks like your list is focused mostly on the effort required to configure and execute plugin CI builds. Valid points, but our "business" is to provide managed plugins for each environment that OCPN can be reasonably run on.
I do not know why the two Frontend templates for plugin building are so visibly unstable.  My experience is that one needs to do little maintenance on a CI build, _once it is configured correctly._  But that is only my experience with three plugins.
`The other problem is we don't have much feedback about how the plugin works in all these environments.`
And we will get no feedback at all if we remove the plugin from that environment.  Instead, we will provide a less than satisfying answer to the CF question: "Where is plugin ABC for my environment XYZ?"
I think it is important to distinguish between problems in the CI build process, versus problems at run-time.  Probably makes good sense to focus effort on run-time problems based upon the popularity of the environment. But the CI process is relatively static, once debugged the first time.
We do not (yet) pay for Cloudsmith resources, so this is not a factor.
In summary, I think our goal should be to to tick all the boxes in the environment vs plugin matrix.
So let us soldier on, fixing as we go.

bdbcat, 2021-02-23:
Why the link to dsr repo at Cloudsmith?  Any point?
Why do you care what I do at Cloudsmith, as long as the metafile links for my plugins are correct and durable?

leamas, 2021-02-23:
@rgleason: Now, the discussion  at the end of this issue is sort of interesting and contains valid data points. However, it's totally unrelated to the issue for this bug.

Could you please open a separate issue for this this discussion so we can close this bug once  [the first bullet](https://github.com/OpenCPN/OpenCPN/issues/2169#issuecomment-782925958) is fixed -- the others seems to be dealt with.

rgleason, 2021-02-23:
Ok Alec will do.
Dave I was trying to find out how many downloads of opencpn  for each environment. That repos is probably not the right one to look at and my guess is that this info is not available.

rgleason, 2021-02-23:
Added this issue with a summary.

closed 2021-03-05 by bdbcat

Hakansv, 2021-03-05:
#1 fix confirmed.
A false url creates a error message but the previous plugin version remains as is.
Thanks

leamas, 2021-03-05:
And thanks for some really useful reporting!

commit 85163b45 (2021-03-05): Downloads: Don't uninstall plugins on failed downloads (#2169).

Don't remove existing plugin until there is a new one successfully
downloaded.

Closes: #2169