OpenCPN/OpenCPN #2173: Give Names to the toolbars so window managers can apply specific rules to them

FredericGuilbault, 2021-02-22:
Hi, 
My next release of lysmarine will run on AwesomeWM which is a tiling window manager. Right now, Opencpn itself play well with the tiling, but the 2 toolbars don't. The window manager consider at least one of them as a full application windows and the other one as some sort of always on top floating popup window with rules relatives to his parent but not transient? (I'm still having trouble to understand how it behaves yet).
 
[image]
[image]
[image]

I tried to write some specific rule set for those 2 toolbars, but I can't identify them correctly. 
Doing `xprop WM_NAME` return a value for the main window but not for the two toolbars. 

[image]

At my level. I'm limited to the freedesktop.org specifications that AwesomeWM implement, So IDK if it will be enough, but in any case. I need a `WM_NAME`  identifier for each toolbar.

bdbcat, 2021-02-22:
How would a name help?  Show us in pseudo code, if possible, the proposed interaction with the WM

FredericGuilbault, 2021-02-22:
I will help me to be able to attach custom set of rules to each specific toolbar. ( Im not sure that I will be able reach acceptable behaviour thou this method. But it will at least allow me to try and understand what are the exact problematic behaviours that are hard coded) 

___
These an actual LUA code that could be tested in AwesomeWM 

This first example maximize every window of opencpn:
```
{
    rule_any = { class = { 'Opencpn' } },  --This match any window that have  WM_CLASS(String) . The main, and the toolbars.
    properties = {
         maximized = true,
    }
  }
```

But don't want to have toolbars and any dialog or popup maximized, so let's match only the main window:

```
{
    rule_any = { name = { 'OpenCPN *' } },  -- Now this match window by WM_NAME using a wildcard for version number
    properties = {
         maximized = true,
    }
  }
```

Now my problem is, toolbars don't have names, so I can't write rules for them. 

I wish I could do: 
```
{
    rule  = { name = { 'The_Zoom_In_And_Out_toolbar' }  }, 
    properties = {
       floating = true,
       focusable = false,
       type = "toolbar",
       sticky = false,
       draggable = true,
       above= false,
       top=true,
       size_hints_honor = false,
       skip_decoration= true,
       placement = awful.placement.bottomLeft,
  }
 }

rgleason, 2021-04-22:
I believe this is a feature request so it really does not belong here.  Please close it, and open a Tracker item  with a link back to this closed Issue.  Thank you

closed 2021-04-24 by FredericGuilbault