OpenCPN/OpenCPN #2176: Debian: FTBFS : glChartCanvas.cpp: glXGetProcAddress undefined

leamas, 2021-02-22:
As heading says.  Compilation fails with
```
.../src/glChartCanvas.cpp:318:36: error: ‘glXGetProcAddress’ was not declared in this scope; did you mean ‘ocpnGetProcAddress’?
  318 | #define systemGetProcAddress(ADDR) glXGetProcAddress((const GLubyte*)ADDR)
      |                                    ^~~~~~~~~~~~~~~~~
.../src/glChartCanvas.cpp:318:36: note: in definition of macro ‘systemGetProcAddress’
  318 | #define systemGetProcAddress(ADDR) glXGetProcAddress((const GLubyte*)ADDR)
      |                                    ^~~~~~~~~~~~~~~~~
```

The root cause seems to be that glXGetProcAddress is defined in GL/glx.h which isn't included. Adding a `#include <GL/glx.h>` solves the Debian issue, but other platforms needs to be considered before making a patch.

leamas, 2021-02-22:
Duplicate...

closed 2021-02-22 by leamas

transmitterdan, 2021-02-23:
FYI, this is fixed in the downloader refactor PR.