OpenCPN/OpenCPN #2177: The oeSENC implementation does not honor color name DEPCN

LennartG-Sve, 2021-02-24:
Color name DEPCN is to be used as 'All depth contours other than the safety contour must use this colour'. I read that as this contour should then be used for all contours not covered by DEPSC.

Color name DEPSC, is to be used as 'This colour is reserved for the selected safety contour', today used to show Depth contour between 'Safety Depth' and 'Deep Depths'.

Referencing the settings in OpenCPN with 'Shallow Depth', 'Safety Depth' and 'Deep Depth' I feel that DEPSC should be used between 'Shallow Depth' and 'Safety Depth' while DEPCN should be used for the other contours.

Note: The quotes above are as stated in the "S-52_PresLib_e4.0.0_Part I_Clean_Draft.pdf' document presented by "INTERNATIONAL HYDROGRAPHIC ORGANIZATION". Please also compare the pattern descriptions (for DEPCN and DEPSC) in 'chartsymbols.xml' with the descriptions in the IHO document.

Verify the respective color names above setting them, one at a time, to e.g r="255 g="0" b="0" and note the display.

/LennartG

LennartG-Sve, 2021-02-24:
The platform used is Ubuntu 2.04, OpenCPN 5.2.4 and oeSENC 4.1.17.1, also tested on oeSENC 4.2.19.1. Same result, still no DEPCN contours.

bdbcat, 2021-02-24:
@LennartG-Sve 
How about some screenshots to illustrate the problem?

LennartG-Sve, 2021-02-24:
[image]
This image illustrates the lacking contour lines between the white and light blue areas (10m) plus the lacking contour limiting the 3 m contour line. The shown contour line is for the 6 m contour (DEPSC) that I feel should be at the 3 m limit (as safety contour line). 
The colors here are from the CF thread 'Colors and symbols for vector charts', Swedish classic setup. Colors are still preliminary but close.

tai-fun, 2021-02-24:
LennartG-Sve
Have a look at the picture below. I have the same setup as you.(Ubuntu
20.04 etc)
What are your depth settings for vector  charts?

LennartG-Sve, 2021-02-24:
tai-fun,
there is no picture attached so nothing to see there. 
My depth settings are the same as mentioned distinguishing the colours: 3, 6 and 10 meters.

tai-fun, 2021-02-24:
[image]

I think the picture will be visible for you now.

LennartG-Sve, 2021-02-24:
tai-fun,
interesting, we have exactly the same setting apart from 2.5 / 3 meters. Please look at the CF forum thread 'Colors and symbols for vector charts', testinstall and select the swedish classic and use the files there plus common files to see if you get the same result.
Could be something in those files then.

bdbcat, 2021-02-24:
Color name DEPCN is to be used as 'All depth contours other than the safety contour must use this colour'. 
Color name DEPSC, is to be used as 'This colour is reserved for the selected safety contour'

I believe that this is what is shown by the Taifun screenshot.

LennartG-Sve, 2021-02-25:
As there is no definition of 'safety contour' in the standard I will not argue on that, the depth setup also states *Safety Depth' so fine with me. 
The main question still remains though: It appears to work on some systems/setups but not on others. Why?

tai-fun, 2021-02-25:
Testing Swedish classic.
What stands out is the missing color between the 6 and 10 m curves. 
The color is uniform in the area between the 3 & 10 m lines.
[image]

LennartG-Sve, 2021-02-25:
tai-fun,
a) have you done the updates listed in #26? Not sure if erkswede has included those yet.
b) Pleas include some more info about your system. Mine will follow shortly.

LennartG-Sve, 2021-02-25:
Some, slightly reduced, system information:

cat /etc/os-release:
NAME="Ubuntu"
VERSION="20.04.1 LTS (Focal Fossa)"
root@mysak:~# uname -vrm
5.4.0-62-generic #70-Ubuntu SMP Tue Jan 12 12:45:47 UTC 2021 x86_64

root@mysak:~# lshw
*-core
description: Motherboard
product: Q1900-ITX
vendor: ASRock
physical id: 0
serial: M80-4C001601699
*-firmware
description: BIOS
vendor: American Megatrends Inc.
physical id: 0
version: P1.40
date: 10/31/2014
size: 64KiB
capacity: 8MiB
capabilities: pci upgrade shadowing cdboot bootselect socketedrom edd int13floppy1200 nt13floppy720 int13floppy2880 int5printscreen int9keyboard int14serial int17printer acpi usb biosbootspecification uefi
*-memory
description: System Memory
physical id: b
slot: System board or motherboard
size: 8GiB

*-cpu
description: CPU
product: Intel(R) Celeron(R) CPU  J1900  @ 1.99GHz
vendor: Intel Corp.
physical id: 14
bus info: cpu@0
version: Intel(R) Celeron(R) CPU J1900 @ 1.99GHz
slot: CPUSocket
size: 1625MHz
capacity: 2416MHz
width: 64 bits
clock: 83MHz

*-pci
          description: Host bridge
          product: Atom Processor Z36xxx/Z37xxx Series SoC Transaction Register
          vendor: Intel Corporation
          physical id: 100
          bus info: pci@0000:00:00.0
          version: 0e
          width: 32 bits
          clock: 33MHz
          configuration: driver=iosf_mbi_pci
          resources: irq:0
        *-display
             description: VGA compatible controller
             product: Atom Processor Z36xxx/Z37xxx Series Graphics & Display
             vendor: Intel Corporation
             physical id: 2
             bus info: pci@0000:00:02.0
             version: 0e
             width: 32 bits
             clock: 33MHz
             capabilities: pm msi vga_controller bus_master cap_list rom
             configuration: driver=i915 latency=0
             resources: irq:95 memory:d0000000-d03fffff memory:c0000000-cfffffff ioport:f080(size=8) memory:c0000-dffff

Hakansv, 2021-02-25:
@tai-fun 
Please remember the SE charts data base has 3, 6 and 10 meters curves. Nothing else in between 0 and 10 meters..
On this pict my settings are 3, 10 and 4 meters. If I set it to 2, 10 and 3 meters the light blue between 3 and 10 meters is of course not shown.
[image]

[image]

LennartG-Sve, 2021-02-25:
tai-fun,
not sure it is recommended to order the depths in that order, think they have to be sequential for the colors to display correct (at least on my system). Try 3 - 4 - 10 to get all colors. 
Still I notice that you show the contours that does not display on my system including contours for depths not in the option settings, deeper than 10 meters.

tai-fun, 2021-02-25:
Not quite correct. Raaaa hbr have an area dredged to 2.5m hence I start
with 2.5 to see that area.

tai-fun, 2021-02-25:
Generally Swedish charts have 3, 6 and 10 m contours. There is no 4m
contour in Swedish charts.
In Danish charts however you have 2,4,6 and 10m contours. This varies with
the different hydrographic offices.
Nothing wrong with the order I have listed the contour, it has worked well
the last  10 years.

erkswede, 2021-02-25:
@LennartG-Sve have you enabled 'Depth contour' in 'User standard options'?
[image]

LennartG-Sve, 2021-02-25:
Erik, 
the depth contour is enabled and I suspect a problem with the graphics system/setup. Have another machine, just like this one but another graphics system and with Ubuntu 16.04 that works. Could really be a graphics problem after all.

LennartG-Sve, 2021-02-25:
Guys,
it was indeed a problem with x11, xfwm4 / lightdm and / or xfce4.
Took the drastic action to remove / purge everything that had to do with the graphics and reinstalled the minimal set I really need and NOW IT WORKS. 
Thank you for all assistance and suggestions and now I have exactly the same setup of OpenCPN as before but with all contour lines showing.
Will close this tomorrow if that is ok.

LennartG-Sve, 2021-02-26:
After further tests and configuration (GUI-related) it still works OK. Closing

closed 2021-02-26 by LennartG-Sve