OpenCPN/OpenCPN #2178: arm64 plugin flatpak builds

leamas, 2021-02-26:
arm64/aarch64 is yet another platform we need to handle. For many reasons, users will migrate from current 32-bit (mostly Raspbian) OSes to 64 bit ones. Some already have.

The 64-bit Arm/Raspberry ecosystem seems to become more fragmented than the reasonably unified Raspbian 32-bit world. We have things like vanilla Debian, Ubuntu and 64-bit Raspbian (in beta) which likely all will be used. Realistically, we probably cannot maintain all of them. 

Therefore, it's seems like this might be the right time to enable the 64-bit flatpak builds (easy) and start builiding flatpak 64-bit plugins (hard). The plugin details are hairy.

---
Using CI services as today, there are not that many options available:

  1. CircleCI has a something in the oven, but it's [yet not available](https://www2.circleci.com/arm.html)
  2. cloud.drone.io has native Arm builders, but only docker instances. flatpak-builder is complicated to run on docker because if it's sandboxing, so using drone.io ir probably [not possible](https://discourse.drone.io/t/newbie-q-custom-seccomp-profile-flatpak-builder/8774)
  3. Gitlab totally open-source, and it's possible run a self-hosted instance. Such an instance could be configured with arm "runners" which should solve the problem.
  4. Travis offers Arm builders, but no free options for open-source projects. The idea to send email to their support to get extra "credits" is tested and failed.
  5. Github Actions (GA) does not provide arm64 builds. However, it's possible to run a self-hosted arm64 runner and [configure it as part of GA](https://docs.github.com/en/actions/hosting-your-own-runners/about-self-hosted-runners)

Of these options CircleCI looks best, besides that it isn't available. While the gitlab offer in many ways looks promising, it's a little more to handle than available for most devs working from home. 

So, while waiting for whatever offer CircleCI comes up with, perhaps resurrecting Github Actions builds combined with a self-hosted runner is the most realistic.

---

Here is also the question of the flatpak runtime. This needs to be updated, but we have been waiting for wxWidgets 3.2 before we doing  this. However, releasing a new, 64-bit build based on current gtk2-based runtime seems questionable, it's probably better to update the runtime for 64-bit version. There will be an ABI break for wxWidgets 3.2 anyway.

rgleason, 2021-02-26:
This is news to me about Travis.  Is that why travis is not building recently?
What should we be doing?   Sign up for  Drone?  Then change the script, 
OR 
just eliminate travis, and ARM, hang tight & wait for change of wxwidgets and Circleci  to support ARM?

bdbcat, 2021-02-26:
Rick...
Travis is building fine for me now.
https://travis-ci.org/github/bdbcat/oesenc_pi/builds/760533425

leamas, 2021-02-26:
Travis supplied a fixed amounts of "credits" for open source users (10000).  The official statement is that such users should be able to get  new ones after contacting their support. However, this doesn't work, neither for me nor others as described   [here](https://www.jeffgeerling.com/blog/2020/travis-cis-new-pricing-plan-threw-wrench-my-open-source-works).

We thus need to migrate from Travis. shipdriver has a working drone.io setup, which replaces and enhances Travis.

rgleason, 2021-02-26:
Thanks Dave and Alec,  I guess we're going to be forced to change eventually.
I also see that Dave linked to travis-ci.org   and there is a note at the top
"lease be aware travis-ci.org will be shutting down in several weeks, with all accounts migrating to travis-ci.com. Please stay tuned here for more information. "

I had moved all of Sean's plugins to their newer site travis-ci.com as they wanted us to do that sometime ago.
https://www.travis-ci.com/
On this site they state  "Start building for free"  I guess that means everybody and opensource does not get free services after using up the credits.

leamas, 2021-02-28:
hm... about to enable aarch64 flatpak builds. This reveals a broken URL, probably some left-over from the download links update:

` file-invalid          : <screenshot> failed to load [http://download.opencpn.org/screenshots/82434783-624dea00-9a61-11ea-847b-347d91b8f0fd.png]
`

@bdbcat: @nohal:  Anyone who knows the updated link?

leamas, 2021-02-28:
Walking around missing screenshot URL for now by attaching it here which brings me a working URL, Still, in the long run it's certainly better ta have it available on opencpn.org
[image]

bdbcat, 2021-02-28:
What is special about this screenshot?  Where in the build process is it referenced?

leamas, 2021-02-28:
The screenshot lives in data/opencpn.appdata.in. This is the picture displayed in the various Linux software "shops" like Software in Debian/Fedora and also on flathub.org.  The problem is revealed in the flathub commit checks.

bdbcat, 2021-03-01:
@leamas 
Here you go:
https://download.opencpn.org/s/QDiMPDfPSQL8TPE/download

leamas, 2021-03-01:
@bdbcat : Thanks. As usual, things are not that simple, filed #2179 :(

leamas, 2021-03-02:
There is now aarch64 builds available on flathub. However, these are using the old runtime, so it's just a transition phase.

Have a preliminary setup which builds aarch64 flatpak plugins. Working with @Rasbats  to get it in shape; it will take some time.

leamas, 2021-03-02:
There are now aarch64 builds available in the flathub beta repo. This repo also contains  an x86_64 build. Both are using the updated 20.08 runtime and are thus incompatible with current plugins.

To use the beta repo:

    $ flatpak remote-add --user flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo

And install flatpak using for example

    $ flatpak install --user org.opencpn.OpenCPN

leamas, 2021-03-05:
Here is now a provisionary way to build aarch64 plugins. These uses the updated 20.08 runtime. The setup also builds  x86_64 plugins for aarch64.

The first step is to create a github actions configuration file. An [example](https://github.com/Rasbats/shipdriver_pi/blob/master/.github/workflows/shipdriver.yml) is available in the shipdriver plugin.

Next step is to enable the builders. These are Leamas's, and not publicly available for security reasons. The procedure to enable them is documented in the [Managed Plugins Wiki](https://github.com/Rasbats/managed_plugins/wiki/Github-Actions)

As I see it, this issue can now be closed. Leaving it visible for some time, though.

leamas, 2021-03-14:
These plugins are blocked on #2185

leamas, 2021-04-11:
This  has now been visible for about a month. Closing.

leamas, 2021-04-20:
It should now be possible to build arm64 plugins without using my private arm64 builder:

  - circleci  has a preview program for arm64. It seems straight-forward to apply to participate in it, at least so for me.
  - travis seems to be back in open-source business (#2206), and by applying for open-source credits in their support it can actually be used.

I have testes both circleci and travis to the "hello arm64" point. There is some work to port the current github actions setup to either one of them or both. However, it seems worthwhile since using the official hardware is so much easier than using my private builder. 

Will make build chain available in https://github.com/Rasbats/shipdriver_pi/issues/171

leamas, 2021-04-20:
Re-opening until we have a stable build chain without dependencies on private hardware.

hreuver0183, 2021-04-27:
Blind testing results in this:

$ sudo apt install flatpak$ flatpak remote-add --user flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo
\<cut\>
$ flatpak list
$

The platform is not ready. I get this result on Ubuntu Focal and on Ubuntu Hirsute.
Needless to say this results in:

$ flatpak install --user  flathub-beta org.opencpn.OpenCPN
error: The application org.opencpn.OpenCPN/aarch64/beta requires the runtime org.freedesktop.Platform/aarch64/20.08 which was not found

To say the least,some documentation effort is needed. The above lines are not sufficient.
If you say the flatpak/aarch64 is ready I'm willing to test, but currently I'm not convinced it is the way to go.
There are huge differences in hardware between aarch64 and x86_64.

hreuver0183, 2021-04-27:
The commands needed are:
$ flatpak install flathub org.freedesktop.Platform//20.08
$ flatpak install --user flathub-beta org.opencpn.OpenCPN

This solves the installation, not the plugin issue.
(There is currently no testing. There is nothing to compare since the system is incomplete.)

leamas, 2021-04-27:
Yes. We need to work on the plugins, agreed.  Work is under way,  using recently available arm64 builders in the CI services.

And, we need write some docs...

leamas, 2021-05-04:
> The commands needed are [...]

Or, somewhat simpler: 

    $ flatpak install --user https://flathub.org/beta-repo/appstream/org.opencpn.OpenCPN.flatpakref

Small steps with the plugins. Added a PR to the shipdriver template: https://github.com/Rasbats/shipdriver_pi/pull/172. Once this is accepted, the first arm64 plugins should be come available. We then need to make other PRs to plugin communities like oesenc, radar and testplugin.

leamas, 2021-05-04:
There is now a stable build chain based on circleci without dependencies on my private hardware.

Still makes sense to track the progress in the plugins for some time, we need this implemented in the major plugin communities: oesenc/oernc, testplugin, radar, shipdriver.

leamas, 2021-05-05:
testplugin is already implemented. This means that besides the open PR against radar_pi everything is initiated w r t arm64 plugins flatpak builds.

Closing

closed 2021-05-05 by leamas