OpenCPN/OpenCPN #2179: opencpn.appdata.in: Broken link

leamas, 2021-03-01:
The file data/opencp.appdfata.in has a broken link to a screenshot as discussed in https://github.com/OpenCPN/OpenCPN/issues/2178#issuecomment-787421157

leamas, 2021-03-01:
At a closer look, it also fails validation due to being too big. And, a bit silly since it's only used on Linux: the windows toolbar at bottom.

@bdbcat: Attaching a new version. Would appreciate if this was uploaded to opencpn.org, and a new URL so we can patch the medata file.

[image]

bdbcat, 2021-03-01:
@leamas 
https://download.opencpn.org/s/ZNRdz7K8XXdTxnR/download
Still pretty big....

leamas, 2021-03-01:
Thanks!

> Still pretty big....

The max and also recommended size is 1600 x 900, and this is more or less that (removing the windows decorations shaved off some pixels).

bdbcat, 2021-03-01:
Ahh, big physical, not big bytes.

PR #2180 (2021-03-01): appdata: Fix broken screenshot link (#2179)
Fix broken link after update on opencpn.org. While on it, scale
image to recommended size and remove windows decorations.

Closes: #2179
commit a1a9be78 (2021-03-01): appdata: Fix broken screenshot link (#2179)

Fix broken link after update on opencpn.org. While on it, scale
image to recommended size and remove windows decorations.

Closes: #2179
closed 2021-03-01 by bdbcat