OpenCPN/OpenCPN #2185: Plugin compatibility check broken (arm64/aarch64)

leamas, 2021-03-14:
Testing 64-bit arms plugins reveals multiple problems with the compatibility check in PluginHandler.cpp:isCompatible():

  1. For flatpak plugins on 64-bit ARM, the platform is ambigous and becomes flatpak-aarch64 or flatpak-arm64  
      depending on  host. 
  1. It has been observed that the host platform is defined as 'notfound-x86_64' which obviously will fail.
  1. The flatpak universe uses flatpak-aarch64 and we should stick to that. The xsd file in the plugins
      project needs to be updated to reflect this.
  1. The initial platform check in isCompatible basically is something like (simplified)
  
         if (metadata.target_arch != os_detail->osd_arch && metadata.target_arch != os_detail->osd_build_arch)

     This is wrong since it sidesteps all the mechanisms available to override the package platform such as the
     environment variable OPENCPN_COMPAT_TARGET, the configuration file variables CompatOS and 
     CompatOsVersion and also the new choices done in the Catalog Settings. These updates  the strings 
     compatOS and compatOsVersion, but the check above might fail without respecting these values

leamas, 2021-03-14:
Looking further into this here is now also `<build-target>` besides `<target>`. I'm  probably missing something here, but: isn't 'build-target'  just a band-aid for the fact  that 'target', which is deduced from the build host, is plain wrong when cross-building? If so, wouldn't it be easier to fix 'target' so it contains the correct value for the target platform rather than adding a new variable?

"Confused"

@bdbcat : Thoughts?

rgleason, 2021-03-14:
I think maybe build-target was added later to fix the gtk3 issue

bdbcat, 2021-03-14:
`This is wrong since it sidesteps all the mechanisms available to override the package platform`
Somehow, updates to this entire method did not get merged into master, although it exists correctly in branch v5.2.x.
I'll push a correction.

leamas, 2021-03-15:
Unfortunately, that fix doesn't really help. It basically provides "better" values for compatOS and compatOSVersion, but these values are not used when the initial test around  line 283 silently fails. 

IMHO, this whole function needs a redesign due to:
  - There should not be a need to consult the runtime version info to figure out the host. We know already at configure time if the code is aimed for for example Android-eabi-v7a, flatpak-aarch64, etc.
  - The only exception is the list of compatible OS:es, this is only available in runtime. OTOH, these checks primary used for Mint doesn't really work today.
  - Not having proper target types for Android has meant introducing some kind of "subtypes" for armhf/arm64 adding more complexity
  - Overall, it's just too much code in a single method to maintain (as can be seen here).

leamas, 2021-03-16:
The #2188 is now tested on flatpak-aarch64 and works occasionally. The  CI builds takes care of most other rplatform, android remains to be tested.

bdbcat, 2021-03-16:
`works occasionally`
What does that mean?

leamas, 2021-03-16:
It means that I have tested one or two times, and it worked those times. Certainly not a comprehensive test suite.

Bear in mind that I'm a dumb Swede, and the nuances in my Englisch  are, well, you know

leamas, 2021-03-16:
With some luck, Mike might be able to do some more flatpak  testing: https://github.com/Rasbats/shipdriver_pi/issues/153#issuecomment-800375741

bdbcat, 2021-03-16:
I have lots of boat electrical hardware that "works occasionally".  If I thought software was subject to the same rules, I'd give up....
Cheers

leamas, 2021-03-17:
After merging #2188 this should be fixed. However, there is #2189 instead. Closing

closed 2021-03-17 by leamas